xCuri0/ReBarUEFI #1: Not compatible with Ivy Bridge E/EP boards

rliwoch, 2022-08-30:
Hi @xCuri0,

First of all - great effort. Stumbled upon your post on reddit. 

I glanced over the source code before building just to check if there's nothing super malicious there. Frankly, I don't have enough EFI knowledge to judge but the C code looked sane, so I went on to build the module all singing and dancing based on the promise of Ivy Bridge **+** readied myself for a treat.  
I've built the ffs (you gotta love that extension) and when I fired up the MMTool it struck me that on X79/C602 chipset BIOSes - that is Ivy Bridge E/EP (high end 4th gen i7s and Xeons v1/v2) - there's no `PciHostBridge`. There's `PciRootBridge`.

Now, seeing that you've got hooks to PciHostBridge I decided not to flash it as I guess it would be pointless to try in this instance. 

Effectively I'm trying to enable Nvidia Tesla M40 24GB on a Supermicro X9 server motherboard - I have a couple of those boards and I'm lead to believe some support ReBAR and some not. Some basically were certified to run with 24GB K80s, so there's no other option for such a memory hog to run without ReBAR.
From newer systems - I'm certain that M40 requires ReBAR/CSM disabled/full blown UEFI to work. The cards is detected but `lspci -vvv` shows all memory regions as disabled. 

On top of that I also have a couple of old X79 boards and double checked their BIOSes and it's a similar story to C602 server chipset. 
Here's the screenshot, it presents the modules in GA-X79-UD3, Supermicro X9DRI-F and for comparison, the BIOS for the board you mentioned in your reddit post GA-B75M.

[image]

BTW. This supermicro board has a MMIOH of up to 1T, only mentioning it because on reddit I think you mentioned Ivy goes only up to 64G, I'm sure you meant consumer boards; But just mentioning this for completeness.
[image]

All in all, I think it would be fair to change the description in this repo to `ivy bridge` instead of `ivy bridge+`. The + may suggest anything above Ivy Bridge, but E/EP are above and there's this clash in the module names.

I'd love to provide you with a patch but my EFI knowledge doesn't go that deep unfortunately.

xCuri0, 2022-08-31:
@rliwoch
> Now, seeing that you've got hooks to PciHostBridge
Actually it hooks the PciBus driver which gets called by PciHostBridge. On your system it's probably called by PciRootBridge but I can't be sure. And I would recommend using UEFITool unless you know MMTool works for you

rliwoch, 2022-08-31:
Cheers for answering. I'll poke around.
Any specific reason you're recommending UEFITool? I remember reading once that MMTool can mess up paddings? is it what you're getting at?

Still just letting you know that as it's stands it's not for every Ivy Bridge and up. Interestingly enough 4G Decoding definitely works on those boards as Quadro M4000 with 8gigs of vRAM works no issues. It's just Teslas are a pain in the butt, requiring ReBAR

xCuri0, 2022-08-31:
@xCuri0
>Any specific reason you're recommending UEFITool

Not any reason besides it working for me (didn't try MMTool), open source and apparently MMTool sometimes causes issues according to some people.

rliwoch, 2022-08-31:
MMTool worked for me for injecting NVMe boot capabilities on X79 back in the day. But I take your point. 

One more question - you did all your work based on the tianocore source code, right?   [PciHostBridge.c](https://github.com/tianocore/edk2/blob/7c0ad2c33810ead45b7919f8f8d0e282dae52e71/MdeModulePkg/Bus/Pci/PciHostBridgeDxe/PciHostBridge.c)? 
Not some reversed engineered Asus bios or something like that.
I'm starting to see the relation between the Host and Root bridge. Need some more time after hours to fully understand what's going on in there.

rliwoch, 2022-09-01:
@xCuri0 

TL;DR; I think it may actually work. Will try it. I somehow missed what you said yesterday
> PciBus driver PciHostBridgeResourceAllocationProtocol which gets called by PciHostBridge

I got panicky when you said "insert before HostBridge" and I don't have a HostBridge :) If it works I'll let you know.

Would you care to share your `lspci | grep -I  bridge` when you got a second?

Mine:
```
root@pve:~# lspci | grep -I  bridge
00:00.0 Host bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 DMI2 (rev 04)
00:01.0 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 1a (rev 04)  (CPU 0 - Physical port 1 bifurcated and sharing lanes with PCH)
00:01.1 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 1b (rev 04) (CPU 0 - Physical port 1 bifurcated)
00:02.0 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 2a (rev 04) (CPU 0 - Physical port 2 NOT bifurcated)
00:03.0 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 3a (rev 04) (CPU 0 - Physical port 3 bifurcated)
00:03.2 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 3c (rev 04) (CPU 0 - Physical port 3 bifurcated)
00:11.0 PCI bridge: Intel Corporation C600/X79 series chipset PCI Express Virtual Root Port (rev 06)  (VIRTUAL BRIDGE? on PCH?)
00:1c.0 PCI bridge: Intel Corporation C600/X79 series chipset PCI Express Root Port 1 (rev b6) (things PCI attached over PCH)
00:1e.0 PCI bridge: Intel Corporation 82801 PCI Bridge (rev a6) (same here)
00:1f.0 ISA bridge: Intel Corporation C600/X79 series chipset LPC Controller (rev 06) (same here)
80:00.0 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port in DMI2 Mode (rev 04) (virtual root bridge representing PCI devs physically to CPU1?)
80:01.0 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 1a (rev 04) (CPU 1 - Physical port 1 NOT bifurcated at present)
80:02.0 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 2a (rev 04) (CPU 1 - Physical port 2 NOT bifurcated at present)
80:03.0 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 3a (rev 04) (CPU 1 - Physical port 3 NOT bifurcated at present)
```
^ the above would make sense in view of the actual system diagram:
[image]

I went to some dark places last night, like:
[EFI Spec /section 12-1](https://www.intel.com/content/dam/doc/product-specification/efi-v1-10-specification.pdf)
[E5 v2 spec vol1](https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/xeon-e5-v2-datasheet-vol-1.pdf)
[E5 v2 spec vol2](https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/xeon-e5-v2-datasheet-vol-2.pdf)
and 
[PCI DXE guidelines](https://edk2-docs.gitbook.io/edk-ii-uefi-driver-writer-s-guide/18_pci_driver_design_guidelines)
as well as
[PCI Enumeration diagrams](https://docplayer.net/87909716-Edk-ii-topology-pci-enumeration.html)

I think the reason why X79/C602/C602J may need this PciRootBridge driver is two fold:

- provide a virtual root port (bridge?) on 2 and 4 CPU boards, especially when NUMA is enabled. In the and physical PCIe are bound to a specific CPU but with NUMA a device can access the entire memory pool
- allow for more granular control of physical PCI ports found on (typically) server boards - i.e. Bifurcation of the port

Intel in fact says DMI2 Host Bridge or PCIe Root Port
[image]
If I'm reading it right - all bifurcated ports on a single physical port are under a PCI-to-PCI bridge

The EDK II Topology PCI Enumeration says:

> PcAtChipsetPkg/PciHostBridgeDxe/PciHostBridgeDxe.inf **(often overridden)**

Which I think further reinforces the point I made above

`PciBus` driver actually says:
>  Single PCI bus driver instance will manager all PCI Root Bridges in one EFI based firmware,
  since all PCI Root Bridges' resources need to be managed together.
  Supported() function will try to get PCI Root Bridge IO Protocol.
  Start() function will get PCI Host Bridge Resource Allocation Protocol to manage all
  PCI Root Bridges. So it means platform needs install PCI Root Bridge IO protocol for each
  PCI Root Bus and install PCI Host Bridge Resource Allocation Protocol.

Since I clearly have a Host Bridge in `lspci` I assert that:

1.  `PciRootBridge` driver might be just an extended version of `PciHostBridge`. 
2. Looking at the `InitializePciHostBridge` function implementation it in fact iterates through all the bridges

``` 
RootBridges = PciHostBridgeGetRootBridges (&RootBridgeCount);
[...]
for (Index = 0; Index < RootBridgeCount; Index++) {
```
and then finally get to trigger `PciHostBridgeResourceAllocationProtocol`

Just leaving it here, in case anyone will look for it in the future.

xCuri0, 2022-09-01:
> Would you care to share your `lspci | grep -I bridge` when you got a second?

@rliwoch 
```
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor DRAM Controller (rev 09)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor PCI Express Root Port (rev 09)
00:1c.0 PCI bridge: Intel Corporation 7 Series/C216 Chipset Family PCI Express Root Port 1 (rev c4)
00:1c.4 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 5 (rev c4)
00:1e.0 PCI bridge: Intel Corporation 82801 PCI Bridge (rev a4)
00:1f.0 ISA bridge: Intel Corporation B75 Express Chipset LPC Controller (rev 04)
```

rliwoch, 2022-09-01:
So before flashing I decompiled the existing BIOS with `ghidra` to the extent I was indeed able to see that there is a HostBridge in there somewhere created and `PciHostBridgeResourceAllocationProtocol` would be called. 

- Compiled the module with ReBar set as 19 = 2^19.
- Injected the `ReBarDXE` between the `PciBus` and `PciRootBridge`
- Unfortunately no worky-worky 

all memory regions are still disabled. Tried various MMIOH settings, of course that's UEFI boot and Above 4G Decoding enabled. 
[image]

xCuri0, 2022-09-02:
@rliwoch send dmesg (if you can) would be useful in seeing why the memory regions get disabled.

also you have to create RebarState efi variable and set that to needed size to enable it. Can you also send the output of lspci that shows the Resizable BAR capability ?

xCuri0, 2022-09-02:
> you mean with `dmpstore`? or mounting `efivarfs` - or either is good.

not too sure about it. i used a windows application i made to set the efi variable because i didn't think of the case of nvidia gpus on linux (amd doesn't need bios support for resizable bar on linux). source code is here [RebarState.cpp](https://gist.github.com/xCuri0/23307bb0cf6f02a8be8df3a3276139cb)

rliwoch, 2022-09-02:
Interestingly enough M40 is the only device that doesn't even attempt to do anything with BAR, 10gig Intel NIC and LSI SAS controller both create BARs. 

[dmesg.txt](https://github.com/xCuri0/ReBarUEFI/files/9480878/dmesg.txt)
[lsi2008.txt](https://github.com/xCuri0/ReBarUEFI/files/9480879/lsi2008.txt)
[m40.txt](https://github.com/xCuri0/ReBarUEFI/files/9480880/m40.txt)

I also checked RTX3070 I have on another linux box and BARs are correctly reported. 

About variable - need to read a bit about it, I can mount and see efivars after mounting `efivarfs`; but need check how to write to it, if simply echoing to the efivarfs is enough

xCuri0, 2022-09-03:
> [    1.498350] pci 0000:84:00.0: reg 0x14: [mem 0x39f000000000-0x39f7ffffffff 64bit pref]

This does appear to be 32GB so the ReBar module is working.

> [    5.427404] nvidiafb 0000:84:00.0: enabling device (0000 -> 0002)
[    5.427546] nvidiafb: Device ID: 10de17fd 
[    5.427549] nvidiafb: unknown NV_ARCH

This appears to be the problem so the issue is with the Nvidia driver ?

rliwoch, 2022-09-03:
No that's a red herring. Forgot to tell you to ignore that. 
This module is normally turned off in mod probe - i just turned it on for completeness for you. 

That's a virtualisation server so the host normally has all nvidia kernel modules (open source or proprietary) turned off in `modprobe`. 
The only thing the host OS has to do is to load the hardware properly and in separate IOMMU groups. If I install NVIDIA proprietary driver on the host then their kernel modules will simply refuse to load. 

If I pass through this card to Linux VM with nvidia drivers it shouts at me that the BAR is 0M
If I do it for a Windows VM - Code12 in device manager, not enough resources. 

I put any other card in - and it works perfectly. I can insert Quadro M4000 8G vRAM into PCIe and it just works. 
I'm using PCIe passthrough on KVM

xCuri0, 2022-09-03:
> If I pass through this card to Linux VM with nvidia drivers it shouts at me that the BAR is 0M
can you show the dmesg of this

rliwoch, 2022-09-03:
[dmesg-guest.txt](https://github.com/xCuri0/ReBarUEFI/files/9481882/dmesg-guest.txt)

```
[    0.609605] PCI: Using ACPI for IRQ routing
[    0.765457] PCI: pci_cache_line_size set to 64 bytes
[    0.765509] pci 0000:00:1a.0: can't claim BAR 4 [io  0xd300-0xd31f]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.765534] pci 0000:00:1a.1: can't claim BAR 4 [io  0xd2e0-0xd2ff]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.765553] pci 0000:00:1a.2: can't claim BAR 4 [io  0xd2c0-0xd2df]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.765662] pci 0000:00:1d.0: can't claim BAR 4 [io  0xd2a0-0xd2bf]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.765680] pci 0000:00:1d.1: can't claim BAR 4 [io  0xd280-0xd29f]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.765698] pci 0000:00:1d.2: can't claim BAR 4 [io  0xd260-0xd27f]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.765838] pci 0000:00:1f.2: can't claim BAR 4 [io  0xd240-0xd25f]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.765857] pci 0000:00:1f.3: can't claim BAR 4 [io  0xd200-0xd23f]: address conflict with PCI Bus 0000:01 [io  0xd000-0xdfff]
[    0.765973] pci 0000:01:00.0: can't claim BAR 0 [mem 0xff000000-0xffffffff]: no compatible bridge window
[    0.765978] pci 0000:01:00.0: can't claim BAR 1 [mem 0xfffffff800000000-0xffffffffffffffff 64bit pref]: no compatible bridge window
[    0.765983] pci 0000:01:00.0: can't claim BAR 3 [mem 0xfffffffffe000000-0xffffffffffffffff 64bit pref]: no compatible bridge window
```

Interesting. Don't know why I didn't think to look in there. I just assumed I'll first try to solve it on the host.

closed 2022-10-30 by xCuri0