xCuri0/ReBarUEFI #2: Howto build on Linux?

hansbogert, 2022-10-02:
I did the following:

```shell
git clone https://github.com/tianocore/edk2.git
cd edk2
git clone https://github.com/xCuri0/ReBarUEFI.git
cd ReBarUEFI/ReBarDxe
python3 buildffs.py

Traceback (most recent call last):
  File "/home/gandalf/src/edk2/ReBarUEFI/ReBarDxe/buildffs.py", line 18, in <module>
    subprocess.run(["build", "--platform=ReBarUEFI/ReBarDxe/ReBar.dsc"], shell=shell, env=os.environ, stderr=sys.stderr, stdout=sys.stdout)
  File "/usr/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'build'

```

xCuri0, 2022-10-03:
You need to setup edk2 environment before building

[https://github.com/tianocore/tianocore.github.io/wiki/Common-instructions](https://github.com/tianocore/tianocore.github.io/wiki/Common-instructions)

closed 2022-10-26 by xCuri0

hansbogert, 2022-10-26:
@xCuri0 never thanked you, hereby, thanks!