xCuri0/ReBarUEFI #4: Boots into windows but unable to access bios with ReBAR at or greater than 4gb

SqueakyAlex, 2022-10-29:
- Motherboard: Gigabyte z170x-Gaming 7 rev. 1.0
 - BIOS Version: F22m
 - GPU: Radeon RX6800

Easily managed to set it up and get it running. 
After some quick testing I found out that...

1gb = According to Radeon drivers (ver22.10.2) and GPU-Z, ReBAR gets enabled, works as expected.
2gb = Halts on bootlogo, unable to enter bios. Need to reset bios.
4gb+ = Boots into Windows but unable to see bootlogo during boot and unable to enter bios, trying to enter bios using the Windows troubleshooting menu also fails. (tested up to 16gb with the same behavior)

Did some benchmarks on 4gb and 16gb to test for system instability, all seemed fine except that I was unable to see the bootlogo on boot or access the bios which can be confusing because during the whole booting process it does seem like the computer has failed to post.

[image]

xCuri0, 2022-10-30:
CSM is off right ? Don't believe what GPU-Z says about CSM or 4G decode because it's often wrong

SqueakyAlex, 2022-10-30:
Yes, verified that CSM is off and 4G decode is enabled.

I do get a performance boost when its set to 4gb+, its just that I'm unable to enter the bios and the screens stay off during boot until windows logon screen shows.

Just noticed on your winraid forum post that someone else is having a similar problem with the same board, just diff gpu.

xCuri0, 2022-10-30:
@SqueakyAlex yeah the bios probably needs patching. I had to patch my bios (gigabyte b75m-d3h) to get BARs over 2GB to work.  But I haven't made a patch to fix this issue since I don't have it.

Can you send a screenshot of Device Manager resources tab for your GPU when using 1GB BAR ?

SqueakyAlex, 2022-10-30:
1GB Bar
[image]

Some unrequested extras
4GB Bar
[image]

16GB Bar
[image]

Had not noticed before, high definition audio bus gets disable when using 4GB+ bar.
[image]

xCuri0, 2022-10-30:
@SqueakyAlex the issue of High Definition Audio Bus not having enough resources can be fixed by patching the PciHostBridge module's call to AddMemorySpace to increase the size it uses (and DSDT QWordMemory too if needed).

SqueakyAlex, 2022-10-30:
Assuming your patches would not work for Skylake, I spent a few hours checking pcihostbrige on ghidra and came to the conclusion that I don’t know what I’m looking for and that I know nothing about assembly. 

Eventually I did try your patches and as expected the result was "No patches can be applied to input file"

I did manage to extract my DSDT and found this

```
                CreateDWordField (BUF0, \_SB.PCI0._Y0E._MIN, M1MN)  // _MIN: Minimum Base Address
                CreateDWordField (BUF0, \_SB.PCI0._Y0E._MAX, M1MX)  // _MAX: Maximum Base Address
                CreateDWordField (BUF0, \_SB.PCI0._Y0E._LEN, M1LN)  // _LEN: Length
                Store (M32L, M1LN) /* \_SB_.PCI0._CRS.M1LN */
                Store (M32B, M1MN) /* \_SB_.PCI0._CRS.M1MN */
                Subtract (Add (M1MN, M1LN), One, M1MX) /* \_SB_.PCI0._CRS.M1MX */
                If (LGreaterEqual (OSYS, 0x07D6))
                {
                    If (LEqual (M64L, Zero))
                    {
                        CreateQWordField (BUF0, \_SB.PCI0._Y0F._LEN, MSLN)  // _LEN: Length
                        Store (Zero, MSLN) /* \_SB_.PCI0._CRS.MSLN */
                    }
                    Else
                    {
                        CreateQWordField (BUF0, \_SB.PCI0._Y0F._LEN, M2LN)  // _LEN: Length
                        CreateQWordField (BUF0, \_SB.PCI0._Y0F._MIN, M2MN)  // _MIN: Minimum Base Address
                        CreateQWordField (BUF0, \_SB.PCI0._Y0F._MAX, M2MX)  // _MAX: Maximum Base Address
                        Store (M64L, M2LN) /* \_SB_.PCI0._CRS.M2LN */
                        Store (M64B, M2MN) /* \_SB_.PCI0._CRS.M2MN */
                        Subtract (Add (M2MN, M2LN), One, M2MX) /* \_SB_.PCI0._CRS.M2MX */
                    }
                }
```

xCuri0, 2022-10-30:
@SqueakyAlex Just send a screenshot of Device Manager showing Resources By Type Large Memory it will show how much the space the DSDT gives.

SqueakyAlex, 2022-10-30:
[image]

Pentium320, 2022-11-07:
Hi,
Looks, like I'm having the same issue on X99 platform, maybe there is something, that could be helpful.

- CPU: Intel Xeon E5-2697 v3 2.6GHz
- Motherboard: ASRock X99 Extreme4
- BIOS Version: 3.60 (with Turbo Unlock mod)
- GPU: AMD Radeon RX 6700 XT 12GB

Behaviour is the same as in this thread after applying ReBAR patch:
up to 1024M (2^10) - ReBAR seems to be working as intended
2048M (2^11) - system doesn't post at all, speaker gives 8 beeps (AFAIK it indicates GPU error). CMOS reset needed.
4096M (2^12) and above - unable to enter BIOS, no Post Screen, speaker gives 5 beeps, PC boots directly into Windows and ReBAR seems to be working. AMD HD Audio is broken (error code: 12). Performance has increased a bit.

[image]

CSM disabled, 4G decode enabled.

If there is anything useful, that I could provide to resolve the issue, tell me. Thanks for your work!

xCuri0, 2022-11-08:
If anyone wants to try and develop patches to fix it you can search for "ffffffff000" in PciBus module on your disassembler find the code that reads the BAR size and categorizes them as 32-bit (4/5) or 64-bit (6/7).

soulpower11, 2022-11-10:
My High Definition Audio Bus got disabled too
[image]

Not sure what to do as I am new to bios modding

xCuri0, 2022-11-13:
Try booting Linux with 4GB+ BAR and seeing what dmesg says.

xCuri0, 2022-12-24:
One Asrock X99 Taichi user on win-raid found this
[image]

So try disabling the option **Above 4G Decoding Patch** using the IFR method if it's hidden

Dadovvv, 2023-06-09:
can anyone tell me how to disable Above 4g Decoding Patch? what line should i edit and what?

xCuri0, 2023-06-09:
@Dadovvv please read wiki common issues and fixes. if you can't find the string then the setting doesn't exist

Igo131, 2024-01-18:
> One Asrock X99 Taichi user on win-raid found this out [image]
> 
> So try disabling the option **Above 4G Decoding Patch** using the IFR method if it's hidden

This works only for mobos that don't have native Above 4G Decoding option or i should try this even if i have the switch in my mobo?

MrAnonymous7122, 2024-03-16:
Hi, I have a H110M-R which has a LGA1151 socket. For me, display blacks out mid way through the booting process (it's past the boot screen with del for bios and then past the asus logo with win11 spin). My CPU does has an iGPU but the problem is I didn't build this PC myself and the motherboard IO shield has something physically blocking the port from being used. Is there anyway around this or do I have to remove the motherboard from the case and remove the blocker? I'm not sure if this is relevant but the mouse and keyboard are still receiving power and it shuts down normally upon pressing the power button on the case.

MrAnonymous7122, 2024-03-16:
Strangely, after writing this comment it works? It's booted to windows normally and display and sound work fine through HDMI. I will add to working motherboards in a sec.

mirh, 2024-03-18:
FWIW I had [found out](https://old.reddit.com/r/buildapc/comments/niucyh/black_screen_when_enabling_csm/jv13obj/) that when I'm just using the igp of my i5-9600K (on a Z390 GAMING X with F10i bios) I'm getting a black screen with aperture sizes higher than 1024MB.

crazyhulk, 2024-07-09:
> One Asrock X99 Taichi user on win-raid found this out一位 ASRock X99 Taichi 用户在 win-raid 上发现了这一点 [image]
> 
> So try disabling the option **Above 4G Decoding Patch** using the IFR method if it's hidden因此，如果该选项被隐藏了，尝试使用 IFR 方法禁用“大于 4G 解码补丁”选项

I have the same case, I can not boot into bios when rebar unlimited . There is an option Above 4G Decoding on my bios。
[image]

“Try disable the option”， Did I see it wrong?

JGSphaela, 2024-09-24:
I believe this issue is **more severe on Intel GPUs**, as referenced in [#66](https://github.com/xCuri0/ReBarUEFI/issues/66) and corroborated by my own testing.

### System Specifications:
- **Motherboard**: Gigabyte Z170x-Gaming 3 (rev. 1.0)
- **BIOS Version**: Modified 22j with Coffee Lake support
- **GPU**: Intel Arc A770

I’ve ensured that:
- CSM support is disabled.
- Above 4G Decoding is enabled.
[image]
[image]

### Steps Taken:
1. I followed the BIOS modification instructions from the wiki and successfully enabled ReBar at 1GB.
2. GPU-Z correctly reports that ReBar is enabled at 1GB.
   
   [image]

3. Despite this, Intel’s driver still shows ReBar support as disabled.
   
   [image]

4. In Device Manager, ReBar at 1GB shows correctly, but when I try settings of 4GB or higher, the GPU stops working. I lose display output, but the system still boots fine into Windows, and I have to use the iGPU to access the system. In Device Manager, the Arc A770 shows error code 12: “This device cannot find enough free resources...”.

   [image]

5. I’ve attempted:
   - Reinstalling drivers
   - Disabling iGPU
   - Connecting both iGPU and GPU at the same time
   - Nothing has worked.

I also experimented with a 2GB ReBar size for fun, and as expected, it failed:

   [image]

Intel Arc GPUs need ReBar enabled to be fully functional, so this issue is really a bummer. Maybe consider adding Intel GPU not working to the **Common Issues** list under the LGA1151 various issues for now. I really want to fix this and if there's anything else I can provide please let me know!