xCuri0/ReBarUEFI #5: UEFIPatch: Increase MMIO space hangs Windows on boot screen. More info on DSDT mod?

slayermaster, 2022-10-30:
**System**
 - Motherboard: Z77X-D3H
 - BIOS Version: F18i
 - GPU: GTX 1070

**Description**
Patching the bios MMIO causes Windows to hang on boot screen and will not proceed to login screen. Seen that it might need DSDT mod, any more info that?

xCuri0, 2022-10-30:
You need to extract the DSDT from the Amiboardinfo module with [AmiboardinoTool](https://github.com/xCuri0/AmiBoardInfoTool). Decompile the DSDT and look for ```CreateQWordField```. It will be something like this probably (stock on my Gigabyte B75M-D3H)

```
                    CreateQWordField (BUF0, \_SB.PCI0._Y0F._LEN, M2LN)  // _LEN: Length
                    CreateQWordField (BUF0, \_SB.PCI0._Y0F._MIN, M2MN)  // _MIN: Minimum Base Address
                    CreateQWordField (BUF0, \_SB.PCI0._Y0F._MAX, M2MX)  // _MAX: Maximum Base Address
                    M2LN = 0x0000000400000000
                    If ((TUUD >= 0x1000))
                    {
                        M2MN = (TUUD << 0x14)
                    }
                    Else
                    {
                        M2MN = 0x0000000100000000
                    }

                    M2MX = ((M2MN + M2LN) - One)
```
Change it to something like (my modded DSDT)
```
                    CreateQWordField (BUF0, \_SB.PCI0._Y0F._LEN, M2LN)  // _LEN: Length
                    CreateQWordField (BUF0, \_SB.PCI0._Y0F._MIN, M2MN)  // _MIN: Minimum Base Address
                    CreateQWordField (BUF0, \_SB.PCI0._Y0F._MAX, M2MX)  // _MAX: Maximum Base Address
                    // 36-bit for Ivy Bridge
		    M2MX = 0x0000000FFFFFFFFF

		    If ((TUUD >= 0x1000))
		    {
		        M2MN = (TUUD << 0x14)
		    }
		    Else
		    {
		        M2MN = 0x0000000100000000
		    }
			
		    M2LN = (M2MX - M2MN) + One
```
Recompile the DSDT. I recommend that you remove what's not needed in the DSDT to make the size be smaller than the original to avoid potential issues. Again use AmiboardinfoTool to rebuild the Amiboardinfo module and replace it with the newly generated Amiboardinfo using UEFITool

slayermaster, 2022-10-30:
I have that exact code, probably because we both have Gigabyte board. Will have a try and report back later.

slayermaster, 2022-10-30:
It works flawlessly, flashing the bios with the new DSDT boots straight into Windows.
I was hoping by increasing the MMIO space, it'll fix Hyper-V's GPU-P _"upper MMIO region is above the supported physical address range"_ but sadly it did not. Error still pops up.

Still, good to know that I can turn on ReBar on my trusty ol mobo.

xCuri0, 2022-10-30:
I will add more info about doing the DSDT mod. Closing

closed 2022-10-30 by xCuri0