xCuri0/ReBarUEFI #7: How do I compile it for Ubuntu?

Tokobotenkai, 2022-11-02:
Do you have a step by step guide on how to go about it?

I am having issues trying to setup the edk2 environment, seems like one of the modules is missing (at least following this guide: https://github.com/tianocore/tianocore.github.io/wiki/Common-instructions).

Or do I have to try a specific release of edk2?

I am running Ubuntu 22.04 as I am trying to get the new Intel Arc GPUs running which seems to require ReBar.

Thanks in advance.

xCuri0, 2022-11-03:
I used the latest EDK2 on Ubuntu 22.04 without problems. What's the exact error that you are getting ?

Tokobotenkai, 2022-11-03:
I realised I wasn't using the latest so I did this:

```
mkdir ~/src
cd ~/src
git clone https://github.com/tianocore/edk2.git
cd edk2
git submodule update --init
make -C BaseTools
. edksetup.sh
git clone https://github.com/xCuri0/ReBarUEFI.git
gedit Conf/target.txt
ACTIVE_PLATFORM       = ReBarUEFI/ReBarDxe/ReBar.dsc
[#TARGET](tg://search_hashtag?hashtag=TARGET)                = DEBUG
TARGET_ARCH           = IA32 X64
TOOL_CHAIN_TAG        = GCC5
MAX_CONCURRENT_THREAD_NUMBER = 9
build
cd ReBarUEFI
cd ReBarDxe
python3 [buildffs.py](https://buildffs.py/)
```

I got this output:
```
Build environment: Linux-5.17.0-1019-oem-x86_64-with-glibc2.35
Build start time: 12:52:01, Nov.03 2022

WORKSPACE        = /home/azrael/src/edk2
EDK_TOOLS_PATH   = /home/azrael/src/edk2/BaseTools
CONF_PATH        = /home/azrael/src/edk2/Conf
PYTHON_COMMAND   = /usr/bin/python3

- Done -
Build end time: 12:52:01, Nov.03 2022
Build total time: 00:00:00

Build failed
```

Any ideas?

xCuri0, 2022-11-03:
What happens when you use the default target.txt ?

Tokobotenkai, 2022-11-03:
Running build without the python script gives an output without the build failed line with the modified target.txt

I could try to run it with the defaults again.

xCuri0, 2022-11-03:
The following worked for me
```
git clone https://github.com/tianocore/edk2.git
cd edk2
git submodule update --init
make -C BaseTools
source ./edksetup.sh
```
Edit ```Conf/target.txt```
```
TARGET = RELEASE
TARGET_ARCH           = X64
TOOL_CHAIN_TAG        = GCC5 (change this if you're on Windows or want to use Clang)
```
```
git clone https://github.com/xCuri0/ReBarUEFI.git
cd ReBarUEFI/ReBarDXE
python3 ./buildffs.py
```

Tokobotenkai, 2022-11-03:
Wanted to say it works now!

Thank you.

I get an error now in the ASUS EZ Flash utility (in the BIOS) saying that selected file is not a proper BIOS. Any idea?

Just to check if I just need to add the FFS file after the last entry of the DXE driver section (the one containing CSMCore) correct?

The efi file generated with the FFS is not required correct?

xCuri0, 2022-11-03:
@Tokobotenkai what motherboard is it ?

Tokobotenkai, 2022-11-03:
The motherboard is an ASUS Z97-PRO, from my understanding [here](https://winraid.level1techs.com/t/howto-get-full-nvme-support-for-all-systems-with-an-ami-uefi-bios/30901/5900?page=296):

> First, use UEFITool on the stock.CAP BIOS to extract the body >> Right click AMI Aptio Capsule and choose extract body, Then do NVME mod to that bin/rom
> This is how you should do that >> open stock.CAP in Hex editor and go to offset 800. Open mod bios rom/bin in second tab of hex editor, select all and copy, then paste at offset 800 in stock.cap, then save this file as M5F.CAP
> 
> Extension name does not matter so much when saving, I mean you can easily rename, but how the BIOS is created and it’s contents are what matters.
> Your mod BIOS is in a capsule, but named rom/bin, renaming should have been OK but it’s probably not accepted due to how you did the mod with capsule included while editing.

Tokobotenkai, 2022-11-03:
I think I managed to get it working?
```

sudo dmesg | grep BAR
[    0.176467] pci 0000:03:00.0: BAR 2: assigned to efifb
[    7.442884] i915 0000:03:00.0: BAR 0: releasing [mem 0xde000000-0xdeffffff 64bit]
[    7.442887] i915 0000:03:00.0: BAR 2: releasing [mem 0xc0000000-0xcfffffff 64bit pref]
[    7.442907] pci 0000:02:01.0: BAR 15: releasing [mem 0xc0000000-0xcfffffff 64bit pref]
[    7.442909] pcieport 0000:01:00.0: BAR 15: releasing [mem 0xc0000000-0xcfffffff 64bit pref]
[    7.442916] pcieport 0000:01:00.0: BAR 15: no space for [mem size 0x200000000 64bit pref]
[    7.442918] pcieport 0000:01:00.0: BAR 15: failed to assign [mem size 0x200000000 64bit pref]
[    7.442920] pci 0000:02:01.0: BAR 15: no space for [mem size 0x200000000 64bit pref]
[    7.442921] pci 0000:02:01.0: BAR 15: failed to assign [mem size 0x200000000 64bit pref]
[    7.442923] i915 0000:03:00.0: BAR 2: no space for [mem size 0x200000000 64bit pref]
[    7.442924] i915 0000:03:00.0: BAR 2: failed to assign [mem size 0x200000000 64bit pref]
[    7.442926] i915 0000:03:00.0: BAR 0: assigned [mem 0xde000000-0xdeffffff 64bit]
[    7.442989] i915 0000:03:00.0: [drm] Failed to resize BAR2 to 8192M (-ENOSPC)
[    7.442991] i915 0000:03:00.0: BAR 2: assigned [mem 0xc0000000-0xcfffffff 64bit pref]
[    7.448375] i915 0000:03:00.0: [drm] Cannot use the full memory 0x000000017c800000 on the device as LMEM BAR size was found to be smaller
```

Does this mean it is working?

xCuri0, 2022-11-03:
@Tokobotenkai no the BIOS should do the BAR resizing itself which is clearly hasn't because the GPU driver tries to resize it. Here it looks like Linux is trying to do BAR resizing and failing because 4G Decoding is disabled ?

Tokobotenkai, 2022-11-03:
I just checked that I have enabled it in the BIOS.

Not too sure how do I upload it but here is the file: https://drive.google.com/file/d/1-5zQmXm8xSRryHiWRTa9qT10BrTAevAc/view?usp=sharing

xCuri0, 2022-11-03:
@Tokobotenkai have you enable the ReBarUEFI driver using ReBarState ?

Tokobotenkai, 2022-11-03:
I set it to like 13 (which is 8GB) as lspci -vvv showed that the BAR size is up to 8GB, the card has 6GB RAM only though

xCuri0, 2022-11-03:
that's normal BAR size can only be power of 2. Does it show the current BAR size is 8GB though ? send the whole lspci

Tokobotenkai, 2022-11-03:
```
lspci -v |grep -A8 VGA
03:00.0 VGA compatible controller: Intel Corporation Device 56a5 (rev 05) (prog-if 00 [VGA controller])
 Subsystem: ASRock Incorporation Device 6004
 Flags: bus master, fast devsel, latency 0, IRQ 44
 Memory at de000000 (64-bit, non-prefetchable) [size=16M]
 Memory at c0000000 (64-bit, prefetchable) [size=256M]
 Expansion ROM at df000000 [disabled] [size=2M]
 Capabilities: <access denied>
 Kernel driver in use: i915
 Kernel modules: i915
```

```
sudo lspci -vvv
03:00.0 VGA compatible controller: Intel Corporation Device 56a5 (rev 05) (prog-if 00 [VGA controller])
 Subsystem: ASRock Incorporation Device 6004
 Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
 Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
 Latency: 0, Cache Line Size: 64 bytes
 Interrupt: pin ? routed to IRQ 44
 Region 0: Memory at de000000 (64-bit, non-prefetchable) [size=16M]
 Region 2: Memory at c0000000 (64-bit, prefetchable) [size=256M]
 Expansion ROM at df000000 [disabled] [size=2M]
 Capabilities: [40] Vendor Specific Information: Len=0c <?>
 Capabilities: [70] Express (v2) Endpoint, MSI 00
  DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
   ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0.000W
  DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
   RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
   MaxPayload 128 bytes, MaxReadReq 128 bytes
  DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq+ AuxPwr- TransPend-
  LnkCap: Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <64ns, L1 <1us
   ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
  LnkCtl: ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
   ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
  LnkSta: Speed 2.5GT/s (ok), Width x1 (ok)
   TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
  DevCap2: Completion Timeout: Range B, TimeoutDis+ NROPrPrP- LTR+
    10BitTagComp+ 10BitTagReq+ OBFF Not Supported, ExtFmt+ EETLPPrefix-
    EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
    FRS- TPHComp- ExtTPHComp-
    AtomicOpsCap: 32bit- 64bit- 128bitCAS-
  DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ OBFF Disabled,
    AtomicOpsCtl: ReqEn-
  LnkCap2: Supported Link Speeds: 2.5GT/s, Crosslink- Retimer- 2Retimers- DRS-
  LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
    Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
    Compliance De-emphasis: -6dB
  LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
    EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
    Retimer- 2Retimers- CrosslinkRes: unsupported
 Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit+
  Address: 00000000fee02004  Data: 0028
  Masking: 00000000  Pending: 00000000
 Capabilities: [d0] Power Management version 3
  Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
  Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
 Capabilities: [100 v1] Alternative Routing-ID Interpretation (ARI)
  ARICap: MFVC- ACS-, Next Function: 0
  ARICtl: MFVC- ACS-, Function Group: 0
 Capabilities: [420 v1] Physical Resizable BAR
  BAR 2: current size: 256MB, supported: 256MB 512MB 1GB 2GB 4GB 8GB
 Capabilities: [400 v1] Latency Tolerance Reporting
  Max snoop latency: 71680ns
  Max no snoop latency: 71680ns
 Kernel driver in use: i915
 Kernel modules: i915
```

xCuri0, 2022-11-03:
13 is correct. The driver isn't working it's probably because it was built on Linux. I think that some UEFI don't like Linux built modules because they have a different IMAGEBASE (although they do work fine with Ovmf QEMU). Either that or the BIOS wasn't flashed correctly. You could try either building on Windows or using the FFS from releases

Tokobotenkai, 2022-11-03:
I noticed that the compiled FFS file is smaller than the precompiled one you have on the releases?

Would using that file be a better shot?

Tokobotenkai, 2022-11-03:
Now it doesn't boot, will swap out the GPU to the previous dGPU to see if it is an ARC Issue as ARC doesn't boot if reBAR is disabled.

This is the patched BIOS: https://drive.google.com/file/d/1mDD7r4XK-Y06pMB4TSTnIEewcTtfV8Te/view?usp=sharing

xCuri0, 2022-11-04:
Try using smaller like sizes 9 (512mb) first in case the BIOS doesn't support large ones. Also sending your whole dmesg can be helpful

Tokobotenkai, 2022-11-04:
Okay, it boots with the Nvidia GPU. I guess installing the Arc Intel drivers screws the iGPU too as from what I am reading they are using the same driver. 

I set the size to 9 and the Intel A380 boots.

Some outputs shows that it is not working:
```
lspci -v |grep -A8 VGA
03:00.0 VGA compatible controller: Intel Corporation Device 56a5 (rev 05) (prog-if 00 [VGA controller])
 Subsystem: ASRock Incorporation Device 6004
 Flags: bus master, fast devsel, latency 0, IRQ 44
 Memory at de000000 (64-bit, non-prefetchable) [size=16M]
 Memory at c0000000 (64-bit, prefetchable) [size=256M]
 Expansion ROM at df000000 [disabled] [size=2M]
 Capabilities: <access denied>
 Kernel driver in use: i915
 Kernel modules: i915
```

```
sudo dmesg | grep BAR
[    0.176505] pci 0000:03:00.0: BAR 2: assigned to efifb
[    6.715885] i915 0000:03:00.0: BAR 0: releasing [mem 0xde000000-0xdeffffff 64bit]
[    6.715888] i915 0000:03:00.0: BAR 2: releasing [mem 0xc0000000-0xcfffffff 64bit pref]
[    6.715907] pci 0000:02:01.0: BAR 15: releasing [mem 0xc0000000-0xcfffffff 64bit pref]
[    6.715909] pcieport 0000:01:00.0: BAR 15: releasing [mem 0xc0000000-0xcfffffff 64bit pref]
[    6.715915] pcieport 0000:01:00.0: BAR 15: no space for [mem size 0x200000000 64bit pref]
[    6.715917] pcieport 0000:01:00.0: BAR 15: failed to assign [mem size 0x200000000 64bit pref]
[    6.715919] pci 0000:02:01.0: BAR 15: no space for [mem size 0x200000000 64bit pref]
[    6.715920] pci 0000:02:01.0: BAR 15: failed to assign [mem size 0x200000000 64bit pref]
[    6.715921] i915 0000:03:00.0: BAR 2: no space for [mem size 0x200000000 64bit pref]
[    6.715922] i915 0000:03:00.0: BAR 2: failed to assign [mem size 0x200000000 64bit pref]
[    6.715924] i915 0000:03:00.0: BAR 0: assigned [mem 0xde000000-0xdeffffff 64bit]
[    6.715988] i915 0000:03:00.0: [drm] Failed to resize BAR2 to 8192M (-ENOSPC)
[    6.715990] i915 0000:03:00.0: BAR 2: assigned [mem 0xc0000000-0xcfffffff 64bit pref]
[    6.723990] i915 0000:03:00.0: [drm] Cannot use the full memory 0x000000017c800000 on the device as LMEM BAR size was found to be smaller
```
I am starting to wonder if it is an issue with the Intel A380 driver. I have no Radeon cards nor any ReBar enabled card to test with in my home.

Here is the dmesg file: https://drive.google.com/file/d/1DTg-9egiF0bvgeO5k3qHpqgJKCcMnv2d/view?usp=sharing

xCuri0, 2022-11-04:
@Tokobotenkai looks like 4G decode isn't enabled or CSM is on and also the ReBarDxe driver isn't working.  ```kernel: pci 0000:03:00.0: reg 0x18: [mem 0xc0000000-0xcfffffff 64bit pref]``` shows a 256MB BAR during boot before Intel driver loads. If 4G decoding was working you won't need the ReBarDxe driver to work though if you're only using Linux.

Tokobotenkai, 2022-11-04:
At least in the BIOS, 4G Decode is enabled and CSM is disabled.

So I guess I did something wrong with the DXE BIOS modification?

xCuri0, 2022-11-04:
yeah probably something wrong. do other sizes like 10, 11 or 12 behave differently ? also when you re run ReBarState does it show the previous value you put ?

Tokobotenkai, 2022-11-04:
The 2nd option always shows as zero for some reason.

Enter ReBarState Value
 0: Disabled 
>0: Maximum BAR size set to 2^x MB 
 32: Unlimited BAR size

It always shows as >0 no matter what value I set.

I just ran CMake and then Make in the ReBarState folder, is that correct?

xCuri0, 2022-11-04:
does it not show "Current ReBarState" ?
[image]
here is mine on windows but linux should be the same

Tokobotenkai, 2022-11-04:
Okay a few findings.

The FFS file compiled by Windows and added into the BIOS and flashed doesn't work with Ubuntu for some reason, I flashed it multiple times.

The reason was because Ubuntu was showing llvmpipe instead of the Intel A380.

So I flashed the one I compiled and the A380 appeared as the display.

The ReBarState on Linux doesn't show Current State and can't change it too, could I have compiled it wrongly?

```
sudo ./ReBarState

Verify that 4G Decoding is enabled otherwise system will not POST with GPU. There is also a possibility of BIOS not supporting large BARs even with 4G decoding enabled.

It is recommended to first try smaller sizes above 256MB in case BIOS doesn't support large BARs.

Enter ReBarState Value
 0: Disabled 
>0: Maximum BAR size set to 2^x MB 
 32: Unlimited BAR size

9
Writing value of 9 / 512 MB to ReBarState 

Failed to remove old variable
Failed to write ReBarState UEFI variable
```
I am considering to flash the original BIOS and see if it is possible to boot (without 4G Decoding, the A380 won't even post).

xCuri0, 2022-11-04:
I pushed an update that prints more information in ReBarState. do you have efivars mounted at ```/sys/firmware/efi/efivars/``` because it looks like you don't.

Tokobotenkai, 2022-11-04:
Here are the outputs:
```
sudo mount -t efivarfs efivarfs /sys/firmware/efi/efivars
mount: /sys/firmware/efi/efivars: efivarfs already mounted on /sys/firmware/efi/efivars.
```

First time
```
sudo ./ReBarState
ReBarState (c) 2022 xCuri0

ReBarState variable doesn't exist / Disabled

Verify that 4G Decoding is enabled otherwise system will not POST with GPU. There is also a possibility of BIOS not supporting large BARs even with 4G decoding enabled.

It is recommended to first try smaller sizes above 256MB in case BIOS doesn't support large BARs.

Enter ReBarState Value
       0: Disabled 
Above  0: Maximum BAR size set to 2^x MB 
Above 19: Unlimited BAR size

9
Writing value of 9 / 512 MB to ReBarState 

Successfully wrote ReBarState UEFI variable
```

Second time
```
sudo ./ReBarState
ReBarState (c) 2022 xCuri0

ReBarState variable doesn't exist / Disabled

Verify that 4G Decoding is enabled otherwise system will not POST with GPU. There is also a possibility of BIOS not supporting large BARs even with 4G decoding enabled.

It is recommended to first try smaller sizes above 256MB in case BIOS doesn't support large BARs.

Enter ReBarState Value
       0: Disabled 
Above  0: Maximum BAR size set to 2^x MB 
Above 19: Unlimited BAR size

9
Writing value of 9 / 512 MB to ReBarState 

Failed to remove old variable
Failed to write ReBarState UEFI variable
```

xCuri0, 2022-11-04:
@Tokobotenkai does dmesg show anything useful after you get those error messages ?

Tokobotenkai, 2022-11-05:
Sorry for not replying, let me get home (I am out for the weekend with no access to my PC) and let me try doing it.

xCuri0, 2022-11-05:
Problem is to do with BIOS nothing else. A user on win-raid figured out how to get it working following a guide meant for hackintosh [https://winraid.level1techs.com/t/experimental-resizable-bar-dxe-driver/89534/37?u=kuri0](https://winraid.level1techs.com/t/experimental-resizable-bar-dxe-driver/89534/37?u=kuri0)

Tokobotenkai, 2022-11-05:
Wow, that's actually cool.

I guess I am probably stuck then.

I am looking at some mining threads because that's where they try to enable 4G Decoding for their GPUs to see if there's any workaround.

Thanks so much for the help so far!

val3nt33n, 2022-11-06:
Have you tried this?
* download Bios Version 1204 from https://www.asus.com/supportonly/z97pro/helpdesk_bios/
* using UEFITool extract (as is) the NvramSmi module and save it to a file
* using UEFITool replace (as is) the NvramSmi in the bios that you are modding
* add the ReBarDxe driver as per @xCuri0's instructions
* using UEFITool extract (body) the AMI Aptio capsule. The result will be saved in a .rom file(this is what you flash)
* Create a bootable FreeDos(using a tool like Rufus) USB drive formatted FAT32 and place the .rom file and AFUDOS on the USB drive together with a .bat file that contains "AFUDOS Z97PRO.rom /GAN" without quotes (see https://winraid.level1techs.com/t/guide-how-to-flash-a-modded-bios-on-a-asus-z97-motherboard/30858)
* boot from the USB drive (you need to enable CSM to boot into FreeDos)
* execute the .bat file and reboot when it's done
* enable above 4G decoding and disable CSM again from BIOS, get into the OS, run ReBarState and set value 9 or 10. Then reboot.

Tokobotenkai, 2022-11-06:
I could try it.

For the body, you can actually open the ROM in a hex editor and paste into the unmodified BIOS file after a specific offset (800 for mine) to reinsert the body ROM into the capsule again.

Put the file into a FAT32 flashdrive and name the file accordingly and use BIOS flashback feature on the mainboard to load it in. 

See this: https://winraid.level1techs.com/t/howto-get-full-nvme-support-for-all-systems-with-an-ami-uefi-bios/30901/5903?page=296

Tokobotenkai, 2022-11-06:
Okay, seems like it works?

Rebarstate now shows the previous state, I can change the value too
```
sudo ./ReBarState
ReBarState (c) 2022 xCuri0

Current ReBarState 9 / 512 MB 

Verify that 4G Decoding is enabled otherwise system will not POST with GPU. There is also a possibility of BIOS not supporting large BARs even with 4G decoding enabled.

It is recommended to first try smaller sizes above 256MB in case BIOS doesn't support large BARs.

Enter ReBarState Value
       0: Disabled 
Above  0: Maximum BAR size set to 2^x MB 
Above 19: Unlimited BAR size

10
Writing value of 10 / 1024 MB to ReBarState 

Successfully wrote ReBarState UEFI variable
```
```
lspci -v |grep -A8 VGA
03:00.0 VGA compatible controller: Intel Corporation Device 56a5 (rev 05) (prog-if 00 [VGA controller])
 Subsystem: ASRock Incorporation Device 6004
 Flags: bus master, fast devsel, latency 0, IRQ 44
 Memory at de000000 (64-bit, non-prefetchable) [size=16M]
 Memory at 80000000 (64-bit, prefetchable) [size=1G]
 Expansion ROM at df000000 [disabled] [size=2M]
 Capabilities: <access denied>
 Kernel driver in use: i915
 Kernel modules: i915
```
```
sudo dmesg | grep BAR
[    0.180927] pci 0000:03:00.0: BAR 2: assigned to efifb
[    7.769746] i915 0000:03:00.0: BAR 0: releasing [mem 0xde000000-0xdeffffff 64bit]
[    7.769750] i915 0000:03:00.0: BAR 2: releasing [mem 0x80000000-0xbfffffff 64bit pref]
[    7.769770] pci 0000:02:01.0: BAR 15: releasing [mem 0x80000000-0xbfffffff 64bit pref]
[    7.769771] pcieport 0000:01:00.0: BAR 15: releasing [mem 0x80000000-0xbfffffff 64bit pref]
[    7.769778] pcieport 0000:01:00.0: BAR 15: no space for [mem size 0x200000000 64bit pref]
[    7.769779] pcieport 0000:01:00.0: BAR 15: failed to assign [mem size 0x200000000 64bit pref]
[    7.769782] pci 0000:02:01.0: BAR 15: no space for [mem size 0x200000000 64bit pref]
[    7.769783] pci 0000:02:01.0: BAR 15: failed to assign [mem size 0x200000000 64bit pref]
[    7.769784] i915 0000:03:00.0: BAR 2: no space for [mem size 0x200000000 64bit pref]
[    7.769785] i915 0000:03:00.0: BAR 2: failed to assign [mem size 0x200000000 64bit pref]
[    7.769787] i915 0000:03:00.0: BAR 0: assigned [mem 0xde000000-0xdeffffff 64bit]
[    7.769849] i915 0000:03:00.0: [drm] Failed to resize BAR2 to 8192M (-ENOSPC)
[    7.769852] i915 0000:03:00.0: BAR 2: assigned [mem 0x80000000-0xbfffffff 64bit pref]
[    7.771560] i915 0000:03:00.0: [drm] Cannot use the full memory 0x000000017c800000 on the device as LMEM BAR size was found to be smaller
```
```
sudo lspci -vvv
03:00.0 VGA compatible controller: Intel Corporation Device 56a5 (rev 05) (prog-if 00 [VGA controller])
 Subsystem: ASRock Incorporation Device 6004
 Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
 Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
 Latency: 0, Cache Line Size: 64 bytes
 Interrupt: pin ? routed to IRQ 44
 Region 0: Memory at de000000 (64-bit, non-prefetchable) [size=16M]
 Region 2: Memory at 80000000 (64-bit, prefetchable) [size=1G]
 Expansion ROM at df000000 [disabled] [size=2M]
 Capabilities: [40] Vendor Specific Information: Len=0c <?>
 Capabilities: [70] Express (v2) Endpoint, MSI 00
  DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
   ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0.000W
  DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
   RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
   MaxPayload 128 bytes, MaxReadReq 128 bytes
  DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq+ AuxPwr- TransPend-
  LnkCap: Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <64ns, L1 <1us
   ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
  LnkCtl: ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
   ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
  LnkSta: Speed 2.5GT/s (ok), Width x1 (ok)
   TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
  DevCap2: Completion Timeout: Range B, TimeoutDis+ NROPrPrP- LTR+
    10BitTagComp+ 10BitTagReq+ OBFF Not Supported, ExtFmt+ EETLPPrefix-
    EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
    FRS- TPHComp- ExtTPHComp-
    AtomicOpsCap: 32bit- 64bit- 128bitCAS-
  DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ OBFF Disabled,
    AtomicOpsCtl: ReqEn-
  LnkCap2: Supported Link Speeds: 2.5GT/s, Crosslink- Retimer- 2Retimers- DRS-
  LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
    Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
    Compliance De-emphasis: -6dB
  LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
    EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
    Retimer- 2Retimers- CrosslinkRes: unsupported
 Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit+
  Address: 00000000fee02004  Data: 0027
  Masking: 00000000  Pending: 00000000
 Capabilities: [d0] Power Management version 3
  Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
  Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
 Capabilities: [100 v1] Alternative Routing-ID Interpretation (ARI)
  ARICap: MFVC- ACS-, Next Function: 0
  ARICtl: MFVC- ACS-, Function Group: 0
 Capabilities: [420 v1] Physical Resizable BAR
  BAR 2: current size: 1GB, supported: 256MB 512MB 1GB 2GB 4GB 8GB
 Capabilities: [400 v1] Latency Tolerance Reporting
  Max snoop latency: 71680ns
  Max no snoop latency: 71680ns
 Kernel driver in use: i915
 Kernel modules: i915
```

I tried resizing to 8GB but similar issues that I am unable to post, the error code is AE which is legacy boot event?

xCuri0, 2022-11-06:
@Tokobotenkai 
> apparently CSM was automatically turned on with Above 4G Decoding On
this automatically happens after boot failure or booting without gpu (which wasn't functioning with 2gb BAR)

val3nt33n, 2022-11-06:
The BIOS automatically switches CSM back to enabled if video is not working. If you have an iGPU you can set that as the primary output in your BIOS. That will keep the CSM disabled and you don’t have to switch cards when you set ReBarState too high.

Anyway, I see that you are also getting rebar working only up to 1GB. The BIOS can’t allocate a BAR bigger than 1GB. This is the same problem that I have. There is a limit enforced somewhere in the BIOS. Needs more investigation to develop a patch that can remove that limit.

Tokobotenkai, 2022-11-06:
Yup, it attempted to boot twice. 4GB and 8GB had this issue. 2GB was a white static screen and 1GB works.

xCuri0, 2022-11-06:
> Needs more investigation to develop a patch that can remove that limit.

Not a limit, seems more like it's forcing it to be allocated into the 32-bit region. The address 0x80000000 is a 32-bit one and you can't find a 2GB of larger BAR in the 32-bit region.

Tokobotenkai, 2022-11-06:
Some interesting things I read as they are interested getting the Xeon Phi to work which requires a large BAR size, supposedly a patched ASUS Gryphon Z87 works (https://www.asus.com/supportonly/gryphon_z87/helpdesk_bios/). I am guessing that it goes correctly to 64 bit for the MMIO so maybe it is something for us to look at? Supposedly an ASUS Z97-WS works with their latest BIOS too.

Someone seems to get it working with a ASUS Gryphon Z97 with ASUS supplied BIOS version 2205.
https://www.pugetsystems.com/labs/hpc/Will-your-motherboard-work-with-Intel-Xeon-Phi-490/#comment-1869578314

Not sure if these help somehow.

aquamacho1, 2022-11-06:
Well, im having exact same issue as you guys on my x79 board.

val3nt33n, 2022-11-07:
> By the way how did you determine BIOS version 1204 was fine?

From this https://www.insanelymac.com/forum/topic/317802-efi-variable-store-on-aptio-v-haswell-e-and-up/page/5/#comment-2535040
I assumed that any Asus BIOS released starting with October 2014 has this problem. So I looked for the last version before that month.

Tokobotenkai, 2022-11-07:
@val3nt33n Great catch there!

val3nt33n, 2022-11-07:
> @val3nt33n Great catch there!

The credit goes to @xCuri0 for pointing me to it.

xCuri0, 2022-11-11:
@Tokobotenkai if 16GB works on the A770 (with all the required patches) I can close this issue.

Tokobotenkai, 2022-11-12:
Yup, consider it solved. Thanks man