xCuri0/ReBarUEFI #10: Z97 board can't allocate above 4G even with above 4g decoding enable in BIOS

val3nt33n, 2022-11-05:
**System**
 - Motherboard: Asus Z97i-Plus
 - BIOS Version: 2704
 - GPU: Intel Arc A750 LE (supports up to 8GB BAR)
 - Above 4G decoding is enabled in BIOS
 - CSM is disabled
 - Applied PciBus | Remove <16GB BAR size limit patch
 - replaced NvramSmi module from an older BIOS(version 2103) to allow new uefi variable creation

**Description**
I modded the BIOS to enable rebar using your instructions but the maximum working BAR size is 1GB. At 2GB the system gets stuck during post, at 4GB and 8GB it boots into the OS(Windows 10 and Ubuntu 20.04) but video output works only from the iGPU.

This is dmesg after booting with 8GB
```
[    0.196147] pci 0000:00:02.0: reg 0x18: [mem 0xe0000000-0xefffffff 64bit pref]
[    0.199521] pci 0000:01:00.0: reg 0x10: [mem 0xf0000000-0xf07fffff 64bit pref]
[    0.199685] pci 0000:00:01.0:   bridge window [mem 0xf0000000-0xf07fffff 64bit pref]
[    0.200388] pci 0000:03:00.0: reg 0x18: [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.200403] pci 0000:03:00.0: reg 0x30: [mem 0x00000000-0x001fffff pref]
[    0.215439] pnp 00:00: disabling [mem 0xfed40000-0xfed44fff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.216329] pnp 00:06: disabling [mem 0xfed1c000-0xfed1ffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.216331] pnp 00:06: disabling [mem 0xfed10000-0xfed17fff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.216333] pnp 00:06: disabling [mem 0xfed18000-0xfed18fff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.216334] pnp 00:06: disabling [mem 0xfed19000-0xfed19fff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.216336] pnp 00:06: disabling [mem 0xf8000000-0xfbffffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.216337] pnp 00:06: disabling [mem 0xfed20000-0xfed3ffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.216338] pnp 00:06: disabling [mem 0xfed90000-0xfed93fff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.216340] pnp 00:06: disabling [mem 0xfed45000-0xfed8ffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.216341] pnp 00:06: disabling [mem 0xff000000-0xffffffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.216342] pnp 00:06: disabling [mem 0xfee00000-0xfeefffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.216344] pnp 00:06: disabling [mem 0xf7fe0000-0xf7feffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.216345] pnp 00:06: disabling [mem 0xf7ff0000-0xf7ffffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.225125] pci 0000:01:00.0: BAR 15: no space for [mem size 0x200000000 64bit pref]
[    0.225127] pci 0000:01:00.0: BAR 15: failed to assign [mem size 0x200000000 64bit pref]
[    0.225129] pci 0000:02:01.0: BAR 15: no space for [mem size 0x200000000 64bit pref]
[    0.225130] pci 0000:02:01.0: BAR 15: failed to assign [mem size 0x200000000 64bit pref]
[    0.225135] pci 0000:03:00.0: BAR 2: no space for [mem size 0x200000000 64bit pref]
[    0.225136] pci 0000:03:00.0: BAR 2: failed to assign [mem size 0x200000000 64bit pref]
[    0.225140] pci 0000:03:00.0: BAR 6: no space for [mem size 0x00200000 pref]
[    0.225141] pci 0000:03:00.0: BAR 6: failed to assign [mem size 0x00200000 pref]
[    0.225176] pci 0000:00:01.0:   bridge window [mem 0xf0000000-0xf07fffff 64bit pref]
[    0.225201] pci_bus 0000:01: resource 2 [mem 0xf0000000-0xf07fffff 64bit pref]
```
and /proc/iomem
```
f0000000-f07fffff : PCI Bus 0000:01
  f0000000-f07fffff : 0000:01:00.0
f7400000-f77fffff : 0000:00:02.0
f7800000-f7afffff : PCI Bus 0000:06
  f7800000-f79fffff : 0000:06:00.0
  f7a00000-f7a07fff : 0000:06:00.0
f7b00000-f7bfffff : PCI Bus 0000:01
  f7b00000-f7bfffff : PCI Bus 0000:02
    f7b00000-f7bfffff : PCI Bus 0000:04
      f7b00000-f7b03fff : 0000:04:00.0
...
100000000-41fdfffff : System RAM
  179e00000-17ae025c7 : Kernel code
  17b000000-17ba87fff : Kernel rodata
  17bc00000-17c0455ff : Kernel data
  17c393000-17c9fffff : Kernel bss
41fe00000-41fffffff : RAM buffer
```
[image]

val3nt33n, 2022-11-05:
I tried looking at the memory range in DSDT but it's using some external symbols that I couldn't find anywhere in the other ACPI tables.
```
        CreateQWordField (BUF0, \_SB.PCI0._Y1E._LEN, M2LN)  // _LEN: Length          
        CreateQWordField (BUF0, \_SB.PCI0._Y1E._MIN, M2MN)  // _MIN: Minimum Base Address         
        CreateQWordField (BUF0, \_SB.PCI0._Y1E._MAX, M2MX)  // _MAX: Maximum Base Address             
        M2LN = M64L /* External reference */
        M2MN = M64B /* External reference */                                         
        M2MX = ((M2MN + M2LN) - One)
```

val3nt33n, 2022-11-05:
After setting pci=realloc at boot it looks like it manages to assign memory to the GPU

```
[    0.213321] pnp 00:00: disabling [mem 0xfed40000-0xfed44fff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.213934] pnp 00:06: disabling [mem 0xfed1c000-0xfed1ffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.213937] pnp 00:06: disabling [mem 0xfed10000-0xfed17fff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.213939] pnp 00:06: disabling [mem 0xfed18000-0xfed18fff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.213940] pnp 00:06: disabling [mem 0xfed19000-0xfed19fff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.213941] pnp 00:06: disabling [mem 0xf8000000-0xfbffffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.213943] pnp 00:06: disabling [mem 0xfed20000-0xfed3ffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.213944] pnp 00:06: disabling [mem 0xfed90000-0xfed93fff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.213945] pnp 00:06: disabling [mem 0xfed45000-0xfed8ffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.213947] pnp 00:06: disabling [mem 0xff000000-0xffffffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.213948] pnp 00:06: disabling [mem 0xfee00000-0xfeefffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.213950] pnp 00:06: disabling [mem 0xf7fe0000-0xf7feffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.213951] pnp 00:06: disabling [mem 0xf7ff0000-0xf7ffffff] because it overlaps 0000:03:00.0 BAR 2 [mem 0x00000000-0x1ffffffff 64bit pref]
[    0.221845] pci_bus 0000:00: max bus depth: 3 pci_try_num: 4
[    0.221854] pci 0000:01:00.0: BAR 15: no space for [mem size 0x200000000 64bit pref]
[    0.221856] pci 0000:01:00.0: BAR 15: failed to assign [mem size 0x200000000 64bit pref]
[    0.221858] pci 0000:02:01.0: BAR 15: no space for [mem size 0x200000000 64bit pref]
[    0.221859] pci 0000:02:01.0: BAR 15: failed to assign [mem size 0x200000000 64bit pref]
[    0.221860] pci 0000:02:01.0: BAR 14: no space for [mem size 0x01800000]
[    0.221861] pci 0000:02:01.0: BAR 14: failed to assign [mem size 0x01800000]
[    0.221863] pci 0000:03:00.0: BAR 2: no space for [mem size 0x200000000 64bit pref]
[    0.221864] pci 0000:03:00.0: BAR 2: failed to assign [mem size 0x200000000 64bit pref]
[    0.221865] pci 0000:03:00.0: BAR 0: no space for [mem size 0x01000000 64bit]
[    0.221866] pci 0000:03:00.0: BAR 0: failed to assign [mem size 0x01000000 64bit]
[    0.221868] pci 0000:03:00.0: BAR 6: no space for [mem size 0x00200000 pref]
[    0.221869] pci 0000:03:00.0: BAR 6: failed to assign [mem size 0x00200000 pref]
[    0.221870] pci 0000:02:01.0: PCI bridge to [bus 03]
[    0.221879] pci 0000:02:04.0: PCI bridge to [bus 04]
[    0.221884] pci 0000:02:04.0:   bridge window [mem 0xf7c00000-0xf7cfffff]
[    0.221892] pci 0000:01:00.0: PCI bridge to [bus 02-04]
[    0.221895] pci 0000:01:00.0:   bridge window [mem 0xf7c00000-0xf7cfffff]
[    0.221900] pci 0000:00:01.0: PCI bridge to [bus 01-04]
[    0.221902] pci 0000:00:01.0:   bridge window [mem 0xf7c00000-0xf7cfffff]
[    0.221903] pci 0000:00:01.0:   bridge window [mem 0xf0000000-0xf07fffff 64bit pref]
[    0.221907] pci 0000:00:1c.0: PCI bridge to [bus 05]
[    0.221914] pci_bus 0000:00: No. 2 try to assign unassigned res
[    0.221918] pci 0000:01:00.0: BAR 15: no space for [mem size 0x200000000 64bit pref]
[    0.221919] pci 0000:01:00.0: BAR 15: failed to assign [mem size 0x200000000 64bit pref]
[    0.221921] pci 0000:02:01.0: BAR 15: no space for [mem size 0x200000000 64bit pref]
[    0.221922] pci 0000:02:01.0: BAR 15: failed to assign [mem size 0x200000000 64bit pref]
[    0.221923] pci 0000:02:01.0: BAR 14: no space for [mem size 0x02000000]
[    0.221924] pci 0000:02:01.0: BAR 14: failed to assign [mem size 0x02000000]
[    0.221926] pci 0000:03:00.0: BAR 2: no space for [mem size 0x200000000 64bit pref]
[    0.221927] pci 0000:03:00.0: BAR 2: failed to assign [mem size 0x200000000 64bit pref]
[    0.221928] pci 0000:03:00.0: BAR 0: no space for [mem size 0x01000000 64bit]
[    0.221929] pci 0000:03:00.0: BAR 0: failed to assign [mem size 0x01000000 64bit]
[    0.221930] pci 0000:02:01.0: PCI bridge to [bus 03]
[    0.221939] pci 0000:02:04.0: PCI bridge to [bus 04]
[    0.221943] pci 0000:02:04.0:   bridge window [mem 0xf7c00000-0xf7cfffff]
[    0.221951] pci 0000:01:00.0: PCI bridge to [bus 02-04]
[    0.221954] pci 0000:01:00.0:   bridge window [mem 0xf7c00000-0xf7cfffff]
[    0.221959] pci 0000:00:01.0: PCI bridge to [bus 01-04]
[    0.221961] pci 0000:00:01.0:   bridge window [mem 0xf7c00000-0xf7cfffff]
[    0.221962] pci 0000:00:01.0:   bridge window [mem 0xf0000000-0xf07fffff 64bit pref]
[    0.221965] pci 0000:00:1c.0: PCI bridge to [bus 05]
[    0.221973] pci_bus 0000:00: No. 3 try to assign unassigned res
[    0.221975] pci 0000:02:04.0: resource 14 [mem 0xf7c00000-0xf7cfffff] released
[    0.221976] pci 0000:02:04.0: PCI bridge to [bus 04]
[    0.221979] pci 0000:01:00.0: resource 14 [mem 0xf7c00000-0xf7cfffff] released
[... cut ...]
	Subsystem: Intel Corporation Device 1021
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Region 0: Memory at f0000000 (64-bit, non-prefetchable) [disabled] [size=16M]
	Region 2: Memory at 800000000 (64-bit, prefetchable) [disabled] [size=8G]
	Capabilities: [40] Vendor Specific Information: Len=0c <?>
	Capabilities: [70] Express (v2) Endpoint, MSI 00
```

val3nt33n, 2022-11-06:
Browsing through AMIBCP I found some PCIe settings that don't show up in BIOS. Not sure what these do or how to make them visible. All settings(both visible and hidden) have selected 'Yes' in the Show column. I'm confused why it says 'Yes' when it's hidden.
The Reserved Memory setting defaults to 10. I wonder if this is the 1GB limit.

[image]

xCuri0, 2022-11-06:
@val3nt33n can you also show the **Prefetchable Memory** description. I think that might be it not **Reserved Memory**

val3nt33n, 2022-11-06:
@xCuri0 it says "Prefetchable Memory Range for this Root Bridge"

xCuri0, 2022-11-06:
try changing both to 13 I guess. but i think **Prefetchable Memory** is more likely the important one

val3nt33n, 2022-11-06:
I tried changing all 3 one by one. No difference. I think some of these settings are hidden because they are not implemented.

val3nt33n, 2022-11-06:
Oddly, if I enable on-aboard audio, only this device gets assigned at the end of the Large Memory. At least this confirms that above 4G decoding is working.
[image]

xCuri0, 2022-11-06:
@val3nt33n my system allocates multiple devices in the 64-bit region (gpu not visible in screenshot because it's in large memory)
[image]

Have you tried with **Reserved Memory** = 13, **Prefetchable Memory** = 13 and **Prefetchable Memory above 4G** = Enabled, all at the same time ?

val3nt33n, 2022-11-06:
> Have you tried with **Reserved Memory** = 13, **Prefetchable Memory** = 13 and **Prefetchable Memory above 4G** = Enabled, all at the same time ?

I have. No change.

Tokobotenkai, 2022-11-06:
I have an i5-4690k and ASUS Z97-PRO but I only have Ubuntu 22.04 running on it. Any idea how can I contribute or something?

cursemex, 2022-11-07:
> Browsing through AMIBCP I found some PCIe settings that don't show up in BIOS. Not sure what these do or how to make them visible. All settings(both visible and hidden) have selected 'Yes' in the [...]
Found this while surfing on the web: https://www.bios-mods.com/forum/Thread-GUIDE-How-to-modify-AMI-MSI-BIOS-to-unlock-full-features-UEFI-Aptio4

Step 5 made those options visible in BIOS

https://i.imgur.com/YVAiniE.png

https://i.imgur.com/gFxoUPa.png

https://cdn.discordapp.com/attachments/813843868763422751/1039006024054603817/20221107_052935.jpg

https://cdn.discordapp.com/attachments/813843868763422751/1039006039032463390/20221107_052950.jpg

val3nt33n, 2022-11-07:
@cursemex yeah, that’s what I did when I tested(it’s enough to set USER on the top menu and it will make everything under it visible). But playing with different setting in these new options didn’t do anything for me.

xCuri0, 2022-11-07:
@val3nt33n in cursemex's screenshot it shows "PCIE Port 1 is assigned to LAN". The one with the GPU on is what needs to be changed.

Tokobotenkai, 2022-11-07:
I wonder if we have a working board and is there a difference in the order of assignment?

xCuri0, 2022-11-07:
@Tokobotenkai 

### Prazola on win-raid's Asrock Z97 Extreme9
[image]
It also is different for Port 5
[image]

### My Gigabyte B75M-D3H
[image]
Port 7 has "Extra Bus Reserved" set to 7
[image]

### Romulus_ut3 on win-raid's Asus P8Z77-V PRO
[image]

Like my board port 7 has "Extra Bus Reserved" set to 7
[image]

all of these don't have the "Prefetchable Memory above 4G"

val3nt33n, 2022-11-07:
> @val3nt33n in cursemex's screenshot it shows "PCIE Port 1 is assigned to LAN". The one with the GPU on is what needs to be changed.

They all say that(there is not option to change this). And I changed the settings on all ports. I also noticed that on higher ports there are more extra bus reserved and higher Reserved I/O. Playing with them didn't do anything for me.

cursemex, 2022-11-07:
Silly question, but how do I know which "root port" my gpu is connected to? HWinfo doesn't state the gpu to be on a "root port", but does  for my sound card.

xCuri0, 2022-11-07:
i think i found the problem. PciBus driver is downgrading 64-bit resources to 32-bit ones automatically on Z97 it seems.

Asus Z97(bad)
[image]

Asus Z77 (good)
[image]

aquamacho1, 2022-11-07:
@xCuri0 can this be related to my problem to ?

xCuri0, 2022-11-07:
here's the hex patch if you want to try it. i don't think it will work though because probably have to fix like 3 other places

in PciBus.efi replace ```83 3E 06 75 06 C7 06 04 00 00 00 83 3E 07 75 06 C7 06 05 00 00 00``` with ```66 90 66 90 66 90 66 90 66 90 66 90 66 90 66 90 66 90 66 90 66 90```

val3nt33n, 2022-11-07:
I was getting ready to re-program the chip in case it doesn't boot anymore and to my surprise I found a tiny chip that I don't think I can program(see macro photo attached). My programmer is for a generic(much bigger chip) with fewer pins. Also I don't have USB flashback. The manual suggests that I can upload a new BIOS from a USB drive but inside the BIOS UI. Looks like I was adventuring in all this without any backup.

[image]

val3nt33n, 2022-11-07:
@Tokobotenkai can you try to patch first since your board has dual BIOS
@cursemex I thin you can try it too.

@xCuri0 does the same code exist on Asus Z97-Pro?

xCuri0, 2022-11-07:
@val3nt33n yes it does (i tried uefipatch and it found and patched). There is the pad file problem though but I've written in README how to fix it.

val3nt33n, 2022-11-07:
@xCuri0 BTW is there any reason why you are suggesting to use MMTool to do the replacement instead of UEFITool?
I tried both and in the final file they have different size for the PciBus module. Not sure which one is correct. I have UEFITool_0.28.0_win32 and MMTool 4.50.0.23

xCuri0, 2022-11-07:
MMTool doesn't mess up the pad file like UEFITool does. But MMTool can only replace the FFS unlike UEFITool which lets you replace the PE32 executable inside the FFS. Which is why in the README is says to create your modified FFS with UEFITool.

val3nt33n, 2022-11-07:
I see. When I replaced the NvramSmi module I used UEFITool because MMTool did not show the string for NvramSmi (it was empty at the line where it should have been.

Tokobotenkai, 2022-11-07:
I am going to work soon so probably get back to you guys in about 15-16 hours

xCuri0, 2022-11-07:
@Tokobotenkai Use UEFIPatch to do it as it's the easiest way you just have to put the UEFIPatch code in patches.txt and run UEFIPatch on the BIOS file in command line. You still do have to do the workaround with MMTool to avoid getting no boot because of pad file problem though (it's in README).

Tokobotenkai, 2022-11-07:
@xCuri0 Just to check, I extracted the body from the CAP, and applied the patch using UEFITool, where do I get the MMTool? Is there a linux version?

val3nt33n, 2022-11-08:
* run UEFIPatch Z97PRO.cap patches.txt -o Z97PRO-patched.cap
* use UEFITool to extract(as is) the PciBus module from the patched file and save the result in a .ffs file
* use MMTool (got to replace tab) and load Z97PRO.cap In the Browse section load the .ffs file
* got to PciBus line, click on it and then click on Replace, then save the result to a new .cap file(this is now the patched bios that you can use as baseline)
* follow @xCuri0's steps to insert the ReBarDxe driver in the new .cap file
* do anything else you need to do(i.e. replace NvramSmi if you need to, etc.)

I got MMTool 4.50.0023 from https://www.tweaktownforum.com/forum/tech-support-from-vendors/gigabyte/30823-latest-overclocking-programs-system-info-benchmarking-stability-tools

cursemex, 2022-11-08:
> 

Thanks for this. I was tryina figure it. I'll do the steps and give you an update

val3nt33n, 2022-11-08:
OK, so after you flashed, the system booted normally? You saw problems only once you experimented with larger bar sizes?

You can try to clear CMOS if a large bar doesn't boot(or you can set iGPU as default right after you flash, if you have one), instead of recovering the BIOS.

cursemex, 2022-11-08:
> 

No problems going up to 8GB from my observation aside from 1 long beep after post.
I'll redo the testing to confirm it.

As for clearing CMOS, I've noticed that it doesn't like my dedicated gpu on. It'll keep beeping and rebooting until I remove the gpu, then it'll switch to iGPU. I haven't tried it this time around, though.

val3nt33n, 2022-11-08:
Can you show device manager (View->Resources by Type) and expand Large Memory with 8GB?

Your GPU has 12GB of VRAM, no? In that case the max BAR you would need is 16GB.
To get 16GB working you probably also need this patch

```
# PciBus | Remove <16GB BAR size limit (Ivy Bridge/Haswell)
3C1DE39F-D207-408A-AACC-731CFB7F1DD7 10 P:B8FFFFFFFF030000004C3B:B8FFFFFFFFFFFFFF004C3B 
```

cursemex, 2022-11-08:
> Can you show device manager (View->Resources by Type) and expand Large Memory with 8GB?
>
> Your GPU has 12GB of VRAM, no? In that case the max BAR you would need is 16GB. To get 16GB working you [...]

I was thinking about applying it too, but didn't. Let me restart the process and remake the roms and test it again.
And yes, my gpu is 6700XT 12GB.

Tokobotenkai, 2022-11-08:
I did another way:
1. I patched the modified Body ROM file
2. Using a hex editor, I pasted the entire patched ROM file into the unmodified Capsule file after the offset
3. Flashed using USB Flashback

For some reason, my USB wireless keyboard is not detected on boot, but if I unplug and replug the dongle it works, else once Ubuntu boots, it is automatically detected.

For now ReBar up to 8GB work, I haven't tried 16GB as the A380 only has 6GB of RAM. I am waiting for a new power supply to connect the A770, or can I just up the limit to 16GB (the A380 from lspci only supports up to 8GB ReBar)?

```
sudo lspci -vvv
03:00.0 VGA compatible controller: Intel Corporation Device 56a5 (rev 05) (prog-if 00 [VGA controller])
 Subsystem: ASRock Incorporation Device 6004
 Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
 Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
 Latency: 0, Cache Line Size: 64 bytes
 Interrupt: pin ? routed to IRQ 44
 Region 0: Memory at de000000 (64-bit, non-prefetchable) [size=16M]
 Region 2: Memory at c00000000 (64-bit, prefetchable) [size=8G]
 Expansion ROM at df000000 [disabled] [size=2M]
 Capabilities: [40] Vendor Specific Information: Len=0c <?>
 Capabilities: [70] Express (v2) Endpoint, MSI 00
  DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
   ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0.000W
  DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
   RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
   MaxPayload 128 bytes, MaxReadReq 128 bytes
  DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq+ AuxPwr- TransPend-
  LnkCap: Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <64ns, L1 <1us
   ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
  LnkCtl: ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
   ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
  LnkSta: Speed 2.5GT/s (ok), Width x1 (ok)
   TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
  DevCap2: Completion Timeout: Range B, TimeoutDis+ NROPrPrP- LTR+
    10BitTagComp+ 10BitTagReq+ OBFF Not Supported, ExtFmt+ EETLPPrefix-
    EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
    FRS- TPHComp- ExtTPHComp-
    AtomicOpsCap: 32bit- 64bit- 128bitCAS-
  DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ OBFF Disabled,
    AtomicOpsCtl: ReqEn-
  LnkCap2: Supported Link Speeds: 2.5GT/s, Crosslink- Retimer- 2Retimers- DRS-
  LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
    Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
    Compliance De-emphasis: -6dB
  LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
    EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
    Retimer- 2Retimers- CrosslinkRes: unsupported
 Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit+
  Address: 00000000fee02004  Data: 0027
  Masking: 00000000  Pending: 00000000
 Capabilities: [d0] Power Management version 3
  Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
  Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
 Capabilities: [100 v1] Alternative Routing-ID Interpretation (ARI)
  ARICap: MFVC- ACS-, Next Function: 0
  ARICtl: MFVC- ACS-, Function Group: 0
 Capabilities: [420 v1] Physical Resizable BAR
  BAR 2: current size: 8GB, supported: 256MB 512MB 1GB 2GB 4GB 8GB
 Capabilities: [400 v1] Latency Tolerance Reporting
  Max snoop latency: 71680ns
  Max no snoop latency: 71680ns
 Kernel driver in use: i915
 Kernel modules: i915
```
```
lspci -v |grep -A8 VGA
03:00.0 VGA compatible controller: Intel Corporation Device 56a5 (rev 05) (prog-if 00 [VGA controller])
 Subsystem: ASRock Incorporation Device 6004
 Flags: bus master, fast devsel, latency 0, IRQ 44
 Memory at de000000 (64-bit, non-prefetchable) [size=16M]
 Memory at c00000000 (64-bit, prefetchable) [size=8G]
 Expansion ROM at df000000 [disabled] [size=2M]
 Capabilities: <access denied>
 Kernel driver in use: i915
 Kernel modules: i915
```
```
sudo dmesg | grep BAR
[    0.176035] pci 0000:03:00.0: BAR 2: assigned to efifb
```

xCuri0, 2022-11-08:
@cursemex 16GB doesn't work because the BIOS only uses 0xC00000000 (48GB) to 0x1000000000 (64GB) for resource allocation (see Large Memory in Device Manager). It's easy to patch though (did it for the Z97 user on win-raid) so I can make a fix for it but DSDT needs to be edited.

cursemex, 2022-11-08:
> 

If you do make them, I'll happily test it out. I may be slow in doing so in the upcoming days due to school though.

Tokobotenkai, 2022-11-08:
From Hardinfo in Ubuntu,
```
-Memory-
<tt>00000000-00000fff </tt>		: Reserved
<tt>00001000-00057fff </tt>		: System RAM
<tt>00058000-00058fff </tt>		: Reserved
<tt>00059000-0008ffff </tt>		: System RAM
<tt>00090000-00090fff </tt>		: Reserved
<tt>00091000-0009dfff </tt>		: System RAM
<tt>0009e000-0009ffff </tt>		: Reserved
<tt>00000000-00000000 </tt>		: PCI Bus 0000:00
<tt>00000000-00000000 </tt>		: PCI Bus 0000:00
<tt>00000000-00000000 </tt>		: PCI Bus 0000:00
<tt>00000000-00000000 </tt>		: PCI Bus 0000:00
<tt>00000000-00000000 </tt>		: PCI Bus 0000:00
<tt>00000000-00000000 </tt>		: PCI Bus 0000:00
<tt>00000000-00000000 </tt>		: PCI Bus 0000:00
<tt>00000000-00000000 </tt>		: PCI Bus 0000:00
<tt>000a0000-000dffff </tt>		: PCI Bus 0000:00
<tt>000f0000-000fffff </tt>		: System ROM
<tt>00100000-bbbac017 </tt>		: System RAM
<tt>bbbac018-bbc6e057 </tt>		: System RAM
<tt>bbc6e058-bbe17fff </tt>		: System RAM
<tt>bbe18000-bbe1efff </tt>		: ACPI Non-volatile Storage
<tt>bbe1f000-bc27efff </tt>		: System RAM
<tt>bc27f000-bc6f3fff </tt>		: Reserved
<tt>bc6f4000-cd165fff </tt>		: System RAM
<tt>cd166000-cdc93fff </tt>		: Reserved
<tt>cdc94000-cdcb1fff </tt>		: ACPI Tables
<tt>cdcb2000-ce1d5fff </tt>		: ACPI Non-volatile Storage
<tt>ce1d6000-ceffefff </tt>		: Reserved
<tt>cefff000-ceffffff </tt>		: System RAM
<tt>cf000000-cfffffff </tt>		: RAM buffer
<tt>d0000000-feafffff </tt>		: PCI Bus 0000:00
<tt>  de000000-df3fffff </tt>		: PCI Bus 0000:01
<tt>    de000000-df3fffff </tt>		: PCI Bus 0000:02
<tt>      de000000-df1fffff </tt>		: PCI Bus 0000:03
<tt>        de000000-deffffff </tt>		: <b><small>PCI</small></b>  Intel Corporation Device 56a5 (rev 05) (prog-if 00 [VGA controller])
<tt>          de102040-de1020bf </tt>		: i915.spi.768
<tt>          de373000-de373ffb </tt>		: i915.mei-gsc.768
<tt>          de374000-de374ffb </tt>		: i915.mei-gscfi.768
<tt>        df000000-df1fffff </tt>		: <b><small>PCI</small></b>  Intel Corporation Device 56a5 (rev 05) (prog-if 00 [VGA controller])
<tt>      df300000-df3fffff </tt>		: PCI Bus 0000:04
<tt>        df300000-df303fff </tt>		: <b><small>PCI</small></b>  Intel Corporation Device 4f92
<tt>          df300000-df303fff </tt>		: ICH HD audio
<tt>  df400000-df4fffff </tt>		: PCI Bus 0000:0c
<tt>    df400000-df407fff </tt>		: <b><small>PCI</small></b>  ASMedia Technology Inc. ASM1042A USB 3.0 Host Controller (prog-if 30 [XHCI])
<tt>      df400000-df407fff </tt>		: xhci-hcd
<tt>  df500000-df5fffff </tt>		: PCI Bus 0000:06
<tt>    df500000-df5fffff </tt>		: PCI Bus 0000:07
<tt>      df500000-df5fffff </tt>		: PCI Bus 0000:0b
<tt>        df500000-df5001ff </tt>		: <b><small>PCI</small></b>  ASMedia Technology Inc. ASM1061 SATA IDE Controller (rev 02) (prog-if 85 [PCI native mode-only controller, supports bus mastering])
<tt>          df500000-df5001ff </tt>		: <b><small>Module</small></b> AHCI SATA low-level driver
<tt>  df600000-df61ffff </tt>		: <b><small>PCI</small></b>  Intel Corporation Ethernet Connection (2) I218-V
<tt>    df600000-df61ffff </tt>		: <b><small>Module</small></b> Intel(R) PRO/1000 Network Driver
<tt>  df620000-df6207ff </tt>		: <b><small>PCI</small></b>  Intel Corporation 9 Series Chipset Family SATA Controller [AHCI Mode] (prog-if 01 [AHCI 1.0])
<tt>    df620000-df6207ff </tt>		: <b><small>Module</small></b> AHCI SATA low-level driver
<tt>  df621000-df6213ff </tt>		: <b><small>PCI</small></b>  Intel Corporation 9 Series Chipset Family USB EHCI Controller #1 (prog-if 20 [EHCI])
<tt>    df621000-df6213ff </tt>		: ehci_hcd
<tt>  df622000-df6223ff </tt>		: <b><small>PCI</small></b>  Intel Corporation 9 Series Chipset Family USB EHCI Controller #2 (prog-if 20 [EHCI])
<tt>    df622000-df6223ff </tt>		: ehci_hcd
<tt>  df623000-df623fff </tt>		: <b><small>PCI</small></b>  Intel Corporation Ethernet Connection (2) I218-V
<tt>    df623000-df623fff </tt>		: <b><small>Module</small></b> Intel(R) PRO/1000 Network Driver
<tt>  dffe0000-dffeffff </tt>		: pnp 00:06
<tt>  dfff0000-dfffffff </tt>		: pnp 00:06
<tt>  e0000000-efffffff </tt>		: PCI MMCONFIG 0000 [bus 00-ff]
<tt>    e0000000-efffffff </tt>		: Reserved
<tt>      e0000000-efffffff </tt>		: pnp 00:06
<tt>fec00000-fec00fff </tt>		: Reserved
<tt>  fec00000-fec003ff </tt>		: IOAPIC 0
[... 18 lines cut ...]
<tt>100000000-42effffff </tt>		: System RAM
<tt>  27ec00000-27fc020bf </tt>		: Kernel code
<tt>  27fe00000-2808bafff </tt>		: Kernel rodata
<tt>  280a00000-280e5dc7f </tt>		: Kernel data
<tt>  28113c000-2815fffff </tt>		: Kernel bss
<tt>42f000000-42fffffff </tt>		: RAM buffer
<tt>800000000-fbfffffff </tt>		: PCI Bus 0000:00
<tt>  c00000000-e007fffff </tt>		: PCI Bus 0000:01
<tt>    c00000000-dffffffff </tt>		: PCI Bus 0000:02
<tt>      c00000000-dffffffff </tt>		: PCI Bus 0000:03
<tt>        c00000000-dffffffff </tt>		: <b><small>PCI</small></b>  Intel Corporation Device 56a5 (rev 05) (prog-if 00 [VGA controller])
<tt>    e00000000-e007fffff </tt>		: <b><small>PCI</small></b>  Intel Corporation Device 4fa1 (rev 01) (prog-if 00 [Normal decode])
<tt>  fbff00000-fbff0ffff </tt>		: <b><small>PCI</small></b>  Intel Corporation 9 Series Chipset Family USB xHCI Controller (prog-if 30 [XHCI])
<tt>    fbff00000-fbff0ffff </tt>		: xhci-hcd
<tt>  fbff10000-fbff13fff </tt>		: <b><small>PCI</small></b>  Intel Corporation 9 Series Chipset Family HD Audio Controller
<tt>    fbff10000-fbff13fff </tt>		: ICH HD audio
<tt>  fbff14000-fbff140ff </tt>		: <b><small>PCI</small></b>  Intel Corporation 9 Series Chipset Family SMBus Controller
<tt>  fbff16000-fbff1600f </tt>		: <b><small>PCI</small></b>  Intel Corporation 9 Series Chipset Family ME Interface #1
<tt>    fbff16000-fbff1600f </tt>		: <b><small>Module</small></b> Intel(R) Management Engine Interface
-DMA-
<tt> 4</tt>		: cascade
```
c00000000 (48GB) to dffffffff (56GB)

So without editing the DSDT, only 8GB is the maximum, am I interpreting correctly?

xCuri0, 2022-11-08:
@Tokobotenkai it's 800000000 (32GB) to fbfffffff (63GB) (PCI Bus 0000:00) which should be good for upto 16GB BAR.

Tokobotenkai, 2022-11-08:
@xCuri0 Okay, will try it with A770, just to clarify a bit, I can set it to 16GB when my system only have a total of 16GB of RAM without crashing right?

xCuri0, 2022-11-08:
@Tokobotenkai yes. only if you use 32GB RAM there might be a problem because no free address space.

Tokobotenkai, 2022-11-08:
@xCuri0 Okay thanks! Let me get back to you then.

val3nt33n, 2022-11-08:
@Tokobotenkai it doesn't matter how much RAM your system has. The GPU BAR is not allocated from the physical RAM.
It's allocated from the virtual space(Haswell's virtual space is 512GB), within the limits allowed by your BIOS.
So, you can have a system with only 4GB of RAM and still be able to use a GPU with a 16GB BAR. They are both independent and are both mapped in the virtual space for easy access by the CPU and applications.

Tokobotenkai, 2022-11-08:
@val3nt33n Understood! Thank you.

val3nt33n, 2022-11-09:
I plan to try the patch myself tomorrow once I'm done with work.

xCuri0, 2022-11-09:
@cursemex you can try this along with the "Remove <16GB BAR size limit" patch. It worked for a user on win-raid with 16GB GPU (RX 6900XT) and 16GB RAM. Not sure if it will work with 32GB RAM but it's worth trying

```
# PciHostBridge | Replace 8-16GB MMIO region with complete use of 64GB address space (Haswell). MAY REQUIRE DSDT MODIFICATION
8D6756B9-E55E-4D6A-A3A5-5E4D72DDF772 10 P:080000004823C1483BC3770C48BE0000000004000000EB7A48BB000000000C000000483BC3770C48BE0000000002000000EB5F48BB000000000E000000483BC3770C48BE0000000001000000EB4448BB000000000F000000483BC37707BE00000080EB2E48BB000000800F000000483BC37707BE00000040EB1848BB000000C00F000000483BC30F87A4FEFFFFBE00000020:010000004821C84839D8480F47D848BE000000E00F0000004829DE9090909090909066909066906690909090909090909066906690909090909090909066909066906690909090909090909066906690909090909090909066909066909090909090669066909090909090909090669090669090909090906690669090909090909090906690909090909090909090909090

```
yes i know it's really long

cursemex, 2022-11-09:
> 

Unfortunately, it doesn't apply the patch.
[image]
Used with "Don't downgrade 64-bit BARs to 32-bit (Haswell)" and "Remove <16GB BAR size limit (Ivy Bridge/Haswell)"

Having "Replace 8-16GB MMIO region with complete use of 64GB address space (Haswell)." by itself didn't work as well.
[image]

xCuri0, 2022-11-09:
@cursemex didn't you mention that you had to use BIOS version 1104 to get ReBarState working or did you fix the NvramSmi problem on the new one ? It applies to BIOS version 1104 without problem for me

cursemex, 2022-11-09:
> 

@xCuri0 my bad for not clarifying it clearly. I did extract Nvramsmi from 1104 and replaced it in 3503 during these tests after reading it from val3nt33n.

cursemex, 2022-11-09:
@xCuri0 I've tested it with the 3 Haswell patches on the 32GB. I get 6 beep error "GPU failure". There's the 1 short beep error(DRAM refresh failure) I've mentioned before when I apply 512MB-8GB. 

On the 16GB, it worked, but the resizable bar isn't working on AMD's Adrenaline.

[image]

[image]

xCuri0, 2022-11-09:
@cursemex i've seen one person with the same problem fix it with a ddu driver reinstall. could try that

i do think that it could be made to work with 32GB GB on the latest BIOS using a different patch which i have to make. the hardware is capable unlike older ivy bridge.

cursemex, 2022-11-09:
> @cursemex i've seen one person with the same problem fix it with a ddu driver reinstall. could try that, worst case will be windows reinstall.
> 
> i do think that it could be made to work with 32GB GB on the latest BIOS using a different patch which i have to make. the hardware is capable unlike older ivy bridge.

@xCuri0 I was about to retract my statement. I couldn't get into safe mode to use DDU (I think it's related to the DRAM refresh failure. Keyboard was not detected), so I uninstalled the drivers normally and reinstalled it.

I'll put a the [1104 firmware](https://mega.nz/file/JIQHiYJQ#pexuV8GxTQP-aGmKcmcb2RKsScTTTUYBnkDmzQ0pdi4) up for those who have 16gb and want to use it :)

[image]

xCuri0, 2022-11-09:
will add your motherboard to the working list with a note of 32GB RAM + 16GB BAR not working.

cursemex, 2022-11-09:
If you get around to rewrite the patch for 3503 later with a possible fix for 32gb ram I'll test it. I'll update my post on winraid with performance differences screenshots and 1104 linked after I wake up later :)

val3nt33n, 2022-11-10:
I confirm that **PciBus | Don't downgrade 64-bit BARs to 32-bit (Haswell)** patch works.  Thanks a lot @xCuri0 
[image]

val3nt33n, 2022-11-10:
I tried some games and none of them work. They crash shortly after they start loading to VRAM.

xCuri0, 2022-11-10:
@val3nt33n does a driver reinstall help ? me and several other amd users don't have any instability but i think you're the first one to try on windows with arc.

val3nt33n, 2022-11-10:
I did reinstall the driver (after DDU) and no change. I tried lowering the BAR size and only at 1GB is stable. Above 1GB all games crash.

[22 more replies]

closed 2022-11-11 by xCuri0