xCuri0/ReBarUEFI #9: Help with GA-X79-UP4 (Sandy Bridge-E) can`t access bios/pc wont boot with ReBAR greater than 1GB

aquamacho1, 2022-11-05:
Hi, so i did manage to unlock ReBAR on my Sandy Bridge - E motherboard (GA-X79-UP4), by modding bios with UEFITool, but values greater than 1GB = hang at boot/black screen and nothing happens.

Can you guide me (if this even possible) how to use UEFIPatch to unlock more of this awesomness? :3 
I do not understand what to do next or what values i need to select in UEFIPatch, niether how to use UEFIPatch :(

[image]

aquamacho1, 2022-11-05:
I`ve checked the bios, seems like i do not have any options called CSM, or above 4G decode.

xCuri0, 2022-11-05:
@aquamacho1 you need to enable the hidden 4G Decoding setting.

i can't be replying to everyone so i created a wiki page detailing how to do it [Enabling hidden 4G decoding](https://github.com/xCuri0/ReBarUEFI/wiki/Enabling-hidden-4G-decoding)

aquamacho1, 2022-11-06:
@xCuri0 thx for your help, but it seems i have a problems with enabling it. 
I did exactly step by step what wiki told, but in the end grub console says that i have GUID mismatch, and values are wrong (im entering 0x4 offset, but it says 0x01)

aquamacho1, 2022-11-06:
Or i messed up 5. step in the guide, do i rename `modGRUBShell.efi` to `bootx64.efi` and replace stock bootx64.efi with modGRUBShell.efi right?

xCuri0, 2022-11-06:
@aquamacho1 GUID mismatch doesn't matter. 0x04 and 0x4 are the same number so not a problem

Yes you do have to rename it

aquamacho1, 2022-11-06:
Well, either i did something wrong or it just doen`t work, 1024mb is still my limit.

xCuri0, 2022-11-06:
@aquamacho1 did you try setting it to 0x0, rebooting,  setting it to 0x1 and then again rebooting ? that's what i had to do to get it working on my mobo

if that doesn't work you can try using ```setup_var_3``` instead of ```setup_var_3``` for all the commands

aquamacho1, 2022-11-06:
Yes, i did tried doing that method to, still nothing, i will try `setup_var_3` now.

aquamacho1, 2022-11-06:
Nope, i think `setup_var_3` and `setup_var` are the same thing right? `var_3` shows same values as before.
I wonder if this is something to do with fact that intel says that my mobo only supports PCI 2.0 by default maybe?
When i had Nvidia 1060 gpu, by default i had to hack PCI 3.0 with that https://nvidia.custhelp.com/app/answers/detail/a_id/3135/~/geforce-gen3-support-on-x79-platform

xCuri0, 2022-11-06:
@aquamacho1 if pcie 3.0 wasn't working you wouldn't be able to use rebar at all. your screenshot also shows working pcie 3.0

maybe you could try using AMIBCP to change it's ``Access/Use``` from default to enabled and then enabling it. also these x79 boards seem to have a ton of hidden bios options so maybe another setting is causing the problem.

aquamacho1, 2022-11-06:
is this important?
```
0x46344 		One Of: PCIe Sub Decode, VarStoreInfo (VarOffset/VarName): 0x12E, VarStore: 0x1, QuestionId: 0xA9, Size: 1, Min: 0x0, Max 0x0, Step: 0x0 {05 A6 9B 03 9C 03 A9 00 01 00 2E 01 10 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00}
0x4636A 			One Of Option: Disabled, Value (8 bit): 0x0 (default) {09 0E 29 03 30 00 00 00 00 00 00 00 00 00}
0x46378 			One Of Option: Enabled, Value (8 bit): 0x1 {09 0E 28 03 00 00 01 00 00 00 00 00 00 00}
0x46386 		End One Of {29 02}
```

xCuri0, 2022-11-06:
@aquamacho1 i thin that it's not needed as it's disabled by default on my motherboard and 4g decode with 8gb bar still works

aquamacho1, 2022-11-06:
> try using AMIBCP to change it's Access/Use from default to enabled and then enabling it
AMIBCP wont open my bios file, or i dont know how to open it

xCuri0, 2022-11-06:
@aquamacho1 correct AMIBCP version for your board is v4.55 it opens for me. Also did you try using the newer F7 version bios maybe it fixes this issue ?

aquamacho1, 2022-11-06:
> Also did you try using the newer F7 version bios maybe it fixes this issue ?

I did not yet, will do if nothing works.

meanwhile found these, how do i enable them? (im sorry, but i`ve never before modded bioses using hex)
[image]

xCuri0, 2022-11-06:
@aquamacho1 don't change anything except **Access/Use** to **USER** on **Above 4G decoding**. Otherwise you might get no boot. All this will do is make the option visible in the BIOS

aquamacho1, 2022-11-06:
Flashed modded bios with 4G enabled by dafault, but no dice, checked with the grub console, it is was enabled afetr bios flash.
Interesting thing, after i got to windows and checked GPU-Z, above 4G Decode listed as disabled.

xCuri0, 2022-11-06:
I guess you can try the F7 bios then

aquamacho1, 2022-11-06:
> I guess you can try the F7 bios then

Nope, all the same, can you give me your mobo name? i will check what options are enabled between your and mine. Maybe will find something.

xCuri0, 2022-11-06:
@aquamacho1 Gigabyte B75M-D3H

aquamacho1, 2022-11-06:
> @aquamacho1 Gigabyte B75M-D3H

what rev?

xCuri0, 2022-11-06:
iirc it's 1.1

aquamacho1, 2022-11-06:
So using your guide on DSDT patching from wiki, i`ve extracted and decoded AmiBoardInfo --> DSDT.aml --> DSDTMod.dsl
and found this in there:

```
            Device (VTDR)
            {
                Name (_HID, EisaId ("PNP0C02") /* PNP Motherboard Resources */)  // _HID: Hardware ID
                Name (_UID, 0x04)  // _UID: Unique ID
                Name (BUF0, ResourceTemplate ()
                {
                    Memory32Fixed (ReadOnly,
                        0x00000000,         // Address Base
                        0x00002000,         // Address Length
                        _Y0E)
                })
                Method (_STA, 0, NotSerialized)  // _STA: Status
                {
                    If ((VTD0 == 0xFFFFFFFF))
                    {
                        Return (Zero)
                    }

                    Return (0x0F)
                }

                Method (_CRS, 0, Serialized)  // _CRS: Current Resource Settings
                {
                    CreateDWordField (BUF0, \_SB.PCI0.VTDR._Y0E._BAS, VBR0)  // _BAS: Base Address
                    CreateDWordField (BUF0, \_SB.PCI0.VTDR._Y0E._LEN, VLT0)  // _LEN: Length
                    VBR0 = VTD0 /* \VTD0 */
                    If ((VTD0 == 0xFFFFFFFF))
                    {
                        VLT0 = Zero
                    }

                    Return (BUF0) /* \_SB_.PCI0.VTDR.BUF0 */
                }
            }

```

Does this somehow related to the issue?

xCuri0, 2022-11-07:
this is the relevant code
```
                QWordMemory (ResourceProducer, PosDecode, MinFixed, MaxFixed, Cacheable, ReadWrite,
                    0x0000000000000000, // Granularity
                    0x0000000000000000, // Range Minimum
                    0x0000000000000000, // Range Maximum
                    0x0000000000000000, // Translation Offset
                    0x0000000000000000, // Length
                    ,, _Y05, AddressRangeMemory, TypeStatic)
```

```                    If ((MALH || MALL))
                    {
                        CreateDWordField (CRS1, \_SB.PCI0._Y05._MIN, MN8L)  // _MIN: Minimum Base Address
                        Local0 = (0xB4 + 0x04)
                        CreateDWordField (CRS1, Local0, MN8H)
                        MN8L = MABL /* \_SB_.PCI0.MABL */
                        MN8H = MABH /* \_SB_.PCI0.MABH */
                        CreateDWordField (CRS1, \_SB.PCI0._Y05._MAX, MX8L)  // _MAX: Maximum Base Address
                        Local1 = (0xBC + 0x04)
                        CreateDWordField (CRS1, Local1, MX8H)
                        CreateDWordField (CRS1, \_SB.PCI0._Y05._LEN, LN8L)  // _LEN: Length
                        Local2 = (0xCC + 0x04)
                        CreateDWordField (CRS1, Local2, LN8H)
                        MN8L = MABL /* \_SB_.PCI0.MABL */
                        MN8H = MABH /* \_SB_.PCI0.MABH */
                        LN8L = MALL /* \_SB_.PCI0.MALL */
                        LN8H = MALH /* \_SB_.PCI0.MALH */
                        MX8L = MAML /* \_SB_.PCI0.MAML */
                        MX8H = MAMH /* \_SB_.PCI0.MAMH */
                    }
```

I do find it weird they are using ```CreateDWordField``` which can't hold a 64-bit address instead of ```CreateQWordField``` . Can you send a screenshot of Device Manager showing View -> Resources by Type Large Memory (like the below screenshot). It should show the resource range the DSDT gives.

aquamacho1, 2022-11-07:
@xCuri0 i don`t have large memory :cry
[image]
[image]

aquamacho1, 2022-11-07:
[image]

xCuri0, 2022-11-07:
@aquamacho1 4G decoding is on ? if you were getting resources error in windows with 4g decoding on that means the BIOS is still allocating the device properly and can be fixed with DSDT. otherwise it's very hard to fix (BIOS patching)

aquamacho1, 2022-11-07:
@xCuri0 problem is i didn`t got to windows a single time after 1G size increase

> 4G decoding is on ?

Well, in the grub console it shows 0x01, so i think it is, unless 4G Decode is broken on this board.

aquamacho1, 2022-11-07:
@xCuri0 can you help me figure out what do i put in these? i have two of them, one is _Y05 and another is _Y0B, both are the same.
```
                    0x0000000000000000, // Granularity - judging by another values in the file, i leave this as is
                    0x0000000000000000, // Range Minimum - what do i put here ?
                    0x0000000000000000, // Range Maximum - here 
                    0x0000000000000000, // Translation Offset
                    0x0000000000000000, // Length - and here ?
```
My cpu address size are 46 bits physical, 48 bits virtual

xCuri0, 2022-11-07:
If you can't POST DSDT edit won't help. It only will help if you have problems booting Windows (with Linux working). I'm not sure about those DSDT values I've only edited the CreateQWordField stuff

aquamacho1, 2022-11-07:
@xCuri0 Yep, you right.
I think those 0x0000000000000000 values this is not the problem. I`ve just checked DSDT file from mobo in #1 post.
That X79 Supermicro x9dri-ln4f+ had working ReBAR, but has the same exact values as mine and the rest X79 bioses that i`ve checked. 
I think key is that it has MMIOH in it, i think this is the problem, or smth else that i dont know.

aquamacho1, 2022-11-07:
I completely forgot to mention that i 've used `REBAR Legacy On.reg` file, without it there are no option to enable bar in amd control panel.

aquamacho1, 2022-11-08:
@xCuri0 soooo i've checked mine PCIBus.efi file, it looks alright, any ideas?

[image]

aquamacho1, 2022-11-09:
@xCuri0 Can you make a screenshot like above of a region where this patch is supposed to be, i will check my PciRootBridge.
# PciHostBridge | Remove <4GB BAR size limit in SubmitResources (Ivy Bridge)

xCuri0, 2022-11-09:
I don't think that's the cause. 2GB would work then

aquamacho1, 2022-11-11:
@xCuri0  so i think i maybe found the problem in my case, on my mobo ReBarUEFI driver is allocating its space not from virtual space, but from System Board Resources instead
```
            Device (IOH)
            {
                Name (_HID, EisaId ("PNP0C01") /* System Board */)  // _HID: Hardware ID
                Name (_UID, 0x0A)  // _UID: Unique ID
                Name (IOHM, ResourceTemplate ()
                {
                    Memory32Fixed (ReadWrite,
                        0xFC000000,         // Address Base
                        0x01000000,         // Address Length
                        )
                    Memory32Fixed (ReadWrite,
                        0xFD000000,         // Address Base
                        0x01000000,         // Address Length
                        )
                    Memory32Fixed (ReadWrite,
                        0xFE000000,         // Address Base
                        0x00B00000,         // Address Length
                        )
                    Memory32Fixed (ReadWrite,
                        0xFEB00000,         // Address Base
                        0x00100000,         // Address Length
                        )
                    Memory32Fixed (ReadWrite,
                        0xFED00400,         // Address Base
                        0x0003FC00,         // Address Length
                        )
                    Memory32Fixed (ReadWrite,
                        0xFED45000,         // Address Base
                        0x000BB000,         // Address Length
                        )
```
Those two sections is what appears to be my ReBAR values (1GB and 256M respectively)

```
                    Memory32Fixed (ReadWrite,
                        0xFEB00000,         // Address Base
                        **0x00100000**,         // Address Length
                        )
                    Memory32Fixed (ReadWrite,
                        0xFED00400,         // Address Base
                        **0x0003FC00**,         // Address Length
                        )
```

aquamacho1, 2022-11-11:
And i think this section probably is what responsible for mine problems booting bigger ReBAR values.
```
                    Memory32Fixed (ReadWrite,
                        0x00000000,         // Address Base
                        0x00000000,         // Address Length
                        _Y0C)
                })
                Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
                {
                    If (LAPB)
                    {
                        CreateDWordField (IOHM, \_SB.PCI0.IOH._Y0C._BAS, APB)  // _BAS: Base Address
                        CreateDWordField (IOHM, \_SB.PCI0.IOH._Y0C._LEN, APL)  // _LEN: Length
                        APB = LAPB /* \LAPB */
                        APL = LAPL /* \LAPL */
                    }

                    Return (IOHM) /* \_SB_.PCI0.IOH_.IOHM */
                }
            }
```

aquamacho1, 2022-11-11:
@xCuri0 when im trying rebuild DSDT.dsl, im getting so much errors, do you by any chance know whe there are so many?
This is stock file without any changes after decompile.
Maximum error count (200) exceeded
[badcode.zip](https://github.com/xCuri0/ReBarUEFI/files/9990079/badcode.zip)

xCuri0, 2022-11-11:
you have to manually fix each error which isn't easy tbh especially when there are so many. my DSDT only had 2 and were fixable with a few google searches.

aquamacho1, 2022-11-11:
@xCuri0 i can't find any of the code from badcode file in the stock file O_o, there is something wrong here.

aquamacho1, 2022-11-11:
@xCuri0 i did compiled DSDT with both 2013 and 2022 IASL without an errors, but resulting files are identical to decompiled. Amiboardinfotool creates amiboardifo.efi from compiled before DSDT, but its 325kb in size and its has some spagetti code at the beginning and the rest is similar to decompiled.
I give up :(

aquamacho1, 2022-11-16:
Update: So from MacOS using MaciASL in legacy mode i`ve edited and recompiled DSDT with those lines removed/edited below and some more, but no luck, these related to something else i guess.
https://github.com/xCuri0/ReBarUEFI/issues/9#issuecomment-1311420328
https://github.com/xCuri0/ReBarUEFI/issues/9#issuecomment-1311438724

xCuri0, 2022-11-16:
@aquamacho1 to my knowledge DSDT editing won't solve problems where you can't even POST. Only if you can't boot OS or OS gives no resources error it can hep

aquamacho1, 2022-11-16:
@xCuri0 i was thinking about to replace PCIBus or Rootbridge\other drivers from another mobo, but im scared it may cause damage to the board or gpu :(

aquamacho1, 2022-11-18:
@xCuri0 do you know if there is a way to modify `PciRootBridge` anyhow? i mean if there no way to edit those, its over for my board, i think problem in my case is MMIO missing from `PciRootBridge` and another file.

xCuri0, 2022-11-19:
@aquamacho1 the problem is it's not allocating addresses in the 64-bit region. i don't know anything about modifying PciRootBridge though

aquamacho1, 2022-11-26:
Well, after two weeks of playing around, i didn't find anything new :D
But while testing I've edited my bios heavily xD

- Modded all values to be visible/editable, and they actually do work and write/save treir values. No efi console needed.

- Added patches from MaciASL

- Bios has up to date microcodes and drivers (i think)

- Some variables are broken/have duplicates, im lazy to fix them :)

<details>
  <summary>Spoiler warning</summary>
  
[image]
[image]
[image]

</details>

Here is a file if anyone is brave enough like me to try it (im not responsible if your PC goes on fire after flashing, seriously.)
[2050.zip](https://github.com/xCuri0/ReBarUEFI/files/10096469/2050.zip)

XManPro79, 2023-03-02:
Just a consideration...while the x79 platform supports PCIE 3.0... the 3930K supports only PCIe 2.0 as per the Ark website; try using a 22nm chip that supports PCIe 3.0 and you may get your ReBAR working.  It does require 3.0 as far as I understand.

xCuri0, 2023-03-02:
@XManPro79 PCIe Gen3 isn't needed for rebar. Rebar does work here but not at full size because of firmware bug

aquamacho1, 2023-04-16:
Well, out of complete boredom, and after AMD borked my 1G rebar, i have returned to this, and i have some progress:
Edited DSDT again, this time using tutorial from [rebar wiki](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching#dsdt-broken-4g-decoding-fix-haswell-and-up)
like this:

>                 QWordMemory (ResourceProducer, PosDecode, MinFixed, MaxFixed, Cacheable, ReadWrite,
>                     0x0000000000000000, // Granularity
>                     0x0000008000000000, // Range Minimum
>                     0x000000FFFFFFFFFFF, // Range Maximum
>                     0x0000000000000000, // Translation Offset
>                     0x0000008000000000, // Length
>                     ,, _Y05, AddressRangeMemory, TypeStatic)

And 

>                 QWordMemory (ResourceProducer, PosDecode, MinFixed, MaxFixed, Cacheable, ReadWrite,
>                     0x0000000000000000, // Granularity
>                     0x0000000000010000, // Range Minimum
>                     0x00000000000FFFFF, // Range Maximum
>                     0x0000000000000000, // Translation Offset
>                     0x0000000000010000, // Length
>                     ,, _Y0B, AddressRangeMemory, TypeStatic)
> 

And now i have this:
[image]
And my sound from Realtek onboard audio is gone. But i still can`t enable rebar larger than 1G

Any ideas @xCuri0 ?

aquamacho1, 2023-04-16:
So i need to move GPU and PCI Bus to Large memory like this right?
[image]

xCuri0, 2023-04-16:
@aquamacho1 With 4G decode on and BAR set to 4GB or larger it will show in Large Memory.

Also DSDT editing won't fix POST issues, only if you can't boot Windows it's DSDT related. 

Try setting these BIOS settings:

**4G Decode**: On
**Full Screen LOGO Show**: On
**CSM Support**: Never
**Boot Mode Selection**: UEFI
**Storage Boot Option Control**: UEFI Only
**Display Boot Option Control**: UEFI Only
**PCI Device ROM Priority**: UEFI OpRom

aquamacho1, 2023-04-16:
> 4G Decode: On
> Full Screen LOGO Show: On
> CSM Support: Never
> Boot Mode Selection: UEFI
> Storage Boot Option Control: UEFI Only
> Display Boot Option Control: UEFI Only
> PCI Device ROM Priority: UEFI OpRom
I had these enabled like this from the start

aquamacho1, 2023-04-16:
@xCuri0 Does it matter in which PCI slot my gpu is plugged in? All this time my gpu was in slot 5 (second x16 slot)

xCuri0, 2023-04-16:
@aquamacho1 i don't think so but sometimes these HEDT boards have different MMIO allocation per slot

xCuri0, 2023-05-27:
@aquamacho1 you can try these patches

```
# PciBus | Increase 256 byte configuration space limit to 4096 byte allowing access to PCIe extended configuration space
3C1DE39F-D207-408A-AACC-731CFB7F1DD7 10 P:FF000000760B48B8:00100000760B48B8

# Runtime | Remove 4GB limit for CpuIo2 (Fixes 4G Decoding USB issues)
CBC59C4A-383A-41EB-A8EE-4498AEA567E4 10 P:B9FFFFFFFF490FAFC14903C0483BC1776C:6690669090490FAFC14903C06690906690

# PciHostBridge | Remove 4GB limit for PciRootBridgeIo.Mem (Fixes 4G Decoding USB isues)
8D6756B9-E55E-4D6A-A3A5-5E4D72DDF772 10 P:493B4C24C0771B:66906690669090
8D6756B9-E55E-4D6A-A3A5-5E4D72DDF772 10 P:493B4C24C0771C:66906690669090
```
maybe it will get video output working

aquamacho1, 2023-05-27:
@xCuri0 Im sorry, but i can`t test this right now, that X79 board currently not in use anymore.
I`ve bought myself an X99 board not long ago, and rebuild my PC around that using spare parts from X79 system (ram cooler etc.).

ZOXZX, 2023-06-12:
Hello!

Also trying to enable greater than 1GB ReBar on x79.
I have dual BIOS MSI X79A GD65 board.
With 3C1DE39F patch from above and 4G enabled, I can boot into Windows and connect remotely (no display).

[image]

As you can see, only PCI Bus is in large memory area.

With ReBar size of 16GB, I can see low res windows boot screen and can get display out, although probably 640x480 black and white.
Anyway this happens:

[image]

Some parts of GPU are in large area, some not.

As mentioned, I have dual BIOS and CH341 so if you think we can work this out, I'm willing to test.

Thank you!

ZOXZX, 2023-06-14:
Sorry for delay, I'm short on time.
I'll have some free time from next tuesday and then I'll sent diagnostics ...

xCuri0, 2023-11-21:
I'm closing this because I think we've solved it at #48

closed 2023-11-21 by xCuri0