xCuri0/ReBarUEFI #11: List of working motherboards

xCuri0, 2022-11-07:
Motherboards which don't support atleast 4GB BAR size are considered not working. Reply to this issue with screenshots of it working and I will add your motherboard.

| User  | Motherboard | GPU | BAR size tested | UEFIPatch req (Y/N) | Notes |
| ------------- | ------------- | ------------- | ------------- | ------------- | ------------- |
| @xCuri0   | Gigabyte B75M-D3H |  RX 580 8GB | 8GB | Y | 
| @romulus2k4  | ASUS P8Z77-V PRO | RX 570 4GB | 4GB | Y |
| prazola (win-raid)  | ASRock Z97 Extreme9 | RX 6900 XT | 16GB | Y |
| Miyconst | Huanzhi X99-QD4 | RTX 3060 | 16GB | N |

sidex15, 2022-11-07:
[image]
Chinese Unbranded X79 motherboard...

Tokobotenkai, 2022-11-08:
Would my Z97-Pro be considered working? I haven't tried the 16GB patch (only tested to 8GB), but it needs the replacement of the older 1204 NvramSmi module and the PciBus UEFIPatch to work. The GPU tested is the Intel Arc A380 (will test the A770 once I get a replacement power supply).

xCuri0, 2022-11-08:
@Tokobotenkai anything 4GB or higher is considered working but if you find a larger size doesn't I can add a note

EDK-Rise, 2022-11-08:
Hi, i have z270 m7 with 7700k. my motherboard has 4G decoding in bios. how to patch my bios?
here my bios link:
https://download.msi.com/bos_exe/mb/7A57v10.zip

xCuri0, 2022-11-08:
@EDK-Rise GTX 1080 doesn't support resizable bar

aquamacho1, 2022-11-08:
> [image] Chinese Unbranded X79 motherboard, RX 570 8GB...

@sidex15 But can you actually check what a name of this board?, im trying figure out how to patch mine X79, checking whats the difference between them, maybe i will find something.

sidex15, 2022-11-10:
> 

well after deep diving in the russian overclocking forums I found that my board is Jingsha X79M-S 1.0 the [Picture is the same as mine](https://forums.overclockers.ru/download/file.php?id=198638&sid=283a7a2da6f2db91e16ff2239d8076ce&mode=view)...

I just followed the X99 guide from Miyconst and use [Enabling hidden 4G decoding](https://github.com/xCuri0/ReBarUEFI/wiki/Enabling-hidden-4G-decoding)...

aquamacho1, 2022-11-10:
@sidex15 This is awesome, thx.

xCuri0, 2022-11-10:
@sidex15 I think macOS doesn't support large BARs not a problem with the bios or anything

sidex15, 2022-11-10:
> @sidex15 I think macOS doesn't support large BARs not a problem with the bios or anything

Yeah i'm using opencore 0.8.3 even with resizeapplegpubars set to 0 won't even boot to macOS...
I guess it's still using the max BARs...
I'll try lowering the BAR size see if it boots...

sidex15, 2022-11-10:
> @sidex15 I think macOS doesn't support large BARs not a problem with the bios or anything

@xCuri0 yep you're right setting a rebar size of 256mb boots to macOS... I also found out that my motherboard is Jingsha X79M-S 1.0 after digging in the Russian overclocking forum...

LatiosAeon, 2022-11-12:
ASUS Z170-WS & Intel ARC A770, BAR size set to unlimited and the full 16GB worked
[image]

kodekq, 2022-11-15:
A big thank you @xCuri0 and @Mak3rde
Asus z170-a, 5700xt and use patch "Don't downgrade 64-bit BARs to 32-bit (by @Mak3rde)"
[image]

gyhmz, 2022-11-16:
MZ32-AR0 (rev. 1.0) +ARC A380 +TESLA M40
Before:

[image]
[image]

After inserting ReBarDxe.ffs and running ReBarState.exe:

[image]

gyhmz, 2022-11-16:
> Motherboards which don't support atleast 4GB BAR size are considered not working. Reply to this issue with screenshots of it working and I will add your motherboard.
> User 	Motherboard 	GPU 	BAR size tested 	UEFIPatch req (Y/N) 	Notes
> @xCuri0 	Gigabyte B75M-D3H 	RX 580 8GB 	8GB 	Y 	
> @romulus2k4 	ASUS P8Z77-V PRO 	RX 570 4GB 	4GB 	Y 	Games crash with 4GB BAR size, 2GB works fine thought to be driver issue.
> prazola (win-raid) 	ASRock Z97 Extreme9 	RX 6900 XT 	16GB 	Y 	
> Miyconst 	Huananzhi X99-QD4 	RTX 3060 	16GB 	N 	
> @sidex15 	Unbranded X79 	RX 570 8GB 	8GB 	N 	
> @dabm-git 	Huananzhi X99-TF 	Vega 64 	8GB 	N 	
> @Tokobotenkai 	ASUS Z97-Pro 	Arc A770 	16GB 	Y 	
> @cursemex 	ASUS MAXIMUS VII HERO (Z97) 	RX 6700 XT 	16GB 	Y 	16GB BAR doesn't work with 32GB RAM
> @val3nt33n 	ASUS Z97i-Plus 	Arc A750 	8GB 	Y 	
> @Mak3rde 	ASUS X99-A II 	RX 6950 XT 	16GB 	Y 	
> @LatiosAeon 	ASUS Z170-WS 	Arc A770 	16GB 	Y 	
> @LatiosAeon 	ASUS Z170 Sabertooth MARK 1 	Arc A770 	16GB 	Y 	
> @kodekq 	ASUS Z170-A 	RX 5700 XT 	8GB 	Y 	
> @gyhmz 	Gigabyte MZ32-AR0 	Arc A380 	8GB 	Y

xCuri0, 2022-11-16:
@gyhmz Tesla M40 doesn't have ReBar it just has a large BAR by default.

CiprianFlorin-Ifrim, 2022-11-17:
I can confirm it works on a Supermicro H11DSi-NT (tested with bios v1.3, should work with the v2.4 as well). Tested with a 4090 FE.
GPU-Z says 3 bars are available - 16MB, 256MB, and 32GB.

Thanks a lot for making this tool.

Divin1ty666, 2022-11-17:
@Tokobotenkai Hey! Do you mind sharing modified bios for z97-pro, please?

Tokobotenkai, 2022-11-18:
> @Tokobotenkai Hey! Do you mind sharing modified bios for z97-pro, please?

There you go: https://drive.google.com/file/d/1yZb9LkaTvvBjAkJfGSO-hPgm2L_W9SWH/view?usp=sharing

I have random issues where my keyboard and mouse is not detected on boot, not sure if it is a dying USB or a BIOS issue though.

cursemex, 2022-11-18:
> There you go: https://drive.google.com/file/d/1yZb9LkaTvvBjAkJfGSO-hPgm2L_W9SWH/view?usp=sharing
>
> I have random issues where my keyboard and mouse is not detected on boot, not sure if it is a dying [...]

@Tokobotenkai I have them on my Hero too. Funny enough they only occur after setting rebar state value.

Tokobotenkai, 2022-11-18:
> @Tokobotenkai I have them on my Hero too. Funny enough they only occur after setting rebar state value.

I suspect it is some device being mapped in the same error, but the error is intermittent. I workaround it by using a PS2 keyboard.

Divin1ty666, 2022-11-19:
@Tokobotenkai Thank you!

kkkyloo, 2022-11-19:
[image]
machinist mr9a pro v2 work with re bar, but windows starts up slowly

supermanz87, 2022-11-19:
Injected FFS rebar into bio's on **Asus x99 Strix** with xeon mod and enabled 4G Decoding  in bio's correctly.

System spec's-
x99 strix motherboard
60gb ddr4 2133mhz
E5-2696 V3 @3.4Ghz 18/36 All turbo unlock FFS Mod
Nvidia RTX 3060Ti
samsung 970 nvme
Windows 11 pro x64 (Uefi Mode)

Rebar exe not injecting script in windows 11 pro x64

Obtained SE_SYSTEM_ENVIRONMENT_NAME
ReBarState variable doesn't exist / Disabled. Enter a value to create it

Verify that 4G Decoding is enabled otherwise system will not POST with GPU. There is also a possibility of BIOS not supporting large BARs even with 4G decoding enabled.

It is recommended to first try smaller sizes above 256MB in case BIOS doesn't support large BARs.

Enter ReBarState Value
      0: Disabled
Above 0: Maximum BAR size set to 2^x MB
     32: Unlimited BAR size

32
Writing value to ReBarState

Failed to write ReBarState UEFI variable
GetLastError: 5
You can close the app now

xCuri0, 2022-11-19:
@supermanz87 you have to use UEFIPatch. it's described in the README how to use it

SEt-t, 2022-11-20:
Asus Z170-P D3 : requires patch "Don't downgrade 64-bit BARs to 32-bit (by @Mak3rde)"
[image]
[image]

cursemex, 2022-11-20:
@xCuri0 I found this:
https://github.com/LongSoft/UEFITool/pull/116#issuecomment-370165778

`842A454A-75E5-408B-8B1C-36420E4E3F21 10 P:9801000072:0000000072`

Vit9696's code allows for nvramsmi to write on UEFI for Z97's Asus motherboards like my Maximus VII hero. Already tested it, and it seems working.

xCuri0, 2022-11-21:
@cursemex thanks for finding this. i've added it to patches.txt

Divin1ty666, 2022-11-21:
So I've modified my Asus Z97 Deluxe bios with FFS DXE module, then patched it with those patches: **# NvramSmi NVRAM whitelist unlock (by @vit9696)** and  **# PciBus | Don't downgrade 64-bit BARs to 32-bit (Haswell)**. 

All successful, flushed bios - it loaded. Enabled 4G Decoding and disable CMS. Although I'm not sure if some ReBar option should appear in bios menu as well? Because it doesn't.

But after using ReBarState.exe and setting value to 32(also tried 16) mobo won't boot after I restart it. At first it makes singe beep with pc-speaker as if its OK, but after 2 seconds there goes another 3 beeps and thats it. Any suggestions?

cursemex, 2022-11-21:
@Divin1ty666 No, you shouldn't see any options for ReBar. Once you've set your Rebar Value, you should aee it in GPU-Z.

If you can't get into windows, then that's because you jumped to a high value than to try smaller values to see which one is more stable. Try clearing CMOS, if that doesn't work, disconnect your GPU (Removing power cable from it should suffice), then connect your display cable into your motherboard if it has that option and reset the value to something smaller.

You didn't mention how much ram you have installed. If it is 16-24GB use this patch too:
`# PciHostBridge | Replace 8-16GB MMIO region with complete use of 64GB address space (Haswell).` 
If you are at 32GB, it won't work as it needs further modification. I have mine working with 16GB rebar at 24GB of ram. I guess I'll open a new thread today so we can see if we can fix that 32GB. 

Here are the patches I've used:
`# PciBus | Remove <16GB BAR size limit (Ivy Bridge/Haswell)
3C1DE39F-D207-408A-AACC-731CFB7F1DD7 10 P:B8FFFFFFFF030000004C3B:B8FFFFFFFFFFFFFF004C3B
# PciBus | Don't downgrade 64-bit BARs to 32-bit (Haswell)
3C1DE39F-D207-408A-AACC-731CFB7F1DD7 10 P:833E067506C70604000000833E077506C70605000000:66906690669066906690669066906690669066906690
# PciHostBridge | Replace 32/48GB - 63GB MMIO region with 32/48GB - 64GB (Haswell)
8D6756B9-E55E-4D6A-A3A5-5E4D72DDF772 10 P:3BC3771548BE000000C00F000000482B:3BC3771548BE0000000010000000482B
# Fix Nvramsmi
842A454A-75E5-408B-8B1C-36420E4E3F21 10 P:9801000072:0000000072`

The "norm" is to have 1 beep after the normal post after you set the rebar value. There will be an issue of USB not working in bios. So set your overclocking, rebar, csm before applying the value of rebar. It'll mess with entering safe mode of windows too, check my win-raid post for guides to workaround it if you need it.

Divin1ty666, 2022-11-21:
**cursemex** I have 16Gb, but  **# PciHostBridge | Replace 8-16GB MMIO region with complete use of 64GB address space (Haswell).** patch cannot be applied: **No patches can be applied to input file**, all other patches are applied already.

xCuri0, 2022-11-21:
@Divin1ty666 what's your GPU ? If it's a 32GB BAR such as 3090 it won't work even with 16GB RAM

Divin1ty666, 2022-11-21:
**[xCuri0](https://github.com/xCuri0)** rx 6700 xt

xCuri0, 2022-11-21:
@Divin1ty666 do smaller bar sizes atleast work ? start with 512mb and increase till you get the issue

Divin1ty666, 2022-11-21:
**@Curi0**  So should I start with "1"? Or what value should enter for 512mb? I don't get I how they correspond, I know 0 means disable and 32 is max, right?)))

Intelevande, 2022-11-24:
Hello, I believe I have rebar working on my MSI Z97 Gaming 5 w/ HP 3060 Ti (OEM) and 32GB RAM
[image]

xCuri0, 2022-11-24:
@Intelevande did you need to use UEFIPatch ? all other Z97 users had to

Intelevande, 2022-11-24:
> @Intelevande did you need to use UEFIPatch ? all other Z97 users had to

I didn't use UEFIPatch, I followed your instructions on inserting the FFS and enabling the hidden 'Above 4G Decoding' option, set 32 in the rebarstate.exe program and it seemed to be working for me.  Do you think I should expect some issues even though gpuz says it's enabled?

DanisMiryasov, 2022-11-24:
Hi everyone, I ran into a problem when using UEFI Patch. When trying to apply patches.txt PowerShell writes a message 'patches.txt file not found or can't be read'. UEFIPatch, UEFITool 0.28.0 and patches.txt are located in the same directory.
Asus x99 Strix

cursemex, 2022-11-24:
> Hi everyone, I ran into a problem when using UEFI Patch. When trying to apply patches.txt PowerShell writes a message 'patches.txt file not found or can't be read'. UEFIPatch, UEFITool 0.28.0 and patches.txt are located in the same directory. Asus x99 Strix

Use Command Prompt with admin privilage instead. You can search up CMD in the task bar search, then right click it and run as admin.
Then "cd [UEFIPatch.exe directory]".
And follow the rest of the instructions from here. Make sure to use "UEFIPatch.exe [filename] [patches.txt] -O [output file]" during the commands with the ".exe" included in the command.

LordAdo96, 2022-11-24:
> 

but where to apply those patches? Im using haswell too, mobo is GA-Z97X-Gaming

LordAdo96, 2022-11-24:
> @Divin1ty666 No, you shouldn't see any options for ReBar. Once you've set your Rebar Value, you should aee it in GPU-Z.
>
> If you can't get into windows, then that's because you jumped to a high value [...]

but where to apply those patches? Im using haswell too, mobo is GA-Z97X-Gaming

SEt-t, 2022-11-24:
@LordAdo96 do you see the name of this issue? It's "List of working motherboards", not "Support for people who don't understand a thing".

@xCuri0 I think with rising popularity such people will swarm this repository. Maybe redirect them somewhere?

xCuri0, 2022-11-24:
@SEt-t i do think that's starting to be a problem but i don't know where they can create support questions. maybe github discussions or setting up a discord ?

cursemex, 2022-11-24:
>  It's "List of working motherboards", not "Support for people who don't understand a thing".
@SEt-t [image]

@xCuri0 I'll be down to help with a discord. I was on the verge of writing a dummy's guide on patching, or just refer them to my winraid step-by-step.

@LordAdo96 You already opened an Issue, I'll just reply there.

LordAdo96, 2022-11-24:
My bad sorry guys really

xCuri0, 2022-11-24:
@cursemex already made a dummy's guide on patching including the MMTool workaround. It's on the wiki

will see about creating a discord server later

DanisMiryasov, 2022-11-24:
> > Привет всем, я столкнулся с проблемой при использовании UEFI Patch. При попытке применить patches.txt PowerShell пишет сообщение "файл patches.txt не найден или не может быть прочитан". UEFIPatch, UEFITool 0.28.0 и patches.txt находятся в одном каталоге. Asus x99 Стрикс
> 
> Вместо этого используйте командную строку с правами администратора. Вы можете найти CMD в поиске на панели задач, затем щелкнуть его правой кнопкой мыши и запустить от имени администратора. Затем «cd [каталог UEFIPatch.exe]». И следуйте остальным инструкциям отсюда. Обязательно используйте «UEFIPatch.exe [имя файла] [patches.txt] -O [выходной файл]» во время команд с включенным в команду «.exe».

Thanks for the advice. CMD turned out, + UEFIPatch.
ASUS STRIX X99 GAMING + MSI RTX 3050 GAMING X
[image]

terminatorul, 2022-11-24:
Gigabyte X470 AORUS GAMING 5 WiFi + Ryzen 7 2700X + Intel Arc A770 LE 16GB
[image]

Worked with 16GB BAR and unlimited BAR (select input value 32 for ReBarState.exe)
Specific UEFI patch required to write the new image with Qflash.

[image]

Divin1ty666, 2022-11-26:
So, I tried entering values 8-10-12 in ReBarState with modded bios and mobo(asus z97 deluxe) booted, but usb keyboard don't work until I re-plug it in. So I can't get into the bios . Thing is GPU-Z says **Resizable Bar enabled in bios - NO** other requirements are YES. What do I do?((

xCuri0, 2022-11-26:
@Divin1ty666 does GPU-Z show the BAR size has increased ?

Divin1ty666, 2022-11-26:
[image]

This is how it is now. Btw valuse 14 isn't working, I guess its because 16Gb size is more than rx6700xt has vram, so I can only go with 8Gb value...

xCuri0, 2022-11-26:
@Divin1ty666 update your GPU-Z. Old version probably has incorrect detection and doesn't show BAR size

Divin1ty666, 2022-11-26:
**@xCuri0** I've tried patch _# PciHostBridge | Replace 16GB MMIO region with complete use of physical address space (Ivy Bridge). MAY REQUIRE DSDT MODIFICATION_ - cannot be applied, it says No patches can be applied to input file. Perhaps I'm trying wrong patch, could you please tell me what exact patch should I use?

xCuri0, 2022-11-26:
@Divin1ty666 ```Remove <16GB BAR size limit (Ivy Bridge/Haswell)```

Divin1ty666, 2022-11-26:
That patch I've applied successfully already

xCuri0, 2022-11-26:
@Divin1ty666 how much RAM do you have installed ? any additional PCIe cards or dual GPU ? I don't see why 16GB won't work if you have less than 32GB RAM

Divin1ty666, 2022-11-26:
I have 16Gb ram, I do have Creative SB X-Fi Titanium HD soundcard, but i've unplugged it for now to be able to get to bios socket)

xCuri0, 2022-11-26:
@Divin1ty666 did you try the ```PciHostBridge | Replace 8-16GB MMIO region with complete use of 64GB address space (Haswell). MAY REQUIRE DSDT MODIFICATION``` patch ?

Divin1ty666, 2022-11-26:
Yes, I have. Same thing it can't be applied.

[1123 more replies]