xCuri0/ReBarUEFI #12: ASUS Z170-WS Partially working with ReBAR

LatiosAeon, 2022-11-08:
**System**
 - Motherboard: ASUS Z170-WS
 - BIOS Version: 3062
 - GPU: RTX 3080 Ti FE

**Description**
The ReBAR module has been placed in the correct volume, it was extracted and replaced per instructions for "ASUS no boot after patching" but with the same failing result, the script appears to be working but only with 512MB, 1024MB and higher always results in the system completing POST checks but nothing more, no display or response, CSM disabled and above 4G Decode enabled but Resizable BAR enabled in BIOS also says No
[image]
[Z170WS_test.zip](https://github.com/xCuri0/ReBarUEFI/files/9963272/Z170WS_test.zip)

xCuri0, 2022-11-11:
@LatiosAeon Try using UEFIPatch with the patches.txt from the UEFIPatch folder on it.

LatiosAeon, 2022-11-12:
Ok, So I managed to figure out how to use UEFIPatch being as simple as drag and drop... and I've now only applied the patches.txt from the main file here and now its working, got the full 16GB BAR on the ARC A770, also done the same method to an ASUS Z170 Sabertooth MK1 and it now also has working ReBAR

closed 2022-11-13 by xCuri0

Rasmus715, 2023-11-10:
I understand it's been a while but maybe could you shared a modded Sabertooth bios, please?