xCuri0/ReBarUEFI #13: Asus X99-A II / 2 Bars / ReBar Still Inactive

Mak3rde, 2022-11-09:
**System**
 - Motherboard: Asus X99A-2 (X99-A II)
 - BIOS Version: 2101 (type: UEFI)
 - GPU: PowerColor AMD Radeon RX 6950 XT
 - Operating System: Windows 10 Pro 64-bit (10.0, Build 19045) (19041.vb_release.191206-1406)

**Description**
I inserted the ReBarDxe.ffs in the efi, and verified that it's actually where it should be.
The system booted without problems,
Set >4g active
Set CSM to Disabled

[image]

Gpuz shows reBar is still set as inactive.
and also 
Bar0 = 256 MB
BAR1 = 2MB

ReBarState.exe ->Failed to write ReBarState UEFI variable

[image]

xCuri0, 2022-11-09:
Hi the problem here is with NvramSmi not allowing EFI variables to be written. Other users were able to fix this by downgrading NvramSmi or BIOS with one from 2014.

A patch needs to be developed for your NvramSmi to remove the variable name restriction. I do think those on Hackintosh forums have already figured it out.

Mak3rde, 2022-11-10:
Thx for that info, i will look into it until i got it working and post a how to here.

Mak3rde, 2022-11-10:
Got it working partially, need to patch the above 1gb restriction it seems

# NVRAM whitelist unlock
54B070F3-9EB8-47CC-ADAF-39029C853CBB 10 P:0F84B300000041F6:90E9B300000041F6

Mak3rde, 2022-11-11:
@xCuri0 got it working.

edited initial post.

xCuri0, 2022-11-11:
your patch for PciBus has been added in commit [16712446c800707822dc4e37964abe0c85fc6e50](https://github.com/xCuri0/ReBarUEFI/commit/16712446c800707822dc4e37964abe0c85fc6e50)

closed 2022-11-11 by xCuri0

Mak3rde, 2022-11-11:
@xCuri0  nice!

Maybe you should add the info that this patch is specifically for the Asus X99a-II because the other patch that already was in the patch.txt and had Haswell written to it wasn't working on my bios, so I would assume thy my patch won't work on other platforms too. Maybe it's working for other Asus x99 boards, but I could test that.

xCuri0, 2022-11-11:
@Mak3rde actually it does seem to apply to an ASUS Z170-WS board with similar issue. waiting to see if it works for them

dawenhaozx, 2023-04-14:
Thank you very much for your efforts to fix this issue.
I tried it but it didn't work.

setup_var_cv IntelSetup 0x10E2 0x04 0x00(0x00~0x07)

0x11339 	VarStore: VarStoreId: 0x1 [EC87D643-EBA4-4BB5-A1E5-3F3E36B20DA9], Size: 0x17D6, Name: IntelSetup {24 21 43 D6 87 EC A4 EB B5 4B A1 E5 3F 3E 36 B2 0D A9 01 00 D6 17 49 6E 74 65 6C 53 65 74 75 70 00}

0x15351 		One Of: MMIOHBase, VarStoreInfo (VarOffset/VarName): 0x10E2, VarStore: 0x1, QuestionId: 0x1A8, Size: 4, Min: 0x0, Max 0x7, Step: 0x0 {05 9A 0C 00 0D 00 A8 01 01 00 E2 10 10 12 00 00 00 00 07 00 00 00 00 00 00 00}
0x1536B 			One Of Option: 56T, Value (32 bit): 0x0 (default) {09 0A 14 11 32 02 00 00 00 00}
0x15375 			One Of Option: 40T, Value (32 bit): 0x1 {09 0A 13 11 02 02 01 00 00 00}
0x1537F 			One Of Option: 24T, Value (32 bit): 0x2 {09 0A 12 11 02 02 02 00 00 00}
0x15389 			One Of Option: 16T, Value (32 bit): 0x3 {09 0A 11 11 02 02 03 00 00 00}
0x15393 			One Of Option: 12T, Value (32 bit): 0x4 {09 0A 10 11 02 02 04 00 00 00}
0x1539D 			One Of Option: 4T, Value (32 bit): 0x5 {09 0A 0F 11 02 02 05 00 00 00}
0x153A7 			One Of Option: 2T, Value (32 bit): 0x6 {09 0A 0E 11 02 02 06 00 00 00}
0x153B1 			One Of Option: 1T, Value (32 bit): 0x7 {09 0A 0D 11 02 02 07 00 00 00}

xCuri0, 2023-04-14:
@dawenhaozx have you verified that the address is actually changing in Windows

Because iirc there are multiple entries for each PCIe port.

Also there's an option to change the MMIOH region size (default 256GB) which you can try changing.

revelator, 2025-01-21:
decided to try this out on my asus x99-A II but i cannot flash the bios neither by bios flashback nor ezflash.
bios flashback just blinks once then stops and it does work when i try to flash the original bios.
ezflash throws an error :/
even tried the built in flasher in the bios but it complains saying that it is not an uefi bios.

my guess is im one of the unlucky ones that have an encrypted bios that checks everything ?.
my question would be if downgrading the bios via flashback to an older version would help.

medenijazbec, 2025-10-18:
seems a bit late, dont know if you have already solved it, but this dude did the same thing here, but instead of flashing .cap he just simply extracted it into a .rom
https://forum.level1techs.com/t/x99-e-10g-ws-post-issues-3x-p40-llm-inference-rig/211715

revelator, 2025-10-18:
tbh none of my two x99 boards would accept any of the prebuilt caps made for them. i also have an x99 deluxe which i recently delegated to developer work. the A II was sold and upgraded with a rog strix x570 board and 3700x cpu which supports rebar from the get go. sadly the old cabinet i use for the deluxe has no space for my 2080 ti which also needs hacks to support rebar so i had to use my rtx 3070 in that one instead.