xCuri0/ReBarUEFI #14: Gigabyte Z97X-Gaming 3 hidden 4G Decoding

BartekJablonski, 2022-11-11:
MOBO: Gigabyte Z97X-Gaming 3
BIOS: F8e (not available on GB page, i got it from Support. This bios add support for TPM 2.0) 
CPU: Intel Xeon E3-1285L v4
GPU: RX 6900XT
CSM disabled, SecureBoot enabled, TPM 2.0 Enabled

Im curiousto check if Re-Bar can work on my motherboard using this instruction: 
https://github.com/xCuri0/ReBarUEFI/wiki/Enabling-hidden-4G-decoding

Im stuck on 4 step. I find 4G Decod using UEFITOOL, extracted body, next i used [IRFExtractor_0.3.6] to save output file.
Output file doesn't contain "4G Decod".

I attached original bios file and txt output.

[Z97XGAMING3.zip](https://github.com/xCuri0/ReBarUEFI/files/9989432/Z97XGAMING3.zip)
[Section_Freeform_subtype_GUID_97E409E6-4CC1-11D9-81F6-000000000000_Setup_Setup_body IFR.txt](https://github.com/xCuri0/ReBarUEFI/files/9989436/Section_Freeform_subtype_GUID_97E409E6-4CC1-11D9-81F6-000000000000_Setup_Setup_body.IFR.txt)

sidex15, 2022-11-12:
seems like the it is not included in the setup configuration in the bios... and only counted as a Bios String...

BartekJablonski, 2022-11-12:
I checked older bioses. F8d and F7 do not include 4G Decod as a string. F6 have one mention, but still not included in txt output.

magicalz, 2022-12-21:
My Gigabyte B85M-D3H(rev.1.2) has the same issue.

magicalz, 2022-12-25:
Update: when I locate the offset in hex-editor which has been found in UEFITool, I can see a string "Above 4G Decoding. Enables or Disables 64bit capable Devices to be Decoded in Above 4G Address Space".

It looks like just a description string but not an configuration.

[image]
[image]

[97E409E6-4CC1-11D9-81F6-000000000000_Setup_Setup_body.zip](https://github.com/xCuri0/ReBarUEFI/files/10299543/97E409E6-4CC1-11D9-81F6-000000000000_Setup_Setup_body.zip)

BartekJablonski, 2023-02-15:
I did the same thing. Added FFS module unlock ReBar <= 1GB without 4G Decoding.
[image]

BartekJablonski, 2023-02-16:
Unfortunately, I have not been able to get ReBar to run above 1GB. Nevertheless I have a higher minimum FPS.

NoUserNameForYou, 2023-03-04:
A guy on winraid has added options from strings into his settings 

https://winraid.level1techs.com/t/add-option-in-ami-uefi-bios-menu-how-to/31609/15

Perhaps we can do the same for 4g decode?

xCuri0, 2023-03-06:
@NoUserNameForYou Can you try this in grub-mod-setup_var

1. ```setup_var 0x3 0x0```
2. Reboot
3. ```setup_var 0x3 0x1```
4. Reboot and see if you have 4G decode working

NoUserNameForYou, 2023-03-07:
@xCrui0 I'm on asus h81m-d rev1 and enabled 1GB rebar already. My 4g decoding strings are in the bios but when I extract the text file, it's not in there. 

You said adding 4g decode in could be possible via asm editing but you weren't planning on doing it. Do you still want me to try the setup_var commands you mentioned?

xCuri0, 2023-03-07:
@NoUserNameForYou don't really think anything more can be tried other than swapping pcihostbridge or maybe pcibus modules with a similar motherboard that supports 4g decoding.

patching is complicated because it would require addition of code which isn't easy

NoUserNameForYou, 2023-03-07:
@xCuri0 I'm willing to experiment. I can come back with a ch341a. 

Let's see, https://findhard.ru/en/motherboards/model?id=2026&m=asus-h81-gamer 
Just tried this bios and it has 4g decoding in its .txt file after ifrextract. 

How do we proceed after this. H81gamer looks a bit different when it comes to ram slots (4 vs my 2) and usb sockets but there shouldn't be that much difference.

NoUserNameForYou, 2023-03-09:
@xCuri0 Well mate?

BartekJablonski, 2023-03-18:
The Grub mod setup var method did not unlock 4G Decoding. I also discovered a strange bug with Unreal Engine games. When I have ReBar enabled for 1GB, games on UE can only allocate 1GB of VRAM, causing performance to drop dramatically. Games on otyher engine, e.g: Dying Light 2 gain performance and the game can allocate more than 6+GB VRAM.

xCuri0, 2023-03-18:
@BartekJablonski I've seen others report similar issues on Polaris GPUs but didn't know it happened on RDNA too. Also 2GB does have this issue too just that it will happen at 2GB VRAM

BartekJablonski, 2023-03-18:
It was a fun adventure to squeeze the last sweat out of that Broadwell. I am glad that I was able to participate in this :)

xCuri0, 2023-03-18:
@BartekJablonski You mention 256mb causes the issue, but isn't that the default size no rebar ?

BartekJablonski, 2023-03-18:
I'll straighten the whole thing out a bit
In ReBar 9 and 10 AMD Settings says that ReBar is supported and I can turn it off. Disabling it will allocate up to 256MB, although GPU-Z will not change the BAR0 and BAR1 values
ReBar State: 0 (disabled). AMD Settings: ReBar Unsupported. UE Games can allocate all memory.
ReBar State: 8 (256MB). AMD Settings: ReBar Unsupported. UE Games can allocate all memory.
ReBar State: 9 (512MB). AMD Settings: ReBar Supported. UE Games can allocate 512MB Only. 
ReBar State: 10 (1024MB). AMD Settings: ReBar Supported. UE Games can allocate 1024MB Only. 
Also check out the pics of what it looks like.
https://imgur.com/a/Zdr5n52

BartekJablonski, 2023-03-27:
Source 2 games is also affected to this VRAM allocation bug.

BartekJablonski, 2023-05-07:
Unity Engine game are also affected to this VRAM allocation bug, but i find out workaround for this. DXVK fixed it. Contraband Police (Unity Game) had this bug. I tried DXVK and now game is can utilize more than 1GB VRAM :p

BartekJablonski, 2023-05-07:
Yeah, DXVK also fixed allocation bug in Sea of Thieves (UE4 game)

xCuri0, 2023-05-07:
@BartekJablonski dxvk often reduces performance tho, we are enabling rebar to get more fps

BartekJablonski, 2023-05-07:
I know that, but it depends on hardware. Obviously im cpu bottlenecked and take f.e Contraband Police.
Native DX11 65FPS -> DXVK 85FPS -> DXVK+ReBar 1GB 92FPS
Sea of Thieves Native VS DXVK+ReBar 1GB gave very similar results. 
I wanted to share this finding

xYuuna, 2023-08-21:
[DSDT.aml.zip](https://github.com/xCuri0/ReBarUEFI/files/12395690/DSDT.aml.zip)
Hi Everyone, need some help with DSDT. 
I am getting errors after doing `iasl DSDTMod.dsl`,  initially it was just a Syntax error,  but after I the removed ARG0, now I am getting 1 error, but a bunch of warnings and remarks. I'm not sure If I did anything wrong or how about to fixing this. I've attached the files. If anyone can help me or guide me through this is appreciated. 

Thanks!

CNduan, 2023-10-06:
试试 UEFITool 替换附件中 pcibus模块 我的主板用的是同样技嘉品牌Z97-HD3 2.1 可以打开1G 以上 Bar 这个模块是我从华擎Z97-EXTREME 中已启用 Bar 中提取 这似乎是技嘉主板PCIBUS模块的问题 地址是：https://winraid.level1techs.com/t/request-resizable-bar-added-to-asrock-z97-extreme-6-v2-80a/91036/2
[PCIBUS.zip](https://github.com/xCuri0/ReBarUEFI/files/12826034/PCIBUS.zip)

xCuri0, 2023-10-23:
@xYuuna go to the lines mentioned and remove Arg0, it says this at the bottom of the DSDT editing wiki page.

NoUserNameForYou, 2023-10-27:
I'm too much of a brainlet to make any sense out of this but thank you for your discoveries.

BTW, sorry I had to delete my discord account.

xYuuna, 2023-10-31:
@xCuri0 thanks for that write up, really appreciate you taking the time. Though this is way beyond my level, I don't even know where to begin. Hopefully someone tries this and figures out an easier way to patch/add this :)

jakeman2048, 2024-03-03:
MOBO: Gigabyte Z97X-UD5H-BK
BIOS: F8
CPU: Intel i5-4670K
GPU: Intel A380

Stock DSDT section:
`
                If ((Local1 >= 0x07D9))
                {
                    If ((M64L == Zero))
                    {
                        CreateDWordField (BUF0, \_SB.PCI0._Y1E._LEN, MSLN)  // _LEN: Length
                        MSLN = Zero
                    }
                    Else
                    {
                        CreateQWordField (BUF0, \_SB.PCI0._Y1E._LEN, M2LN)  // _LEN: Length
                        CreateQWordField (BUF0, \_SB.PCI0._Y1E._MIN, M2MN)  // _MIN: Minimum Base Address
                        CreateQWordField (BUF0, \_SB.PCI0._Y1E._MAX, M2MX)  // _MAX: Maximum Base Address
                        M2LN = M64L /* External reference */
                        M2MN = M64B /* External reference */
                        M2MX = ((M2MN + M2LN) - One)
                    }
                }
`

My attempt at bypassing:
`                If ((Local1 >= 0x07D9))
                {
                        CreateQWordField (BUF0, \_SB.PCI0._Y1E._LEN, M2LN)  // _LEN: Length
                        CreateQWordField (BUF0, \_SB.PCI0._Y1E._MIN, M2MN)  // _MIN: Minimum Base Address
                        CreateQWordField (BUF0, \_SB.PCI0._Y1E._MAX, M2MX)  // _MAX: Maximum Base Address

                        // 39-bit for Haswell and up
                        M2MX = 0x8000000000 - One
                        If ((TUUD >= 0x1000))
                        {
                            M2MN = (TUUD << 0x14)
                        }
                        Else
                        {
                            M2MN = 0x100000000
                        }
                }
`

I'm also unable to find the 4G Decode option with UEFITool. The stock DSDT seems to account for 4G Decoding, but when I bypass that M64L conditional I get the expanded root bus in the OS, but anything above 1G resizebar won't POST. I'm guessing that M64L bit needs to be actually toggled somehow, instead of bypassed in order for the PCIBus to work?

Joris-truly, 2024-03-24:
MOBO: GA-X99-Gaming 5
BIOS F23c
CPU: Intel i7-5930K
GPU: RTX 3070

Same issue, stuck on finding the '4G Decod in txt' step.
I appreciate the support and write-up above, @xCuri0, but it's a bit too impenetrably written for non-programmers. I'm unsure where to start :)

I've added my extracted text. Perhaps it could serve as an example of how to trace the '4g decod' flag?

[Section_PE32_image_Setup_Setup_body.efi.0.0.en-US.ifr.txt](https://github.com/xCuri0/ReBarUEFI/files/14735627/Section_PE32_image_Setup_Setup_body.efi.0.0.en-US.ifr.txt)

ziddey, 2024-04-02:
@jakeman2048 aren't you then missing `M2LN = ((M2MX - M2MN) + One)`

Also, after removing the stray Arg0, I still can't compile:

    PS D:\Work> D:\Programs\iasl.exe -ve .\DSDT.dsl

    Intel ACPI Component Architecture
    ASL+ Optimizing Compiler/Disassembler version 20230628
    Copyright (c) 2000 - 2023 Intel Corporation

    ./DSDT.dsl   9112:         Device (SDMA)
    Error    6141 -   Missing dependency ^  (Device object requires a _HID or _ADR)

    ./DSDT.dsl   9161:         Device (I2C0)
    Error    6141 -   Missing dependency ^  (Device object requires a _HID or _ADR)

    ./DSDT.dsl   9282:         Device (I2C1)
    Error    6141 -   Missing dependency ^  (Device object requires a _HID or _ADR)

    ./DSDT.dsl   9400:         Device (SPI0)
    Error    6141 -   Missing dependency ^  (Device object requires a _HID or _ADR)

    ./DSDT.dsl   9481:         Device (SPI1)
    Error    6141 -   Missing dependency ^  (Device object requires a _HID or _ADR)

    ./DSDT.dsl   9574:         Device (UA00)
    Error    6141 -   Missing dependency ^  (Device object requires a _HID or _ADR)

    ./DSDT.dsl   9655:         Device (UA01)
    Error    6141 -   Missing dependency ^  (Device object requires a _HID or _ADR)

    ./DSDT.dsl   9748:         Device (SDHC)
    Error    6141 -   Missing dependency ^  (Device object requires a _HID or _ADR)

    ASL Input:     ./DSDT.dsl -  596415 bytes   7624 keywords      0 source lines

    Compilation failed. 8 Errors, 134 Warnings, 169 Remarks
    No AML files were generated due to compiler error(s)

Is it safe to force with -f? Or should I be doing something with these:

    Scope (_SB.PCI0)
    {
        Device (SDMA)
        {
        }
    }

ZLHuo, 2024-05-02:
> 试试 UEFITool 替换附件中 pcibus模块 我的主板用的是同样技嘉品牌Z97-HD3 2.1 可以打开1G 以上 Bar 这个模块是我从华擎Z97-EXTREME 中已启用 Bar 中提取 这似乎是技嘉主板PCIBUS模块的问题 [...]

I've attempted to replace the pcibus module in the Z97-HD3 firmware; however, I'm still unable to set BARs larger than 1G. Could you provide a more detailed explanation on how to successfully implement this?

xCuri0, 2024-05-02:
@ZLHuo there a bunch of other modules too not just pcibus

i think pcihostbridge and some sa related module.

i dont think it will be easy to implement tbh

ZLHuo, 2024-05-02:
@xCuri0  
I understand, there's a lot of work to be done.
Achieving a BAR size of 1 GB is truly satisfying. Thank you for your exceptional project!

SaulGoodman1337, 2024-05-14:
> @ZLHuo there a bunch of other modules too not just pcibus
> 
> i think pcihostbridge and some sa related module.
> 
> i dont think it will be easy to implement tbh

Hi,
I have a GA-Z97N-Gaming 5 here that I would like to use. I can't quite get my head around the patching here yet. 
Can you tell me roughly whether it works with my board? I was able to add the module to the bios, but I've already failed with the 4G. 
I can't find anything with UEFITool. Will DSDT patching help me here? 
I am not planning to use windows, only Linux.

xCuri0, 2024-05-14:
if you're using linux you don't need to even flash rebardxe, there's a linux section on dsdt patching wiki just follow that

ivan49k, 2024-06-01:
@xCuri0 Can you tell me about method directly manipulating PCH and PCI setup variables to enable 4G decoding?

xCuri0, 2024-06-01:
@ivan49k someone tried and it doesn't work, the uefi has been built without the code for 4g decoding.

Shranobxtha, 2024-11-27:
[image]
@xCuri0 I am able to reach till here by following the guide, I'm using the motherboard GA-H81M-DS2 rev 3.0, couldn't enable Above 4G decode through grub tho sadly.

geotaxho, 2024-12-08:
I have the same mb and result as @Shranobxtha. However i found above 4g in grub but changing it seems do no effect in gpu-z report (tried both setupvar/setupvar3).
I also found somebody else with a same mb who successfuly enabling above 4g decoding, but i dont understand what he mean with replacing the pcibus and the PciHostBridge.
Can anybody help? i attached grub txt and my patched bios in case.

https://github.com/xCuri0/ReBarUEFI/issues/231
[5.bin.1.0.en-US.ifr.txt](https://github.com/user-attachments/files/18054067/5.bin.1.0.en-US.ifr.txt)
[REBAR.zip](https://github.com/user-attachments/files/18054068/REBAR.zip)

Shranobxtha, 2024-12-12:
@geotaxho I got the same result again with a different motherboard this time with an MSI-H81M-P33, I swapped my board with this cause it lets me set all core multipler.

GTX1080Ti11G, 2025-01-08:
Just thought I'd chime in here as I'm on a haswell platform as well. I own a Gigabte Z87X-OC. For me I had 4G Decod visible in the body, it was called Memory Remap in the extracted text file at 0x2FE offset. It was also set to enable by default.

I compiled the DSDL using the steps, and it compiled successfully. My bios has both the nvme & rebar dxes. I proceeded to use the UEFI patches and then flash the bios, it surprisingly worked.

Above 4G Decoding is now enabled, I don't have a rebar gpu to test yet, but if I do in the future I'll be sure to update it in the list of working motherboards.

Large Memory in device manager was not visible prior to all this.

[image]

yonexsp11, 2025-01-10:
> MOBO: Gigabyte Z97X-Gaming 3 BIOS: F8e (not available on GB page, i got it from Support. This bios add TPM 2.0) CPU: Intel Xeon E3-1285L v4 GPU: RX 6900XT CSM disabled, SecureBoot enabled, TPM 2.0 Enabled
> 
> Im curious to check if Re-Bar can work on my motherboard using this instruction: https://github.com/xCuri0/ReBarUEFI/wiki/Enabling-hidden-4G-decoding
> 
> Im stuck on 4 step. I find 4G Decod using UEFITOOL, extracted body, next i used [IRFExtractor_0.3.6] to save output file. Output file doesn't contain "4G Decod".
> 
> I attached original bios file and txt output.
> 
> [Z97XGAMING3.zip](https://github.com/xCuri0/ReBarUEFI/files/9989432/Z97XGAMING3.zip) [Section_Freeform_subtype_GUID_97E409E6-4CC1-11D9-81F6-000000000000_Setup_Setup_body IFR.txt](https://github.com/xCuri0/ReBarUEFI/files/9989436/Section_Freeform_subtype_GUID_97E409E6-4CC1-11D9-81F6-000000000000_Setup_Setup_body.IFR.txt)

Hi,

Did you ever figure it out? I have th esame board and am anbout to junk it as I can;t get it to work.

It keeps reverting back to th ebackup BIOS file after not being able to boot

GTX1080Ti11G, 2025-01-11:
@yonexsp11 I took the bios in the zip of the first post F8e, the 4G is located at the offset 0x309. It is enabled by default, you can try the setup_var offset method to see if it works, and whether you get Large Memory enabled in Control panel. 
[image]

Else you can try to flash the bios I attached, I added the UEFI patches, patched the DSDT and I added the ReBar dxe to it. This is onto the same bios shared in the OP.
[modZ97XGaming3.zip](https://github.com/user-attachments/files/18386354/modZ97XGaming3.zip)

yonexsp11, 2025-01-12:
Thx!

I will give it a try & let you know.

The challenge continues!

Cheers, Ken./

yonexsp11, 2025-01-15:
Alas that bricked my MB and it reverted back to F6 after failing to boot,
even without the Video card installed in the machine.

I think the modded FM that was posted that you used as a base is messed up.

The stock F8e is probably a better bet.

Cheers, Ken./

GTX1080Ti11G, 2025-01-15:
Sorry to hear that, As per my understanding, there is no stock F8e. The one posted in the OP is a modified F8d with just TPM enabled. I can try working off the F8d from the motherboard's page if you'd like to try again.

yonexsp11, 2025-01-15:
Yes, please!

GTX1080Ti11G, 2025-01-15:
I'm a bit confused with the F8d bios. It has memory remap at 0x302. 
[image]

The DSDT shown in the body is different than the code as per the guide. The code segment is within an IF block so I am confused how to modify it. I tried adding it only to the first block and got 14 errors. The 1st edit is already present as Y1E instead of YAF so its off by 2 bytes only. There are too many duplicate checks for Local1 >= 0x07D9, maybe the bios already has above 4G enabled.
[image]

Also F8e is not on the motherboard support page, and as per user in the OP, its just got TPM. We can try the F8e you might have. You can try the grub-setup method with 0x302 offset however its enabled by default.

yonexsp11, 2025-01-15:
Ok, I also got that value with version D.

Perhaps just the version e, the first link in that page.

yonexsp11, 2025-01-15:
https://winraid.level1techs.com/uploads/short-url/9XOmzxAQaoEB0q6pdu366UT2cVe.zip

yonexsp11, 2025-01-17:
> I'm a bit confused with the F8d bios. It has memory remap at 0x302. [image]
>
> The DSDT shown in the body is different than the code as per the guide. The code segment is within an IF block so I am [...]

How did oyu make out with that basic verison of F8e? I think if like the OP of this thread, we can get the FFS module to stick at least that woudl be a win :)

GTX1080Ti11G, 2025-01-24:
@edisionnano You might need to check and patch your DSDT.
@yonexsp11 I didn't get time to check F8e, but as per my understanding F8d with just the UEFI patches and ReBar Dxe should work. If its not, then perhaps Z97 needs its own DSDT patch tutorial.

edisionnano, 2025-01-24:
> [@edisionnano](https://github.com/edisionnano) You might need to check and patch your DSDT. [@yonexsp11](https://github.com/yonexsp11) I didn't get time to check F8e, but as per my understanding F8d [...]

Did you have to patch your DSDT? What exactly did you do, can you describe in detail?

xCuri0, 2025-02-28:
@paul-tao no one figured that out yet

yuunayuu, 2025-07-23:
I was able to get a Custom DSDT working in Linux. 
Asus Z87-A Mobo + RX 580 8GB. (Motheboard doesn't have 'Above 4G Decoding')
I am booting the dsl file via Grub.

Here is my DSDT file in case anyone wants to check it:
[DSDT.dsl.gz](https://github.com/user-attachments/files/21393743/DSDT.dsl.gz)

[image]

SaulGoodman1337, 2026-08-18:
**Working ReBAR with Gigabyte Z97N-Gaming 5 F5 BIOS - made with the help of ChatGPT**

I modified the original **Gigabyte Z97N-Gaming 5 F5 BIOS** to add working Resizable BAR support.

Changes compared to the stock F5 BIOS:

* Added `ReBarDxe.ffs`
* Applied the standard ReBarUEFI patches
* Applied `HswAbove4G.txt` for Haswell, including the 39-bit / 512 GB MMIO fixes
* Applied the Z97 `BdwUSB3.txt` workaround
* Added a small board-specific `PciHostBridge` patch to force the existing Above-4G MMIO allocation path to run

The board-specific patch bypasses the firmware policy check that was preventing `PciHostBridge` from registering the 64-bit MMIO region above 4 GB.

Before the patch, MMIOPrint showed the large region above RAM as `TYPE: 0` (non-existent). After the patch, it is registered as `TYPE: 3` (MMIO).

With the patched BIOS, Linux now sees the Above-4G root bus window directly from ACPI/firmware, without `pci=nocrs` or `pci=realloc`, and my RX 6950 XT is running with a full **16 GB Resizable BAR**.

Relevant `PciHostBridge` patch:

```text
80 78 20 00 0F 84 59 FF FF FF
->
80 78 20 00 90 90 90 90 90 90
```

[Z97NG5_F5_ReBar_ForceAbove4Gv2.zip](https://github.com/user-attachments/files/31183090/Z97NG5_F5_ReBar_ForceAbove4Gv2.zip)
[Z97NG5_F5_ReBar_patch_notes.txt](https://github.com/user-attachments/files/31182900/Z97NG5_F5_ReBar_patch_notes.txt)