xCuri0/ReBarUEFI #15: ReBar on my ga-z97x Gaming5

LordAdo96, 2022-11-11:
**System**
 - Motherboard:  ga-z97x Gaming5
 - BIOS Version:  F7
 - GPU:   RX 580
[image]

So i managed to mod bios and flash but i can only have 1024mb.. if i go 2gb or 4 or 8 my pc wont boot.. is there any fix or im stuck with 1gb?

xCuri0, 2022-11-11:
4G Decoding on and CSM off ?

UEFIPatch applied ?

LordAdo96, 2022-11-11:
> 

CSM is disabled but 4g Decoding i cant find.

xCuri0, 2022-11-11:
Gigabyte Z97 boards don't have the 4G decoding option even in hidden settings. I'm sure it can be patched to enable it tho.

LordAdo96, 2022-11-11:
Hmmm i there any way i can do it or is it risky.. or myb to stay at 1024?

xCuri0, 2022-11-11:
@LordAdo96 patch to fix it doesn't exist yet. I had a 12% FPS gain in Cyberpunk 2077 with 2GB BAR size on RX 580 8GB I'm sure 1GB gives an increase too.

LordAdo96, 2022-11-11:
Yeah you right.. i think i got as far as i can go.. i mean im new in this whole thing. Thank you very much!

cursemex, 2022-11-24:
1. Download [UEFIPatch and UEFITools](https://github.com/LongSoft/UEFITool/releases/tag/0.28.0) and put them in a folder together.

2. Download your bios from [Gigabyte's website. ](https://www.gigabyte.com/Motherboard/GA-Z97X-Gaming-5-rev-10/support#support-dl-bios)

3. Open up the bios zip and extract the Z97XG7.F8(In this example), and put in the same folder as UEFITool and UEFIPatch

Use these 2 patches from the [patches.txt](https://github.com/xCuri0/ReBarUEFI/blob/master/UEFIPatch/patches.txt) (Create a patches.txt text file in the same directory and add the patches. Make sure they don't have space in the end of the code part before you save it.)

> # PciHostBridge | Replace 8-16GB MMIO region with complete use of 64GB address space (Haswell). MAY REQUIRE DSDT MODIFICATION
`8D6756B9-E55E-4D6A-A3A5-5E4D72DDF772 10 P:080000004823C1483BC3770C48BE0000000004000000EB7A48BB000000000C000000483BC3770C48BE0000000002000000EB5F48BB000000000E000000483BC3770C48BE0000000001000000EB4448BB000000000F000000483BC37707BE00000080EB2E48BB000000800F000000483BC37707BE00000040EB1848BB000000C00F000000483BC30F87A4FEFFFFBE00000020:010000004821C84839D8480F47D848BE00000000100000004829DE9090909090909066909066906690909090909090909066906690909090909090909066909066906690909090909090909066906690909090909090909066909066909090909090669066909090909090909090669090669090909090906690669090909090909090906690909090909090909090909090`
> # PciBus | Remove <16GB BAR size limit (Ivy Bridge/Haswell)
`3C1DE39F-D207-408A-AACC-731CFB7F1DD7 10 P:B8FFFFFFFF030000004C3B:B8FFFFFFFFFFFFFF004C3B`

5. open up Command Prompt with Admin privilege(right click, run as administrator), you can use the taskbar search bar to find it.

6. write "cd [UEFIpatch directory]"

7. run UEFI patch by running this command `uefipatch.exe Z97XG7.F8 Gigabyte.txt -o Z97XG7Patched.F8` 
This will generate a patched bios named "Z97XG7Patched.F8"

8. insert rebardxe.ffs using UEFITool

done

cursemex, 2022-11-24:
You may need to patch your DSDT. Unfortunately, I have no knowledge in the matter as of yet.

LordAdo96, 2022-11-24:
aa ok. thank you for help mate..

xCuri0, 2022-11-24:
> Hmm.. i try it but i only managed get 1024mb if i go 2gb or more just black screen

you need 4g decode for sizes larger than 1gb to work. but you still do get an fps increase even with 1gb

xCuri0, 2023-03-04:
Duplicate of #14

closed 2023-03-04 by xCuri0