xCuri0/ReBarUEFI #16: Laptop Support

5l1v3r1, 2022-11-25:
Great project, but i want to ask if it is possible to apply the patches for laptop bioses like asus n550jk-gtx850m or lenovo y520-gtx1060?

xCuri0, 2022-11-25:
The patches developed are all for AMI UEFI which laptops don't use. But there's no reason the module shouldn't work on a laptop if it has 4G decoding and large BAR support.

SEt-t, 2022-11-25:
@5l1v3r1 while the motherboard's BIOS can be patched, the BIOS on gtx850m/gtx1060 can not so far. So, you won't get the working ReBAR with just this project.

5l1v3r1, 2022-11-28:
> @5l1v3r1 while the motherboard's BIOS can be patched, the BIOS on gtx850m/gtx1060 can not so far. So, you won't get the working ReBAR with just this project.

> The patches developed are all for AMI UEFI which laptops don't use usually. But there's no reason the module shouldn't work on a laptop if it has 4G decoding and large BAR support.

Oh i see then we need to patch the vbios'es as well ? But what kind of patching for the vbios? Because as far as i know the vbioses can be found at https://www.techpowerup.com/vgabios/ link.

xCuri0, 2022-11-28:
@5l1v3r1 you'd need some advanced knowledge to be able to patch VBIOS and also Pascal (GTX 1060) VBIOS is encrypted atleast on desktop so you can't edit that. Though some people say laptops have it unencrypted so might be able to be edited.

The oldest Nvidia GPUs that might be able to have it hacked in are probably GTX 16 series and RTX 20 series. They have the Resizable BAR capability but with maximum supported size of 256MB. 

You can override the supported sizes list through editing DXE driver or Linux kernel source which will allow you to set a larger BAR size. I doubt it will work but no one has tested it yet.

Linux kernel has an override for the Sapphire RX 5600 XT Pulse which comes with a buggy VBIOS which incorrectly lists the maximum size as 2GB. [A Linux Fix Is On The Way For Some GPUs Having AMD Smart Access Memory Issue](https://www.phoronix.com/news/Linux-Quirk-SAM-GPU-Quirk)

The same code is used in the DXE driver since it's basically code from Linux kernel adapted for UEFI. [ReBar.c line 164](https://github.com/xCuri0/ReBarUEFI/blob/master/ReBarDxe/ReBar.c#L164)

5l1v3r1, 2023-03-03:
> 

Have you bypassed igpu? Because i realized that by the help of minidisplay port i could bypass the optimus and could only use nvidia gpu and applied the patches here accordingly and voila! it worked.

psyborg55, 2023-03-15:
Are there any restrictions on BIOS version?
I have AMI Aptio V in laptop. Above 4G setting is visible when opened BIOS dump in AMIBCP, but i don't see it in actual BIOS

xCuri0, 2023-03-15:
@psyborg55 No it should work on any UEFI BIOS you don't even need 4G decode if you are fine with being limited to smaller (often 1GB) BAR sizes. You still do get a performance gain though not as much

Have you tried the enabling hidden 4G decode guide in the wiki ? I've used the same method to enable hidden settings on laptops (not 4G decode but will be same)

psyborg55, 2023-03-15:
No, I'll check that guide. Hidden settings were made visible by modder from win-raid few years ago, i can change them on machine. Just this PCI Subsystem page that contain 4G setting is not visible. It looks like it is broken or removed from BIOS and the page shown in AMIBCP being leftover as it has no settings to modify (nothing for failsafe or optimal config)
[image]

psyborg55, 2023-03-15:
Went through wiki guide and 8 pages on oc forum.
Few more questions:

1. No way to run this with legacy BIOS? I have WIN10 UEFI install but is on HDD, win7 legacy on SSD is what i use most of the time
2. There are mentions of CPU in forums. Does the CPU need to support the feature as well? My system is FX9830P with RX460

xCuri0, 2023-03-15:
@psyborg55 

1. legacy may work but on alot of systems you will just end up with black screen if you try to enable 4g decode without disabling csm

2. there are no cpu requirements afaik

psyborg55, 2023-04-15:
Another shot at this:
Clean win10 uefi/ csm disabled/ amernime 22.10 driver - there isn't anything in adrenalin software SAM related, only some new feature smartshifx max disabled (not available)

Module was inserted in main bios and flashed, rebarstate shows value is preserved across restarts - tried 11,32 and 10

Not sure if PCI bus driver that is installed after AMD driver and restart has any effect on this. On amernime it doesn't update automatically and windows doesn't prompt for restart due to hardware change, i updated it manually, still no changes in GPU-Z

psyborg55, 2023-04-15:
Tried to enable 4G decoding, system crashed and restarted.
Then booted ubuntu (legacy install only) to see the messages.
Looks like quite a bit of stuff needs to be changed...
[image]

xCuri0, 2023-04-15:
@psyborg55 looks like a DSDT issue

xCuri0, 2023-04-15:
@psyborg55 
> Module was inserted in main bios and flashed, rebarstate shows value is preserved across restarts - tried 11,32 and 10
>
> Not sure if PCI bus driver that is installed after AMD driver and restart has [...]

Did you check the Resizable BAR section in Advanced tab ? If the module is working and GPU supports rebar it will increase

psyborg55, 2023-04-15:
Yes, that's the only thing i check in gpu-z. Nothing changes there.
Before going to BIOS mod i was trying to get feature at least shown in adrenalin (disabled, greyed out or whatever) but never got it. Not even now after BIOS mod.

PCI varstore appears to have 6 variables, var0 and var2 are not displayed in ifr output, var2 defaults to 1 after CMOS reset, others default 0. Tried also var6 here i got different values like a0, 3d - is it a parser bug, or something is here as well?

xCuri0, 2023-04-15:
@psyborg55 have u verified that the GPU VBIOS actually supports large rebar sizes using HWInfo or lspci ?

If it does support then maybe try secure boot off ?

psyborg55, 2023-04-15:
There is no resizable bar in PCI features. It should appear there first, regardless of system BIOS Rebar support?

xCuri0, 2023-04-15:
@psyborg55 
[image]

Even without BIOS support HWInfo64 will list supported BAR sizes and GPU-Z will show the default (usually 256mb)

psyborg55, 2023-04-15:
There wasn't any listed
[image]

psyborg55, 2023-04-18:
I checked vBIOS with https://github.com/galkinvv/PolarisBiosEditor-xml

PCI config does not hold capabilities. Do i understand correctly it's either required to do sBIOS mod, or change pin straps on the board to advertise rebar capability in PCI space?

xCuri0, 2023-04-18:
@psyborg55 sorry I don't have much knowledge about VBIOS so can't answer this.

But VBIOS is typically embedded in the BIOS on laptops ?

psyborg55, 2023-04-18:
Yes, it is embedded. I am just focused now on getting a capability advertised to PCI space. Seems like it is first step having that, before trying other modifications.

I think none of the straps in the document i sent you will change it, probably need to add dedicated ROM chip for configuration change

xCuri0, 2023-04-18:
@psyborg55 
> I am just focused now on getting a capability advertised to PCI space

Isn't the ```F0_BAR_EN``` in the screenshot exactly that ?

psyborg55, 2023-04-22:
So, the strap table content is in vBIOS loc 0x70-0xF4.
33 dwords for GPIO 0-32
Zero out this table, the laptop boots and GPU works as usual.
Which means ROM straps have no effect if dedicated ROM is not present and configured.
So no rebar for most of the laptop users. Unless they could somehow patch system BIOS to set feature in pci space

xCuri0, 2023-04-22:
Some 5600XT users would be interested in trying that also. AMD messed up the stock vbios and it only supports 1GB BAR instead of the 8GB it should

psyborg55, 2023-04-22:
Some or all 5600xt are affected? If all, i don't know, maybe try to copy straps from 5700 cards, one by one or few at a time. If it is as simple as that, one of them should fix bar size

xCuri0, 2023-04-22:
@psyborg55 all 5600xt are

psyborg55, 2023-04-28:
> @psyborg55 looks like a DSDT issue from that kernel log

Do you know what changes are needed to fix it?

xCuri0, 2023-04-28:
@psyborg55 it's some error in the code for CRS which describes the PCI resources

psyborg55, 2023-04-28:
Is it this https://www.techinferno.com/index.php?/topic/5874-guide-dsdt-override-to-fix-error-12/

psyborg55, 2023-04-30:
Maybe i was wrong about rebar support on laptops.
I tried to boot with another vBIOS and got DSDT errors again, but this time i did not change any system settings.
Could be, there is another method of enabling resizable bar independent of pin/rom straps
[image]

xCuri0, 2023-04-30:
@psyborg55 Your DSDT probably doesn't work well with larger BAR that's what it looks like

psyborg55, 2023-04-30:
I don't have any experience patching DSDT. can you take a look at it, the dump is from live system with SWBRT33170.001 vBIOS. Also been busy with comparing various vbios, didn't manage to check if the tutorial i linked above would make any difference
[DSDT.zip](https://github.com/xCuri0/ReBarUEFI/files/11361285/DSDT.zip)

xCuri0, 2023-04-30:
@psyborg55 I don't really know how to do this type of DSDT patching, only know fixing PCI resources sorry.

psyborg55, 2023-04-30:
Ok, I will look elsewhere or ask on winraid. On techinferno they say linux boot with pci=noCRS should workaround the problem that happens with 4G decode. I did try that, the messages are still printed and the screen goes blank afterwards.
noCRS also didn't help with the bug that shows up with new vBIOS.

psyborg55, 2023-04-30:
it seems new vbios doesn't properly set PCI up

when GPU works lspci reports

```
   Capabilities: [270 v1] Secondary PCI Express
       LnkCtl3: LnkEquIntrruptEn- PerformEqu-
       LaneErrStat: 0
```

with this one i only got

`   Capabilities: [270 v1] #19`

which result in GPU crash at SetEngineClock

psyborg55, 2023-04-30:
I tried to apply patch from techinferno, DSDT already have Qwordmemory entries, but they were all zero.
After modification, i can see large memory appear in device manager, but all the devices are still assigned with small memory.
Enabling 4G decode still crashed system.

There are CRS1 and CRS2, but i only modified first one. Not sure when is the second one used and if i should put same memory range values in there

xCuri0, 2023-05-01:
@psyborg55 you need to fix the DSDT so that 4G decoding on works. ```pci=nocrs``` kernel option can make it work till it's fixed

psyborg55, 2023-05-01:
Ok, pci=nocrs applied, but the problem is still somewhere else:

Internal graphics crash and no output on display

```
[    0.000000] kernel: Linux version 5.19.0-32-generic (buildd@lcy02-amd64-026) (x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #33~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Jan 30 17:03:34 UTC 2 (Ubuntu 5.19.0-32.33~22.04.1-generic 5.19.17)
[    0.000000] kernel: Command line: BOOT_IMAGE=/boot/vmlinuz-5.19.0-32-generic root=UUID=5ae9d492-9826-4e10-8da4-9d8f2f5c144d ro quiet splash pci=nocrs vt.handoff=7
[    0.000000] kernel: KERNEL supported cpus:
[    0.000000] kernel:   Intel GenuineIntel
[    0.000000] kernel:   AMD AuthenticAMD
[    0.000000] kernel:   Hygon HygonGenuine
[    0.000000] kernel:   Centaur CentaurHauls
[    0.000000] kernel:   zhaoxin   Shanghai  
[    0.000000] kernel: [Firmware Info]: CPU: Re-enabling disabled Topology Extensions Support.
[    0.000000] kernel: x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] kernel: x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] kernel: x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] kernel: x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] kernel: x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] kernel: signal: max sigframe size: 1776
[    0.000000] kernel: BIOS-provided physical RAM map:
[    0.000000] kernel: BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] kernel: BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x0000000000100000-0x00000000f497afff] usable
[    0.000000] kernel: BIOS-e820: [mem 0x00000000f497b000-0x00000000f542cfff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x00000000f542d000-0x00000000f545efff] ACPI data
[    0.000000] kernel: BIOS-e820: [mem 0x00000000f545f000-0x00000000f58eefff] ACPI NVS
[    0.000000] kernel: BIOS-e820: [mem 0x00000000f58ef000-0x00000000f5beffff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x00000000f5bf0000-0x00000000f6ffffff] usable
[    0.000000] kernel: BIOS-e820: [mem 0x00000000f7000000-0x00000000fbffffff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x00000000fe200000-0x00000000fe200fff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x00000000fe600000-0x00000000fe6fffff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x00000000feb80000-0x00000000fec01fff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x00000000fedc0000-0x00000000fedc0fff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x00000000fedc2000-0x00000000fedc8fff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x00000000fee00000-0x00000000feefffff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] kernel: BIOS-e820: [mem 0x0000000100000000-0x0000000286ffffff] usable
[    0.000000] kernel: BIOS-e820: [mem 0x0000000287000000-0x0000000306ffffff] reserved
[    0.000000] kernel: NX (Execute Disable) protection: active
[    0.000000] kernel: e820: update [mem 0xea5b3018-0xea5c2e57] usable ==> usable
[    0.000000] kernel: e820: update [mem 0xea5b3018-0xea5c2e57] usable ==> usable
[    0.000000] kernel: e820: update [mem 0xea5a4018-0xea5b2057] usable ==> usable
[    0.000000] kernel: e820: update [mem 0xea5a4018-0xea5b2057] usable ==> usable
[    0.000000] kernel: e820: update [mem 0xea595018-0xea5a3c57] usable ==> usable
[    0.000000] kernel: e820: update [mem 0xea595018-0xea5a3c57] usable ==> usable
[    0.000000] kernel: extended physical RAM map:
[    0.000000] kernel: reserve setup_data: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] kernel: reserve setup_data: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] kernel: reserve setup_data: [mem 0x0000000000100000-0x00000000ea595017] usable
[    0.000000] kernel: reserve setup_data: [mem 0x00000000ea595018-0x00000000ea5a3c57] usable
[    0.000000] kernel: reserve setup_data: [mem 0x00000000ea5a3c58-0x00000000ea5a4017] usable
[    0.000000] kernel: reserve setup_data: [mem 0x00000000ea5a4018-0x00000000ea5b2057] usable
[    0.000000] kernel: reserve setup_data: [mem 0x00000000ea5b2058-0x00000000ea5b3017] usable
[    0.000000] kernel: reserve setup_data: [mem 0x00000000ea5b3018-0x00000000ea5c2e57] usable
[    0.000000] kernel: reserve setup_data: [mem 0x00000000ea5c2e58-0x00000000f497afff] usable
[    0.000000] kernel: reserve setup_data: [mem 0x00000000f497b000-0x00000000f542cfff] reserved
[    0.000000] kernel: reserve setup_data: [mem 0x00000000f542d000-0x00000000f545efff] ACPI data
[    0.000000] kernel: reserve setup_data: [mem 0x00000000f545f000-0x00000000f58eefff] ACPI NVS
[    0.000000] kernel: reserve setup_data: [mem 0x00000000f58ef000-0x00000000f5beffff] reserved
[    0.000000] kernel: reserve setup_data: [mem 0x00000000f5bf0000-0x00000000f6ffffff] usable
[    0.000000] kernel: reserve setup_data: [mem 0x00000000f7000000-0x00000000fbffffff] reserved
[... cut ...]
[   26.719954] kernel: audit: type=1400 audit(1682926773.249:9): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/NetworkManager/nm-dhcp-client.action" pid=557 comm="apparmor_parser"
[   26.720159] kernel: audit: type=1400 audit(1682926773.249:10): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/NetworkManager/nm-dhcp-helper" pid=557 comm="apparmor_parser"
[   26.720164] kernel: audit: type=1400 audit(1682926773.249:11): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/connman/scripts/dhclient-script" pid=557 comm="apparmor_parser"
```

Sid127, 2023-05-14:
I was able to add the ReBarDxe driver to my bios, as well as use ReBarState and have the value be persistent across reboots, but I have a feeling I need the DSDT patch, which I'm a bit lost on because my laptop has an InsydeH20 BIOS and not an AMI one and hence has no AmiBoardInfo module. 

I know I need the DSDT patch because of the following output
```
[sidpr@strogg build]$ sudo dmesg | grep "root bus resource"
[    1.137242] pci_bus 0000:00: root bus resource [io  0x0000-0x0cf7 window]
[    1.137246] pci_bus 0000:00: root bus resource [io  0x0d00-0xffff window]
[    1.137248] pci_bus 0000:00: root bus resource [mem 0x000a0000-0x000bffff window]
[    1.138280] pci_bus 0000:00: root bus resource [mem 0x3f800000-0xdfffffff window]
[    1.138283] pci_bus 0000:00: root bus resource [mem 0x4000000000-0x7fffffffff window]
[    1.138285] pci_bus 0000:00: root bus resource [mem 0xfc800000-0xfe7fffff window]
[    1.138287] pci_bus 0000:00: root bus resource [bus 00-fe]
```

In addition, I'm not entirely sure if this is any helpful to my device without a patch for the GPU as well, since lspci lists the nvidia GPUs BAR size as follows
```
Capabilities: [bb0 v1] Physical Resizable BAR
	BAR 0: current size: 16MB, supported: 16MB
	BAR 1: current size: 256MB, supported: 64MB 128MB 256MB
	BAR 3: current size: 32MB, supported: 32MB
```

Device details:
Acer Nitro 7 AN715-51
Core i5-9300H @ 2.4GHz (has an integrated graphics unit, UHD 630)
NVIDIA 1660Ti 6GB
Using unlocked InsydeH2O BIOS from the win-raid forums, v1.29

Any help is appreciated <3

Sid127, 2023-05-14:
I tried forcing ReBAR through the dxe driver for my 1660Ti as in the commit here, and also tried adding the same code to the linux kernel's PCI driver, but no change in supported BAR size, meaning it's definitely a VBIOS issue, which I don't know how to patch, or even if it can be done... I do have the vbios extracted from my laptop, but not much else.

Also realized I'd need the Skylake/UEFIToolsNE version of the DSDT patch, but was confused on what to do when my DSDT.dsl had a matching QWordMemory block, except it was NonCacheable instead of Cacheable and had a different identifier (Y0E instead of YAF)

xCuri0, 2023-05-15:
@Sid127 did you check if the override is getting executed in the linux kernel ? you will need to enable rebar in nvidia driver also for the kernel method to work `NVreg_EnableResizableBar`

also your laptop appears to have proper dsdt already. maybe 4g decode needs to be enabled but that's not an issue unless you use windows

Sid127, 2023-05-15:
I already have 4g decode enabled, as confirmed with
```
[sidpr@strogg ~]$ sudo lspci -v | grep "Memory.*64-bit"
	Memory at 4080000000 (64-bit, non-prefetchable) [size=16M]
	Memory at 4000000000 (64-bit, prefetchable) [size=2G]
	Memory at 4081200000 (64-bit, non-prefetchable) [size=32K]
	Memory at 4081215000 (64-bit, non-prefetchable) [size=4K]
	Memory at 4081214000 (64-bit, non-prefetchable) [size=4K]
	Memory at 54400000 (64-bit, non-prefetchable) [size=64K]
	Memory at 408120c000 (64-bit, non-prefetchable) [size=8K]
	Memory at 4081213000 (64-bit, non-prefetchable) [size=4K]
	Memory at 408120f000 (64-bit, non-prefetchable) [size=4K]
	Memory at 4081211000 (64-bit, non-prefetchable) [size=4K]
	Memory at 4081210000 (64-bit, non-prefetchable) [size=4K]
	Memory at 4081212000 (64-bit, non-prefetchable) [size=4K]
	Memory at 4081208000 (64-bit, non-prefetchable) [size=16K]
	Memory at 4081100000 (64-bit, non-prefetchable) [size=1M]
	Memory at 408120e000 (64-bit, non-prefetchable) [size=256]
	Memory at 40000000 (64-bit, prefetchable) [size=256M]
	Memory at 50000000 (64-bit, prefetchable) [size=32M]
	Memory at 4081000000 (64-bit, prefetchable) [size=256K]
	Memory at 4081040000 (64-bit, prefetchable) [size=64K]
	Memory at 54300000 (64-bit, non-prefetchable) [size=16K]
	Memory at 54200000 (64-bit, non-prefetchable) [size=256K]
	Memory at 54100000 (64-bit, non-prefetchable) [size=16K]
```

I do have the nvidia module option enabled, and while the override is being executed, pci_rebar_set_size is failing with the following error
```
[    2.295967] pciRebarSetSize called
[    2.295974] NVRM: BAR resizing failed with error `-2`.
```

xCuri0, 2023-05-15:
doesn't really make sense pci_rebar_find_pos is failing, sure it's that ?

Sid127, 2023-05-15:
not entirely sure, but a [well educated guess](https://github.com/torvalds/linux/blob/f1fcbaa18b28dec10281551dfe6ed3a3ed80e3d6/drivers/pci/pci.c#L3761)

Sid127, 2023-05-15:
okay, I was wrong, it seems to be failing in nvidia kernel module [here](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/6dd092ddb7c165fb1ec48b937fa6b33daa37f9c1/kernel-open/nvidia/nv-pci.c#L242)

Sid127, 2023-05-15:
Further poking around revealed this string of functions: [pci_resize_resource](https://github.com/torvalds/linux/blob/f1fcbaa18b28dec10281551dfe6ed3a3ed80e3d6/drivers/pci/setup-res.c#LL425C1-L476C36) calls [pci_reassign_bridge_resources](https://github.com/torvalds/linux/blob/f1fcbaa18b28dec10281551dfe6ed3a3ed80e3d6/drivers/pci/setup-bus.c#L2219), which is where the error arises from.

Exact error arises [here](https://github.com/torvalds/linux/blob/f1fcbaa18b28dec10281551dfe6ed3a3ed80e3d6/drivers/pci/setup-bus.c#L2265)

I am unsure how to continue ahead though, but it does look like it's got something to do with [add_to_list](https://github.com/torvalds/linux/blob/f1fcbaa18b28dec10281551dfe6ed3a3ed80e3d6/drivers/pci/setup-bus.c#L60)

xCuri0, 2023-05-15:
@Sid127 can you send a DSDT dump ? i think sometimes the DSDT prevents BAR from moving but it can be fixed

Sid127, 2023-05-15:
of course, [here you go](https://cloud.sloughland.dev/s/KeGtTDksK8KLbfT). Uploaded DSDT to Nextcloud because GitHub wouldn't let me attach it here.

Also, if Boot Config had to be preserved, wouldn't that error out right [here](https://github.com/torvalds/linux/blob/f1fcbaa18b28dec10281551dfe6ed3a3ed80e3d6/drivers/pci/setup-res.c#L434) though?

xCuri0, 2023-05-15:
@Sid127 does it error at `kzalloc` ? https://github.com/torvalds/linux/blob/f1fcbaa18b28dec10281551dfe6ed3a3ed80e3d6/drivers/pci/setup-bus.c#L66

Sid127, 2023-05-15:
It does not seem to error out at that kzalloc, the if block checking if `tmp` is empty is not executed. Using ReBarState to set reBAR to 128MB also does not change the BAR size of my GPU.

xCuri0, 2023-05-15:
>  Using ReBarState to set reBAR to 128MB also does not change the BAR size of my GPU.

Linux may be changing it back to 256mb

Sid127, 2023-05-15:
I'm not entirely sure how to work around that...

xCuri0, 2023-05-16:
Also if you're using that override 128mb won't work because it's not in that supported caps value afaik

Sid127, 2023-05-16:
Removed the override, and reducing the BAR from 256 to 128 does work

xCuri0, 2023-05-16:
it's the VBIOS then, NVIDIA doesn't lock down laptop VBIOS so modification is technically possible but requires some very advanced reverse engineering knowledge

Sid127, 2023-05-16:
Thought as much, thanks for the help regardless <3

Would UEFITools be able to parse a vBIOS? Or is there any other tool I shoukd be using to poke around the vBIOS, just to see if I can manage to enable it myself?

Sid127, 2024-01-22:
a *while* after my initial attempt, I decided to take another look at things on the same laptop, and my dmesg now has:
```
[    0.497373] pci 0000:00:02.0: BAR 2: assigned to efifb
[    0.636093] pci 0000:01:00.0: can't claim BAR 6 [mem 0xfff80000-0xffffffff pref]: no compatible bridge window
[    0.636114] pci 0000:00:15.0: BAR 0: assigned [mem 0x674000000-0x674000fff 64bit]
[    0.636388] pci 0000:00:15.1: BAR 0: assigned [mem 0x674001000-0x674001fff 64bit]
[    0.636658] pci 0000:00:1e.0: BAR 0: assigned [mem 0x674002000-0x674002fff 64bit]
[    0.636927] pci 0000:01:00.0: BAR 6: assigned [mem 0xb2000000-0xb207ffff pref]
```
which is definitely different compared to the last time I tried.

I recalling not patching my DSDT despite having this
```
[sidpr@strogg ~]$ sudo dmesg | grep -iE "root bus resource"
[    0.499242] pci_bus 0000:00: root bus resource [io  0x0000-0x0cf7 window]
[    0.499246] pci_bus 0000:00: root bus resource [io  0x0d00-0xffff window]
[    0.499249] pci_bus 0000:00: root bus resource [mem 0x000a0000-0x000bffff window]
[    0.499251] pci_bus 0000:00: root bus resource [mem 0x8d800000-0xdfffffff window]
[    0.499253] pci_bus 0000:00: root bus resource [mem 0xfc800000-0xfe7fffff window]
[    0.499255] pci_bus 0000:00: root bus resource [bus 00-fe]
```
and being told my DSDT was already fine and I know for a fact I've got 4G decoding enabled as well

Any help would be appreciated :)

xCuri0, 2024-01-23:
@Sid127 If you want rebar on Turing 20 / 16 series there's a new method called NvStrapsRebar look at the repo or thread in discussions

HolimaX, 2024-02-05:
Any chance to get this for Lenovo T430?

See https://github.com/n4ru/1vyrain/issues/134