xCuri0/ReBarUEFI #20: Asus Z97 Maximus VII Hero

cursemex, 2022-12-01:
**System**
 - Motherboard: Asus Z97 Maximus VII Hero
 - BIOS Version: 3503
 - GPU: 6700 XT

Alright, so to summarize what happened last time, we've managed to get my motherboard to run with up to 24 GB of ram while 32 GB failed to run. 

Let me know if you require anything else that wasn't put on [GitHub](https://github.com/xCuri0/ReBarUEFI/issues/10#issuecomment-1310080110) or [Winraid](https://winraid.level1techs.com/t/semi-completed-offer-asus-z97-maximus-vii-hero/90930)

Milor123, 2023-03-29:
I have the same problem

Motherboard: Asus Z97 Maximus VII Hero
BIOS Version: 3503
GPU: 6750 XT

Question, what happend if the CSM is not disabled?

cursemex, 2023-03-29:
> I have the same problem
> 
> Motherboard: Asus Z97 Maximus VII Hero BIOS Version: 3503 GPU: 6750 XT
> 
> Question, what happend if the CSM is not disabled?

In order to get REBAR working, you need to have 4G decoding ON and CSM compatibility OFF. Otherwise, the REBAR won't work.

Going to 32 GB of ram won't work unless further patching is done, and I have no experience in the topic to do it myself.

Best of luck

IamDB90, 2024-01-12:
I have the same Mobo here "Asus Maximus VII Hero" with i7 4790K & ROG RX5700XT 8GB. I am unable to Enable the ReBAR feature on BIOS. Can you please share me your Modified BIOS file? I have Enabled "Above 4G Decoding" though but there is no ReBAR or Resizable BAR available. Please save your BIOS to a File & share it.

Milor123, 2024-01-12:
> I have the same Mobo here "Asus Maximus VII Hero" with i7 4790K & ROG RX5700XT 8GB. I am unable to Enable the ReBAR feature on BIOS. Can you please share me your Modified BIOS file? I have Enabled [...]

[https://winraid.level1techs.com/t/asus-z97-maximus-vii-hero-readjustable-bar/90930](https://winraid.level1techs.com/t/asus-z97-maximus-vii-hero-readjustable-bar/90930)

Pheoxy, 2025-04-03:
Using this with an Intel Arc Battlemage B580, I seem to constantly get `Misaligned address error` even though ReBAR appears to be enabled via the BIOS update and setting an 8GB BAR size seems to have worked.

xCuri0, 2025-11-03:
Trying to fix the Haswell 32GB RAM issue, can you run this EFI application MMIOPrint that prints the UEFI memory map and show a picture of the output. Doesn't matter how much RAM is installed.

https://github.com/xCuri0/ReBarUEFI/issues/48#issuecomment-1763741478

closed 2025-11-09 by xCuri0