xCuri0/ReBarUEFI #22: ReBar on my Asus P8B75-M (Ivy Bridge)

HelpBIOS, 2022-12-12:
**System**
 - CPU: Intel Core i7-3770
 - Motherboard: Asus P8B75-M
 - BIOS Version: [P8B75-M BIOS 1701](https://dlcdnets.asus.com/pub/ASUS/mb/LGA1155/P8B75-M/P8B75-M-ASUS-1701.zip)
 - GPU: Intel Arc A770

**Help**
@xCuri0 or someone with a good heart can help me to modify the BIOS for my motherboard, I don't trust myself and I'm afraid of brick my motherboard.

1.  When I open the official or modified BIOS in UEFITool I get different messages depending on the version of UEFITool:

[UEFITool non-NE v0.28.0](https://github.com/LongSoft/UEFITool/releases/tag/0.28.0)
[image]

[UEFITool NE vA61](https://github.com/LongSoft/UEFITool/releases/tag/A61)
[image]

2. First I followed the guide to [Adding ReBarDxe FFS module](https://github.com/xCuri0/ReBarUEFI/wiki/Adding-FFS-module) with **UEFITool non-NE v0.28.0** and then follow the guide on how to [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) and I get this output:
```
parseImageFile: Aptio capsule signature may become invalid after image modifications
patch: replaced 29 bytes at offset EDBh 77B6488B0F493BCF73AE48FFC1E81BFFFFFF488B1748FFC8483BD0759B -> 669066909066909066906690909090669090488B176690906690906690
patch: replaced 11 bytes at offset 8A44h B8FFFFFFFF030000004C3B -> B8FFFFFFFFFFFFFF004C3B
Image patched

File Hashes:
MD5: 187A214A7CC9899812803D8595D1E78D
sha1: 513992E878A48A5CB0AEA7FB478F52720C78FD75
sha256: FAA5CAAF19523DA3BBF65C6D6EDBDC6993E6CE37D773539796AC11F44B8645F3
```

3. I did not understand the [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) guide at all.

4. I can flash that modified BIOS or not? and if the modified BIOS is wrong please can you help me, I have no idea what I'm doing.

5. In the BIOS of this motherboard it does not have the 4G Decoding option and I followed the [Enabling hidden 4G decoding](https://github.com/xCuri0/ReBarUEFI/wiki/Enabling-hidden-4G-decoding) guide and **it gave me these two offset values** and I don't know which one to use after you confirm me or share a modified BIOS correctly to flash on my motherboard:

```
0x9E67F 		One Of: Above 4G Decoding, VarStoreInfo (VarOffset/VarName): 0x1, VarStore: 0x1, QuestionId: 0x8, Size: 1, Min: 0x0, Max 0x0, Step: 0x0 {05 A6 61 09 62 09 08 00 01 00 01 00 10 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00}
0x9E6A5 			One Of Option: Disabled, Value (8 bit): 0x0 (default) {09 0E 05 00 30 00 00 00 00 00 00 00 00 00}
0x9E6B3 			One Of Option: Enabled, Value (8 bit): 0x1 {09 0E 04 00 00 00 01 00 00 00 00 00 00 00}
0x9E6C1 		End One Of {29 02}

0xACC9D 			One Of: Above 4G Decoding, VarStoreInfo (VarOffset/VarName): 0x1, VarStore: 0x1, QuestionId: 0x17, Size: 1, Min: 0x0, Max 0x0, Step: 0x0 {05 A6 61 09 62 09 17 00 01 00 01 00 10 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00}
0xACCC3 				One Of Option: Disabled, Value (8 bit): 0x0 (default) {09 0E 05 00 30 00 00 00 00 00 00 00 00 00}
0xACCD1 				One Of Option: Enabled, Value (8 bit): 0x1 {09 0E 04 00 00 00 01 00 00 00 00 00 00 00}
0xACCDF 			End One Of {29 02}
```

I await your help.

Thanks.

HelpBIOS, 2022-12-13:
@xCuri0 please can you make the modified BIOS for my motherboard I would appreciate it, I don't have the knowledge or the tools to do it.

xCuri0, 2022-12-13:
@HelpBIOS wouldn't recommend doing unless you have a BIOS flasher (such as CH341A) since P8B75-M doesn't have USB Flashback or dual BIOS. This also makes flashing BIOS harder.

I recommend you ask on [win-raid](https://winraid.level1techs.com/) if you can't figure out flashing the BIOS or want to request a modded one.

All the tools required (UEFITool, AmiBoardInfo and iasl) are linked in the wiki instructions.

HelpBIOS, 2022-12-13:
I don't have any tool that you mention, I better not try it.

Thank you very much for his time.

closed 2022-12-13 by HelpBIOS