xCuri0/ReBarUEFI #23: Asus X99M WS motherboard could not write ReBarState UEFI variable

terryjx, 2022-12-19:
**System**
 - Motherboard:  ASUS X99M-WS， 
 - BIOS Version:  4001
 - GPU:   RTX A6000 / Titan RTX

**Description**
Describe the issue here. **Do not ask if it will work on a certain device try and see it yourself.** **Make sure CSM is off and 4G decoding is on**

Doubled checked and confirm CSM is off / all UEFI only, and above 4G decoding is enabled.

Followed miyconst's youtube instruction to append the ReBar.fss to the BIOS downloaded from ASUS website, and flashed the modified BIOS back to the mobo using BIOS flashback successfully.

And then run into the following issue (as shown in photo) when running the ReBarState as Administrator .

`Writing value of 10 / 1024 MB to ReBarState 

Failed to write ReBarState UEFI variable
GetLastError: 5
You can close the app now.`

Tried different inputs, for example 10, 20, 32 and 0, the returned errors are the same.

[image]

I just quickly looked into the code but was not able to  figure out what does GetLastError 5 mean.

xCuri0, 2022-12-19:
You have to use UEFIPatch to fix this. It's described in the readme/wiki how to do it.

terryjx, 2022-12-21:
Thanks, just tried to apply UEFIPatch, and the I also checked the Pad-file content to make sure it's the same as the stock BIOS. And the Pad-file content does not seem to have issue. 
[image]

So I wrote the patched BIOS back to the motherboard, enabled Above 4G decoding, disabled secure boot, and then ran ReBarState as Administrator, and it was able to modify the reg successfully this time.

However, after rebooting, the system just hang at boot screen code 99 (For ASUS motherboard, it means USB or legacy boot device error) though I've removed all USB drives already before booting. The screen is all black without even showing BIOS self-test or mobo vendor logo.

Just wondering where else could be wrong.

terryjx, 2022-12-21:
BTW, I'm able to restore the system by flashing the stock BIOS back using ASUS BIOS flashback without any issue, but re-flashing the patched/modified BIOS will still cause black boot screen.

xCuri0, 2022-12-21:
@terryjx you have to disable CSM

terryjx, 2022-12-21:
Sorry it's a typo, the one disabled is CSM, not secure boot.

terryjx, 2022-12-21:
Just tried it again from scratch, and it's working now even with unlimited resizable bar size (key in 32).

Two interesting findings:
1. In GPU-z, it's still showing CMS not disabled, although I've disabled CSM in BIOS, is it the same for other people's case or just me?
2, The value of "Resizable BAR enabled in BIOS" is  still NO, does it matter? 
[image]

In HWINFO, I can see the Resizable BAR size is now 64GB, does it mean that it's successful already?  The VRAM of RTX A6000 is 48GB, so I assume it's more than enough already.  
[image]

xCuri0, 2022-12-21:
@terryjx update your GPU-Z and it will display properly

closed 2022-12-21 by xCuri0

terryjx, 2022-12-21:
Just tried with GPU-Z 2.51.0 which should be the latest version,  weirdly the incorrect information is still there. 

[image]

Probably it's just a problem with GPU-Z, thanks for the help!

terryjx, 2022-12-21:
Posting my patched BIOS here, just in case anybody else wants to try it by themselves.

I only tried flashing it using ASUS BIOS Flashback function, so suggest you do it this way.

BTW, the .CAP file in the zip package is already renamed to the right format, so just extract it to a FAT16 or FAT32 USB thumb drive's root directory would be good to go. 

[X99MWS.zip](https://github.com/xCuri0/ReBarUEFI/files/10274248/X99MWS.zip)

xCuri0, 2022-12-21:
@terryjx I recommend you report this bug to GPU-Z developers since HWInfo has no problem detecting the 64GB BAR.

W1zzardTPU, 2024-02-08:
> Just tried with GPU-Z 2.51.0 which should be the latest version, weirdly the incorrect information is still there.
> 
> [image]
> 
> Probably it's just a problem with GPU-Z, thanks for the help!

Can you confirm that this still happens with GPU-Z 2.57.0?