xCuri0/ReBarUEFI #24: Help: Asus Rampage IV Extreme (RIVE)

1337-537h, 2022-12-21:
**System**
 - Asus Rampage IV Extreme: 
 - BIOS Version:4901
 - AMD Radeon RX Vega 64
 - [Y] CSM is turned off
 - [N] 4G decoding is enabled
 
I cant find 4G decoding in my bios. 
Propablyt it is hidden.. but i lost when i trying this guide "Enabling hidden 4G decoding"
What can i do?

xCuri0, 2022-12-22:
You can also use AMIBCP to set 4G decoding **ACCESS** to **USER** but the method in "Enabling hidden 4G decoding" is typically safer.

1337-537h, 2022-12-22:
I dont know AMIBCP and i cant find it.
Also i can do it via "Enabling hidden 4G decoding"

xCuri0, 2022-12-22:
@1337-537h then use that method

1337-537h, 2022-12-23:
Sorry my mistake i mean i CAN NOT do that method.
Im stuck after some steps..

AbraxasVi, 2022-12-26:
> Sorry my mistake i mean i CAN NOT do that method. Im stuck after some steps..

[RIVE-4901-NVMEBAR-4G.zip](https://github.com/xCuri0/ReBarUEFI/files/10302464/RIVE-4901-NVMEBAR-4G.zip)
I had insert the modules into the bios file, and used AMIBCP to set 4G decoding ACCESS to USER.
But I don't have a RIVE board to make a test, so USING IT ON YOUR OWN RISK😂

1337-537h, 2022-12-26:
Thanks but your file is in *.rom file.
My biosfiles are *.bin
I also have bios flashback if that maters.

colepcsx2, 2022-12-30:
Try the ASUS Bios Renamer for USB BIOS Flashback

1.Put Bios file and Bios renamer in same folder
2.Execute the Utility
3.Use new Bios file for USB BIOS Flashback.

1337-537h, 2022-12-30:
OK thanks i"ll try it.
I didn't know about ASUS Bios Renamer.

Squall-Leonhart, 2023-01-03:
it'll just change the file name to R4E.CAP anyway.

post7794, 2023-01-15:
My motherboard is also rampage iv extreme, when I enable above 4g decode, I can't get into the BIOS when I boot, I just get a black screen, I can't select the boot drive, it shows boot0af:error

1337-537h, 2023-01-15:
How did you enable above 4g decode?
I still can't do that.

post7794, 2023-01-15:
It's the tutorial in the Wiki, you have to note that the USB drive has to be in gpt format and choose uefi boot, but the modification didn't help, I was going to install Hackintosh

xCuri0, 2023-01-16:
@post7794 make sure everything CSM/legacy is off or set to UEFI only

What GPU do you have ?

post7794, 2023-01-16:
> @post7794 make sure everything CSM/legacy is off or set to UEFI only
> 
> What GPU do you have ?

rx560

xCuri0, 2023-01-16:
@post7794 is it on the stock vbios ? GCN/Polaris GPUs commonly have problems with GOP that are fixed by restoring original VBIOS or updating GOP

post7794, 2023-01-16:
Thanks, I'll give it a try

xCuri0, 2023-03-04:
Closing due to inactivity

closed 2023-03-04 by xCuri0

xCuri0, 2023-05-03:
@franckythebest english please

franckythebest, 2023-05-03:
hello little deterrent. I have an extreme rampage iv with a Xeon e5 1680V2, 32GB DDR3 2400MHZ and a RX5700XT THICC III.
bios 4901 modded with menu access "Above 4G". I have the same problem as @poste7794 . once enabled and CSM disabled, black screen. impossible to kick. you have an idea. THANKS

xCuri0, 2023-05-03:
@franckythebest same as #48

dimitrogen, 2024-03-18:
Hi everyone,

I registered on this forum because I had the same problem with rbar and 4g above decoding on rampage 4 extreme. I spent a lot of time trying to solve the issue, and it was easier than you might think.

Let me just say that from what I finally saw in the hidden menu in the BIOS, the Asus Rampage was a really serious server motherboard. It has an overclocking section called Extreme Tweaker added to the BIOS, and the main one is hidden, so it became Rampage 4 Extreme.

Here's something interesting: Back then and now, Windows titles old systems like mine as PCI bus, but in this case, it's recognized as PCI root complex (which is PCI 4.0 5.0) and it's 0000000000000000 - 00003FFFFFFFFFF (4398 GB) (seriously, this motherboard became a beast :D).

Since I didn't do DSDT patching, I just followed the steps for importing the rbar element (thanks xCuri0). Let me just add that in our BIOS (since it's quite different with this hidden menu from the regular one), the values from the patch are distributed in several bodies (as you can see in picture 2).

The command (setup_var) doesn't work because above 4g decoding has administrator rights once, and the second time it's in the hidden menu, something gets confused and the screen goes black. You need to use \AMIBCP-4.53.0050, and to open it, you need to be an extended user, not a regular user. If you open it as the first option, after activating 4g x.m.p, it stops working.

I've been trying to do this for years, now I wanted to explain the steps to you so you can do it yourself, but... :), but come on :) So I'm uploading the BIOS without the unlocked menu (because there are settings there that you shouldn't play with). In this menu, 4g is activated by default and as an extended user, and it should work with xmp. I don't know if it works because Iм not close to my computer yet, but the only thing that can happen is xmp not working (everything else should be perfect). If that happens, I'll send you the dangerous BIOS (in picture 3 I show some of the things inside).

I hope this helps you guys. I'm personally happy that the system is finally working.
[image]
[image]
[image]
https://drive.google.com/file/d/1gX_2FWhbgVujc-7qsKQkUSH4VKMWSxHR/view

> @franckythebest

franckythebest, 2024-03-18:
If you are interested, I can provide the latest R4E bios with all the fixes, nvme driver, rebar, hidden menu. it is extremely stable. my e5 2697v2 holds 114mhz with the "extreme tweaker" settings in auto except for the vccsa which must be set to 1.1v.

Hokomn, 2024-03-18:
@franckythebest @dimitrogen can you just upload bios file ? I Have e5-2697v2 on Asus Rampage 4extreme (my pc&english knowledge is so poor)

dimitrogen, 2024-03-18:
> If you are interested, I can provide the latest R4E bios with all the fixes, nvme driver, rebar, hidden menu. it is extremely stable. my e5 2697v2 holds 114mhz with the "extreme tweaker" settings in auto except for the vccsa which must be set to 1.1v.

If you wan i can try it...:) i dont make DSDT... But this post says i don`t need yt if my nubers bigger(https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching#sandyivy-bridge-dsdt-patch) i have all hidden menus - but don`t upload bios here...Is your bios have DSDT, and do my need  it?(in pic from 1st post)

franckythebest, 2024-03-18:
yes my bios has DSDT with correction of filling errors, maxpayload and maxreadrequest do you have a tutorial and what is the point? you have a normal i7 that you use the multiplier to increase the cpu frequency. the xeon e5s are all blocked except the 1600 series. my e5 2697v2 being blocked, I use BCLK despite everything this cpu is a beast of performance.

[Uploading R4E.zip…]()

dimitrogen, 2024-03-18:
@franckythebest 
in that hidden menu maxpayload and maxreadrequest ( https://enterprise-support.nvidia.com/s/article/understanding-pcie-configuration-for-maximum-performance) . there a values to adapt our boards to 4g and rebar...No problem with the default values, but they are the smallest, probably because it doesn't take into account that 4G decoding and RBAR will be enabled... just make sure you don't exceed the PCIe 3.0 bandwidth. The biggest you can set for PCIe without flooding is max payload 2048 and max request 4096, but there's no point in going to the limit.PCIe 3.0 bandwidth: PCIe 3.0 has a maximum bandwidth of 32 GB/s.
Max payload: The max payload is the maximum amount of data that can be transferred in a single PCIe transaction.
Max request: The max request is the maximum number of PCIe transactions that can be outstanding at a time.
Flooding: Flooding occurs when the PCIe bus is overloaded with too many transactions. :))
Yes my i7 is normal for a 1.2k euro in that times when i buy it - and yes your server proc is beast....What the point of DSDT modification in my scenario if i dont have any limitations in anything

xCuri0, 2024-03-19:
@dimitrogen

> maxpayload and maxreadrequest 

do these actually make a difference in gaming performance ? seems like it's mostly used for compute tasks limited by PCIe bandwidth ?

and you can manipulate those settings on consumer motherboards too through registers (like we do for rebar) if anyone wanted that.

dimitrogen, 2024-03-19:
@xCuri0

Hi!

In the hidden menu of my motherboard, the two settings are from 256 to 4096 (these settings are usually server settings and are well-coded in the BIOS, so I'm not sure if changing them through the registers will have any effect).

Otherwise, how much better are compute tasks limited by PCIe than games (of course, there are such other tasks)?

To be honest, yesterday I managed to unlock everything correctly in my BIOS (because I dare to assume that here (Adding FFS module · xCuri0/ReBarUEFI Wiki (github.com)) some steps need to be added at least for Rampage. More precisely, it should be added that the values that are patched in the patch can be 4 (for example), but they exist in different bodies and if you extract only one body and the other with this value does not - you get boot errors and incomplete compatibility problems with the voltages. I played around with this a lot and didn't have to make a DSDT and at the end all perfect 64gb bar i scared to try 128:). As I said, I unlocked the BIOS yesterday and very quickly changed the settings. Personally, I have set maxpayload and maxreadrequest to 1024 4096,(2048 4096 pci3.0 limit) and the difference for me at least (I haven't looked in detail) is that, for example, when there are stutters in a game (I only tested one before loading the payload :) ), the processor load seems to increase somehow and the stutter itself is much smoother on the moments flat, and the frame representation is more responsive. Maybe I'm just imagining things.

Excuse me for the inconvenience, can I ask you - I can't find and I don't know if there is an article on how to change BAR 0 and BAR 2-Can you point me? Thank you in advance.

xCuri0, 2024-03-19:
@dimitrogen have you checked using lspci on linux that the gpu supports and uses larger read/payload sizes ?

dimitrogen, 2024-03-19:
@xCuri0
Unfortunately, I haven't tested on Linux. To be honest, I haven't had Linux for at least 3 4 years. Life took me in a different direction and I didn't have time to deal with Linux stuff much .  Regarding read/payload sizes, I noticed that in that menu when I set something wrong, it doesn't boot at all (it stays  codes that I haven't seen). That's why I think it works at 1024x4096 because it boots :) but i testted today in first impressions  maybe 1, 1.5frame (but this is in teoretical error range) but definitely   more responsiveness feel in the mouse movements on screen and visual how i say upper + frame is more fluid in my prespective....

dimitrogen, 2024-03-20:
@franckythebest  
you have very good range....
in your post upper the link says bios i still uploading and can`t download it to try it :)

dimitrogen, 2024-03-21:
Hi everyone, a few days ago I managed to activate ReBAR, but the fact that DSDT could also be modified didn't leave me alone, so I decided to try it... I did everything in the video, but in the end, no matter what I did, the structure would get lost when importing AmiBoardInfoMod.efi. Finally, I contacted a friend who is a professional in this field (but doesn't have much time to deal with it) and we came to the following conclusions and steps... After editing and importing the AmiBoardInfoMod.efi file, it also makes changes to the following files,which in turn corrupts the file structure
[image]

According to him, this is because you should not use "Replace All" when importing (Name\s+\(_UID,\s*\"PCI\d\-CP(.*)\"\) Name \(_UID, 0x0$1\) and oтхer osnes)But not have tame to investigate...That's why we came up with an alternative method. He first said that corrupting the structure is not a problem as long as the values that are changed in the non-AmiBoardInfoMod.efi body are the ones that work (since some values are changed, as I said, after importing the body, and theoretically we only made corrections to the body).
That's why we need to restore the old values outside of this file... In my case, after UEFIPatch, the values in the following bodies change:
[image]
So now in these files we have the correctly changed values (note that they also rebuild exactly those parts that were previously changed without having to do so from importing AmiBoardInfoMod). So we follow all the steps, but if the structure is corrupted when importing AMIBoard, all the elements from picture 2 need to be imported again. This way the structure remains different, but everything is correct. According to the person, this simply adds an empty or additional step before the initial computer signal, which is not a problem. It would be a problem if, for example, the other values changed after importing AMIBoard were not corrected...So now i have perfectly workin all with in firs place (raw pad raw) and after (pad raw pad raw) 10 hours testing - no problems with auto voltages  overcklocking and any kind of stress tests...

dimitrogen, 2024-03-21:
!!!!I am making this a separate post because I think it is important!!!!As we established above, we make changes to AmiBoardInfoMod, and it makes changes based on its changes, which we don't want... But we make these changes to adapt the BIOS to the two new elements, ReBAR and NVMe... What I want to say is that I noticed that after importing the elements, making a save, and then adding AmiBoardInfoMod later, everything works fine no proplem but no rebuild to rebat nvme. However, if you take the BIOS, uefipatch it first, then add ReBAR and NVMe with uefitool, and at the same time and last add AmiBoardInfoMod, it also makes changes to those two files... And as we established above, this file still changes other values based on its own changed values. I want to say that in my opinion, this way AmiBoardInfoMod communicates much better with ReBAR and NVMe because it has changed them for itself :)... And then the other elements can be replaced with MM Tool.

loudhippo840, 2026-09-15:
i was able to take the bios linked above (google drive - R4E.CAP and inject my working nvme 4 module into it (it didnt have an nvme module) and was finally able to get above 4g working so i could use an nvidia tesla p40 for my llama.cpp setup (system wouldnt boot before i did this when the card was installed) now it boots installed correctly and sees all 24gb of the vram
i have attached my updated bios file here for anyone else who runs into this same issue to save them the trouble 

[R4Erebar4gnvme.zip](https://github.com/user-attachments/files/32225285/R4Erebar4gnvme.zip)

loudhippo840, 2026-09-15:
pics for reference attached here , thank you so much for your help, i couldnt have done it without you guys!

[image]

[image]

1337-537h, 2026-09-15:
Thank you too. :)