xCuri0/ReBarUEFI #25: Asus B85-Plus : "4GB decoding" feather failed

AbraxasVi, 2022-12-23:
**System**
 - Motherboard: Asus B85-Plus
 - BIOS Version: Both 2304 and 2305 tried
 - GPU: RX 5700
 - [ √ ] CSM is turned off
 - [ √ ] 4G decoding is enabled
 - [ √ ] UEFIPatch is applied (see README for more information)
 
**Description**

Describe the issue here. 
In Asus's bios, it uses "Memory Remap" as the "4GB decoding" in other motherboard.
Manual:
https://dlcdnets.asus.com/pub/ASUS/mb/LGA1150/B85-PLUS/E7928_B85-PLUS.pdf
In the 2.6.4 section, and it is enabled always, even you reset the bios.

As the wiki,
I inserted both the "NvmExpressDxe_5.ffs" and "ReBarDxe.ffs" to the 2304 or 2305 bios (extracted with UEFITool 0.28.0) , 
patched with UEFIPatch (but it seems only the "**PciBus | Remove <16GB BAR size limit (Ivy Bridge/Haswell)**" patched), 
i have got that my Pci Bus range ends at 000DFFF, when i try to patch the DSDT, but there is no **CreateQWordField** in my "DSDT.dsl", so this is different to the wikis,
finally i flash the "unmodded DSDT bios" with AFUWINx64 (with /GAN) and it succeced, but it seems the "Above 4G decoding" didn't work, even in the original asus bios,
i am not sure if it is a bug to this motherboard's bios.

[image]

[image]

AbraxasVi, 2022-12-23:
[image]

and now the GPU-Z says like this

xCuri0, 2022-12-23:
@AbraxasVi Memory Remap and 4G Decoding aren't the same thing.

I'm sure it is possible to enable 4G decode through assembly patching and DSDT edits but it's going to take alot of effort to do

AbraxasVi, 2022-12-25:
Interesting, I can find "Above 4G Decoding" string to token '0x0F97' in the bios file with AMIBCP v4.53. 
But it is unable to find this string in the "Setup Configuation", it's even not a hidden option.

I'm not good at DSDT modification, so I choose to wait someone to take it down😀

[image]
[image]

AbraxasVi, 2022-12-26:
Feedback
Using ReBarState.exe to set 1GB BAR size.

[image]
[image]

xCuri0, 2023-03-04:
Duplicate of #14

closed 2023-03-04 by xCuri0