xCuri0/ReBarUEFI #26: ASUS Z10PA-D8. Doesn't but only after ReBarState.exe with 32

SlimRG, 2022-12-23:
**System**
 - Motherboard: ASUS Z10PA-D8
 - BIOS Version: Z10PA-D8-ASUS-3801 (https://disk.yandex.ru/d/46p8AXX9CawTVw)
 - Modded BIOS: (https://disk.yandex.ru/d/uA7qCh1AiP-fxw)
 
 - GPU: Gainward RTX 3070 Phantom
 - [ Y] CSM is turned off
 - [Y ] 4G decoding is enabled
 - [Y ] UEFIPatch is applied (see README for more information)
 
1) I update VGA driver for GPU
2) I get original firmware
3) I turn on 4G and turn of CSM
4) I add [ReBarDxe.ffs](https://github.com/xCuri0/ReBarUEFI/releases/download/0.2/ReBarDxe.ffs) in firmware
5) I insert patch by UEFIPatch_0.28.0_win32 
6) I install firmawe and reboot
7) I reconfigure bios and reboot
8) I run [ReBarState.exe](https://github.com/xCuri0/ReBarUEFI/releases/download/0.2/ReBarState.exe), set 32 and tried to reboot.
9) PC doesn't start. If I install back original bios - it boot, if I install patched bios - it boot. But after this app - it doesn't boot.

Pls, help me. I will give you all info you want.

xCuri0, 2022-12-23:
Try starting with smaller sizes such as 9 (512MB) and increase till it stops working

SlimRG, 2022-12-23:
> Try starting with smaller sizes such as 9 (512MB) and increase till it stops working. Also check device manager resources to see if it's being allocated a 64-bit address or not

Any value doesn't work

SlimRG, 2022-12-23:
I think - I made smth wrong with bois rom. But - don't know what wrong I made...

xCuri0, 2022-12-24:
@SlimRG take a screenshot of device manager resources showing the GPU with just 4G decode on

xCuri0, 2023-01-06:
@SlimRG did you see this ?

SlimRG, 2023-01-07:
> @SlimRG did you see this ?

yes

But, I'm not at home - will be at home at 09.01.2023

SlimRG, 2023-01-23:
[image]
Sorry for waiting
Today is the first day, I'm not drinking)

savvadesogle, 2023-12-21:
> **System**
> 
>     * Motherboard: ASUS Z10PA-D8
> 
>     * BIOS Version: Z10PA-D8-ASUS-3801 (https://disk.yandex.ru/d/46p8AXX9CawTVw)
> 
>     * Modded BIOS: (https://disk.yandex.ru/d/uA7qCh1AiP-fxw)
> 
>     * GPU: Gainward RTX 3070 Phantom
> 
>     * [ Y] CSM is turned off
> 
>     * [Y ] 4G decoding is enabled
> 
>     * [Y ] UEFIPatch is applied (see README for more information)
> 
> 
>     1. I update VGA driver for GPU
> 
>     2. I get original firmware
> 
>     3. I turn on 4G and turn of CSM
> 
>     4. I add [ReBarDxe.ffs](https://github.com/xCuri0/ReBarUEFI/releases/download/0.2/ReBarDxe.ffs) in firmware
> 
>     5. I insert patch by UEFIPatch_0.28.0_win32
> 
>     6. I install firmawe and reboot
> 
>     7. I reconfigure bios and reboot
> 
>     8. I run [ReBarState.exe](https://github.com/xCuri0/ReBarUEFI/releases/download/0.2/ReBarState.exe), set 32 and tried to reboot.
> 
>     9. PC doesn't start. If I install back original bios - it boot, if I install patched bios - it boot. But after this app - it doesn't boot.
> 
> 
> BIOS POST Code: **87**
> 
> Pls, help me. I will give you all info you want.
> 
> P.S. I have programmator - so I can flash bios)

Hello, did you install rebar patch? I have same motherboard (but z10pe-d8 WS)