xCuri0/ReBarUEFI #27: Asus ROG STRIX Z270-I GAMING

KuRue, 2022-12-30:
MB: Asus ROG STRIX Z270-I GAMING
CPU: 7700K
GPU: RTX 3080
RAM: 16GB

With the latest BIOS for my motherboard (Version 2001), I added ReBarDxe and unencapsulated using UEFITool.
I flash the modified BIOS with AFUWin64 as described in the linked guide, enable 4G decoding and disable CSM.
BAR at 256MB
[image]

BAR at 1024MB - Freeze at BIOS logo, removing GPU boots back into Windows.
BAR at Unlimited - Bluescreen at during boot, Stop code: WDF VIOLATION

I repeated this with UEFIPatch and the linked patches.txt and checked for page-files (there are none), and I get the same results.

I'm currently trying DSDT patching but the only thing under Large Memory is: 
[0000002000000000 - 0000002FFFFFFFFF] PCI Express Root Complex
And I am not able to find AmiBoardInfo using UEFITool.

Guidance or incite would be greatly appreciated.

xCuri0, 2022-12-30:
@KuRue does 512MB BAR work ?

also post a screenshot of device manager resourcing showing the GPU (when its working)
like this 
[image]

xCuri0, 2022-12-30:
@KuRue when i run uefipatch it finds and patching
[image]

it should fix it because what you describe is exactly what happens without **Don't downgrade 64-bit BARs to 32-bit** patch

KuRue, 2022-12-30:
After patching and flashing again it looks like it's working correctly. I must not have saved patches.txt after replacing the text 🤦‍♂️. 
Setting BAR to unlimited shows BAR1 at 16GB and everything shows Yes or Enabled in GPU-Z now.

Thank you for the help and great work!

closed 2022-12-30 by KuRue