xCuri0/ReBarUEFI #28: Asus Z270-A Prime - ReBarState doesn't enable ReBar

WillyTP, 2023-01-15:
**System**
 - Motherboard:  Asus Z270-A
 - CPU: i7700K
 - BIOS Version: 1302
 - GPU:  Intel ARC380
 - [X] CSM is turned off
 - [X] 4G decoding is enabled
 - [N] UEFIPatch is applied (see README for more information)
 
**Description**
I modded the BIOS and flashed it via Asus AI Suite (all the other methods didn't work).
PC posts properly, however after running ReBarState.exe and rebooting into GPU-Z ReBar status is still disabled.

Are there some further checks I may do?

1. I'd like to know if new BIOS has really been flashed properly (I've got no errors, however I can't be sure)
2. Something else?

Thanks for help

[PRIME-Z270-A-ASUS-1302-ReBar.zip](https://github.com/xCuri0/ReBarUEFI/files/10419392/PRIME-Z270-A-ASUS-1302-ReBar.zip)

xCuri0, 2023-01-15:
@WillyTP does disabling secure boot help ?

maybe @KuRae (ASUS ROG Z270-I) can help because this UEFI layout is weird to me

WillyTP, 2023-01-15:
Hi @xCuri0 , thanks for support.
Unfortunately disabling Secure Boot has made no effects.

If anyone of you have some suggestion to what to do / further check is welcome.

Kind regards

[image]
[image]

xCuri0, 2023-01-15:
@dsanke UEFI layout on Z270 seems weird I saw you had it working on a Z270 board did anything have to be done differently ?

dsanke, 2023-01-16:
@xCuri0 I guess there isn't a successful BIOS mod, the Ai Suite won't flash unofficial BIOS.

WillyTP, 2023-01-16:
Hi @dsanke, thanks for your support.

Just for being clear: in order to flash the Bios you posted, is it OK via EZ Flash or shall I proceed through other ways?
For setting Bios Lock disabled, what do you mean exacly? Removing Secure Boot?

Thank you

dsanke, 2023-01-16:
> is it OK via EZ Flash or shall I proceed through other ways?
No you can't use any official flash method. You can use SPI Programmer hardware to read backup and write modified BIOS. Or follow the thread I have linked before. Set BIOS Lock first and use Intel FPT read backup and flash modified BIOS.
> For setting Bios Lock disabled, what do you mean exacly?
Go and see the winraid forum link above.
> Removing Secure Boot?
Yes in order to set BIOS Lock disabled it should be disabled.

WillyTP, 2023-01-16:
Hi @dsanke, thanks for your support.
Unfortunately now I don't have time for such a complicated process, however your help has been very appreciated.
One more question for you if you've time to tell:
I've checked with UEFITool the Bios you modded, and I've seen you've inserted Rebar under a different section than mine (I put it at the bottom of the section with all DXE drivers).
Why?

Thank you

dsanke, 2023-01-17:
@WillyTP It might be my mistake because I modified these BIOS using script, I modified all ASUS 100/200 series BIOS.
Will correct it.

nook24, 2023-01-30:
I also have an Asus Z270-A with an Intel i7700k. Were you able to successfully patch the BIOS?

WillyTP, 2023-01-30:
Hi @nook24
Unfortunately I haven't had the time to flash the one provided by @dsanke due to the complex procedure to do (lack of time)

Should be able to, please let me know how you manage it.

Thank you

WillyTP, 2023-02-12:
Hi @nook24 , everyone
I've been able to flash modded bios, even tough now I cannot check ReBar functionality since I changed GPU with a GTX 1660 Super.

Anyway, procedure has been smoother than what I tought.
I quickly resume in a few points:

1. Download Intel FPT and tools package from link [above](https://winraid.level1techs.com/t/guide-grub-fix-intel-fpt-error-280-or-368-bios-lock-asus-other-mod-bios-flash/32725)
2. Backup current BIOS (FPTw -bios -d backup.bin)
3. Mod backupped bios
4. Disable Secure Boot
5. Disable Bios Lock via EFI shell (0x90C -> 0x00)
6. Flashe new BIOS (fptw -bios -f Z170A_ModdedBackup.bin)

Kind regards

nook24, 2023-03-04:
Thanks for your replay. Unfortunately I'm always ending up with the `Pad-File` issue when i try to modify my BIOS, even if i go the MMTool route.

So I'm not going to try to flash my patched bios as i can not risk to brick this system.

xCuri0, 2023-03-25:
@WillyTP this is an issue that happens when flashing with ASUS AI Suite. See #42 

You need to use another method such as afuwin/afudos

closed 2023-03-25 by xCuri0

OmgStreaker, 2023-11-09:
> @xCuri0 Seen some TE image not aligned? I guess there isn't a successful BIOS mod, I think the Ai Suite won't flash unofficial BIOS. Follow this: https://winraid.level1techs.com/t/guide-grub-fix-intel-fpt-error-280-or-368-bios-lock-asus-other-mod-bios-flash/32725 The offset of BIOS Lock is 0x90C, and I have modified BIOS provided below. This BIOS have ReBAR default enabled so you dont need use ReBarState. Set BIOS Lock disabled and backup your BIOS then flash the modified BIOS I provided. Just flash BIOS Region, the Flash Descriptor and ME Region need more operation to write. This BIOS can support Coffee Lake when you flash Flash Descriptor and ME Region(and need pin mod for U0/P0/R0 step CFL CPU), but I guess you don't need that. [PRIME-Z270-A-ASUS-1302.zip](https://github.com/xCuri0/ReBarUEFI/files/10433959/PRIME-Z270-A-ASUS-1302.zip)

Can someone plz help me patching my z170 pro gaming bios? (intel i7 6700k)

OmgStreaker, 2023-11-09:
Anyone ?

dsanke, 2023-11-13:
[Z170-PRO-GAMING-ASUS-3805.zip](https://github.com/xCuri0/ReBarUEFI/files/13331356/Z170-PRO-GAMING-ASUS-3805.zip)
@OmgStreaker

OmgStreaker, 2023-11-13:
> [Z170-PRO-GAMING-ASUS-3805.zip](https://github.com/xCuri0/ReBarUEFI/files/13331356/Z170-PRO-GAMING-ASUS-3805.zip) @OmgStreaker

Thankyou so much bro but i mod it myself yesterday 😁

OmgStreaker, 2023-11-13:
Now, i have new issues on the way 🥲