xCuri0/ReBarUEFI #30: AB350 Gaming  Rev:1 AM4 Problem in above 64 and 2g in BAR , for Rx580 Nitro 4g

RRM5, 2023-02-03:
**System**
 - Motherboard: AB350 Gaming Rev:1
 - BIOS Version:51g
 - GPU: 1600AF 
 - [Y] CSM is turned off
 - [Y] 4G decoding is enabled
 - [Y ] UEFIPatch is applied (see README for more information)
 
**Description**

Well, that with above64 and 2g of BAR on a RX580Nitro 4g, it gives me microblocks and an FPS drop of about 80 to 40.
The value of the BAR 512, 1024, 2048 does not matter. 2048 is the limit for graphics with 4g of vram (as I understand).
But if I don't activate above64 in the bios and I put 1024 of BAR, it behaves perfectly. Without the above64 it is the maximum 1024 because it does not matter if they change the MMIO, it does not admit more than 1024 and below.

RRM5, 2023-02-28:
Well after seeing the behavior a bit if I activate BAR and 11=2048 what I see is that the voltage of 1135 goes to peaks of 1190, and the watts go from about 150w in doom eternal without BAR, to about 120w with bar.
Let's go the world backwards, it puts more voltage and consumes less. With a drop of almost 50FPS.

RRM5, 2023-03-26:
With an RX 6600 Eagle, rebar works correctly for me with the bios mod with the 8G

xCuri0, 2023-03-26:
@RRM5 what about with 2gb bar size. Because with 8gb BAR you won't be using the part with slow access unless you use close to the VRAM limit

xCuri0, 2023-03-26:
> **System**
> 
> * Motherboard: AB350 Gaming Rev:1
> * BIOS Version:51h MOD
> * GPU: 1600AF
> * [Y] CSM is turned off
> * [Y] 4G decoding is enabled
> * [Y ] UEFIPatch is applied (see README for more information)
> * RX 6600 Eagle 8g Vram  , with the Nitro there was no way for it to work well RX580 Nitro 4096 Vram
> * 2 Nvme , one on board port and one with pcie adapter.
>   **Description**
> 
> Well, that with above64 and 2g of BAR on a RX580Nitro 4g, it gives me microblocks and an FPS drop of about 80 to 40. The value of the BAR 512, 1024, 2048 does not matter. 2048 is the limit for graphics with 4g of vram (as I understand). But if I don't activate above64 in the bios and I put 1024 of BAR, it behaves perfectly. Without the above64 it is the maximum 1024 because it does not matter if they change the MMIO, it does not admit more than 1024 and below.
> 
> As seen with SAM ON 2048 the minimum FPS drops to half, it's strange, it seems that SAM and my Hardware don't get along.
> 
> [image] [image] [image] [image]
> 
> This is problem: [https://gitlab.freedesktop.org/drm/amd/-/issues/1864](https://gitlab.freedesktop.org/drm/amd/-/issues/1864)
> 
> The memory range FFFFFFFFFF
> 
> [image]

RRM5, 2023-03-26:
@xCuri0  With the RX 580 or with the value 10, 12, 13 and others it was not stable, the FPS dropped a lot because of the Vram range even with all stock. Come on, the PC was slowing down.
On the other hand, with the RX 6600 that doesn't happen, the bios is the same, and the entire PC is the same with the same OC/Undervolt values. I leave some screenshots so you can see the ranges of the RX 6600. Greetings and thanks.
[image]
[image]

Fenivan, 2023-04-01:
> @xCuri0 With the RX 580 or with the value 10, 11, 12 and others it was not stable, the FPS dropped a lot because of the Vram range even with all stock. Come on, the PC was slowing down. On the other hand, with the RX 6600 that doesn't happen, the bios is the same, and the entire PC is the same with the same OC/Undervolt values. I leave some screenshots so you can see the ranges of the RX 6600. I think it's because of the Graphics/Bios of the RX 580 4g. But I don't know much about it. Greetings and thanks. [image] [image]

Hey, which value did you use for RX 6600?
I got same gpu and want to enable it.

RRM5, 2023-04-02:
> Hey, which value did you use for RX 6600? I got same gpu and want to enable it.

First a bios mod with the RebarDXE added, then in windows RebarState with the value 13= 8G of vram. Without modifying the registry Legacy.
[image]

DnnhSilva, 2024-04-30:
Can you replace BIOS mod file?

Link is broken. Thank u

DnnhSilva, 2024-04-30:
> Thanks @xCuri0 and @Koekieezzz . Well, in the end, doing tests, it worked at 10= 1024, at 11=2048 the performance drops a lot. I leave the ROM in case someone uses this GA-AB350 GAMING rev1 board (just plain, it's not gaming 3). It worked for me, I installed the bios, adjusted csm disabled, above 4g enabled, boot security enabled, before leaving I saved the profile to load it later. I turn off before entering w11, remove the battery for a few minutes, put it on, boot and enter bios and restore the serial configuration, and load the saved profile and modify the rest, save, enter w11, REBAR Legacy On.reg for me rx580 , then ReBarState.exe value 10= 1024 and reboot. Now I don't lose FPS in Doom Eternal, if I don't do it like this I lost about 50 FPS.
> 
> [AB350G.51h](https://mega.nz/file/kd9B3RhS#thRudnNVP5om1VDWSdoatrwdxym6lBkpVtIewNsDK2w) [Program to change bios image (it is customized for me)](https://mega.nz/file/5YlXVDIS#jrPqTHdivC1xOiHwx5V36jAWkLawXqTWJbYtULU6xNI)
> 
> [image] [image]

Can you replace BIOS mod file?

Link is broken. Thank u