xCuri0/ReBarUEFI #31: [Question] Does this support Nvidia 2000 gpus?

jxnding, 2023-02-06:
Is this for adding support to unsupported motherboards or GPUs too?

I've read through the README and looked through the supported mobos list, which seems to imply that this is just for hacking CPU/mobo support and not GPU support.

Would this work on my system?
5800X / B550 / 2080 Ti

xCuri0, 2023-02-06:
@jxnding Only motherboards. You can force a BAR resize on Turing GPUs by adding an override to this module but I doubt it would work

Sid127, 2023-05-14:
out of curiosity, what'd be the correct way to force a BAR resize?

xCuri0, 2023-05-15:
@Sid127  See the 5600 XT override https://github.com/xCuri0/ReBarUEFI/blob/master/ReBarDxe/ReBar.c#L167

You should try doing these modifications in Linux kernel first though since its easier to test there.