xCuri0/ReBarUEFI #33: Flash Ultility 2 Selected file is not a proper BIOS / not a UEFI BIOS

Schlappenschleuder, 2023-02-18:
**System**
 - Motherboard: Asus Sabertooth Z97 MARK S
 - BIOS Version: 2702
 - GPU: AMD Gigabyte Aorus 5700XT
 - CPU: i7 4770k
 - [y] CSM is turned off
 - [y] 4G decoding is enabled
 - [y] UEFIPatch is applied (see README for more information)
 
**Description**

Did UEFITool Method. It created a .ROM file After that i used the UEFIPatch with Powershell which corrected one path: 114Ah 75080FBAE80F89442430 - EB080FBAE80F89442430. Then i got Pad file created at PcIHostBridge which i corrected with MMtool.

The i renamed the .rom to .cap but Flash Utility says file is not proper Bios.

The Modded file is in .rom format but i need .cap.

Have seen a youtube video where someone extraced the modded.rom file from UEFITool and then used HxD to switch the Capsual HEX binaries from the original to the modded and then just renamed the rewritten file to the original one.

https://youtu.be/aISCfhK05Rk?t=1375

I did the same but in Flash Utilitily then says it's not a UEFI Bios and doesn't even start to flash.

This Motherboard isn't in the list yet as i see but other z97 types worked.

Im pretty new to this so please don't harass me to much :D

xCuri0, 2023-02-19:
Mentioning @Tokobotenkai @cursemex @Divin1ty666 since they all own ASUS Z97 boards and probably know how to solve this.

cursemex, 2023-02-19:
I used this as one way to flash.
https://rog.asus.com/forum/showthread.php?99490-Flash-any-most-Asus-motherboard-Bios-in-DOS-with-USB-tutorial-Intel-AMD-roll-back

Using EZFlash won't work afaik because of CAP & Rom modifications.
Best of luck.

Tokobotenkai, 2023-02-19:
> **System**
> 
>     * Motherboard: Asus Sabertooth Z97 MARK S
> 
>     * BIOS Version: 2702
> 
>     * GPU: AMD Gigabyte Aorus 5700XT
> 
>     * CPU: i7 4770k
> 
>     * [y] CSM is turned off
> 
>     * [y] 4G decoding is enabled
> 
>     * [y] UEFIPatch is applied (see README for more information)
> 
> 
> **Description**
> 
> Did UEFITool Method. It created a .ROM file After that i used the UEFIPatch with Powershell which corrected one path: 114Ah 75080FBAE80F89442430 - EB080FBAE80F89442430. Then i got Pad file created at PcIHostBridge which i corrected with MMtool.
> 
> The i renamed the .rom to .cap but Flash Utility says file is not proper Bios.
> 
> The Modded file is in .rom format but i need .cap.
> 
> Have seen a youtube video where someone extraced the modded.rom file from UEFITool and then used HxD to switch the Capsual HEX binaries from the original to the modded and then just renamed the rewritten file to the original one.
> 
> https://youtu.be/aISCfhK05Rk?t=1375
> 
> I did the same but in Flash Utilitily then says it's not a UEFI Bios and doesn't even start to flash.
> 
> This Motherboard isn't in the list yet as i see but other z97 types worked.
> 
> Im pretty new to this so please don't harass me to much :D
> 
> USB-Stick is In FAT32 Format

@Schlappenschleuder 

I had issues flashing so this worked for me:

I used a hex editor using instructions here (https://winraid.level1techs.com/t/howto-get-full-nvme-support-for-all-systems-with-an-ami-uefi-bios/30901/5900?page=296).
```
    First, use UEFITool on the stock.CAP BIOS to extract the body >> Right click AMI Aptio Capsule and choose extract body, Then do NVME mod to that bin/rom
    This is how you should do that >> open stock.CAP in Hex editor and go to offset 800. Open mod bios rom/bin in second tab of hex editor, select all and copy, then paste at offset 800 in stock.cap, then save this file as M5F.CAP

    Extension name does not matter so much when saving, I mean you can easily rename, but how the BIOS is created and it’s contents are what matters.
    Your mod BIOS is in a capsule, but named rom/bin, renaming should have been OK but it’s probably not accepted due to how you did the mod with capsule included while editing.
```
How I used the hex editor would be the same here: https://winraid.level1techs.com/t/howto-get-full-nvme-support-for-all-systems-with-an-ami-uefi-bios/30901/5903?page=296

Hope it helps.

Schlappenschleuder, 2023-02-19:
Thanks for the quick replies,  will try those out later.

Will let you know if those methods worked to share for others.

Schlappenschleuder, 2023-02-19:
Hello again, @cursemex @Tokobotenkai @xCuri0 

Got the Bios flashed with the USB method @cursemex mentioned. Didn't need to change the HEX Binaries. Booted up fine and went into BIOS do disable CSM and enable 4G Decoding. GPU-Z shows that only Resizable BAR isn't enabled.

Launched ReBarState and set value to 14 for 16gb and got GetLastError 5.

I've got 16 GB Ramm installed on Motherboard.

Does it have domething to do with overclock ? 

Is there a way to solve this? 

Can't find anything related to that in @cursemex or Githubs Thread.

xCuri0, 2023-02-20:
@Schlappenschleuder you aren't using the correct patches.txt that's provided in this repo. i downloaded your bios file and none of the patches are applied.

[image]

ofc you have to do the MMTool thing to not get any pad files when patching

Schlappenschleuder, 2023-02-20:
@xCuri0 It works! Thank you very much!

I just overlooked the txt file in that thread thank you for mentioning.
Only little issue is on boot my keaboard doesn't get detected but in Windows it does but doesn't bother me so no big deal :)

I am goin to look into games soon to see what performance increase i get.

Set bar size in ReBarState.exe directly to value 14 for 16 GB. Should i try more ? This is the Size for now.
I have 8192mb Memory size on my card and 16 GB of Ram if this influences working ReBarSize please let me know.

Also thanks to @cursemex for giving me Thread and step by step instructions & @Tokobotenkai for letting me know how to use HXE editor even if i didn't needed it :)
[image]

Schlappenschleuder, 2023-02-22:
@xCuri0 I do have a question, when i set barsizevalue in ReBarState to 15 for 16gb and reboot, i still got 8gb shown in GPU-Z.
Does this mean i am at maximum?

xCuri0, 2023-02-23:
@Schlappenschleuder yes because 8gb VRAM GPU

closed 2023-02-23 by xCuri0