xCuri0/ReBarUEFI #35: DSDT cannot be extracted | AmiBoardInfo2

glikched, 2023-02-26:
[image]

I tried Extracting the dsdt included in to this DXE driver but failed.

[bios](https://mirror.rex-drive.workers.dev/0:/rebar.fd?a=view)

glikched, 2023-02-26:
I am trying that way as after i enable Above 4G it can't see sata drives.

xCuri0, 2023-02-27:
@glikched you gave zero information about your system

And the issue with drives not showing is because you installed windows in legacy mode. the wiki common issues says how to fix it

xCuri0, 2023-02-27:
@glikched Does Linux boot with above 4G decode on ? DSDT patch won't fix the issue you have, if the drive doesn't show in the BIOS it's probably some legacy option being on

glikched, 2023-02-27:
It doesn't

glikched, 2023-02-27:
Sir. I don't want to harm but i said i have everything set on uefi mode. I will send you some pitrures after i setup_var 0x377 0x1 again.

xCuri0, 2023-03-04:
Wiki now has instructions for extracting and reinserting DSDT on these boards. One user on win-raid was successful doing it

glikched, 2023-03-04:
Ok thank i didn't got time to check and give you more info.

dormancygrace, 2023-03-12:
> [image]
> 
> I tried Extracting the dsdt included in to this DXE driver but failed.
> 
> [bios](https://mirror.rex-drive.workers.dev/0:/rebar.fd?a=view)

First of all, you need raw section from DsdtAsl. You should search>GUID>header only

xCuri0, 2023-03-14:
Continuing in #38

closed 2023-03-14 by xCuri0