xCuri0/ReBarUEFI #36: Asus H81M-D v1 4g decode is in the strings but not in the text file

NoUserNameForYou, 2023-03-04:
**System**
 - Asus h81m-D rev 1:
 - My custom made 3602 with microde 6 for all core turbo + nvme patch
 - 3080ti
 - [ ] CSM is turned off? I don't know how to do it yet, so it's turned on I guess
 - [ ] UEFIPatch is applied (see README for more information) - Don2T know what that is either
 
-----------

I remember changing some var settings in order to flash the bios with fpt with the grub shell. 

UEFItool finds the 4g decode in this guid: "Section_Freeform_subtype_GUID_97E409E6-4CC1-11D9-81F6-000000000000_Setup_Setup_body" but its text file doesn't have it.
AMIBCP4 in bios strings section states "0x0F84 Above 4G decoding" and "0x0F85 Enable or disables 64 bit capable devices to be decoded..." its text explanation.

There's also "0x079D Enable or disable memory map above 4G" string when I search for "above".

When I look at the empty section under "main" in Setup confiuration part, I can't see anything related to 4g decoding. Can I patch in strings in there? Am I SOL?

Please help me. Thank you.

xCuri0, 2023-03-04:
Duplicate of #14

closed 2023-03-04 by xCuri0

NoUserNameForYou, 2023-03-04:
Thanks mate. I also posted in your thread on winraid. Apparently some guy was able to patch in strings in his bios. 

https://winraid.level1techs.com/t/add-option-in-ami-uefi-bios-menu-how-to/31609/

Can you at least direct me in the correct path so I can experiment?

NoUserNameForYou, 2023-03-04:
Oh I see you said 4gb can be patched with asm code but you're not planning it.

Well thanks for your efforts anyway, I was able to activate 1GB resizable bar.

xCuri0, 2023-03-05:
@NoUserNameForYou 

>Oh I see you said 4gb can be patched with asm code but you're not planning it.

It's very complicated and I probably won't do it, best to try another way like what you linked