xCuri0/ReBarUEFI #37: Unpatchable atm../DSDT too large even with _OSI removed

264-bit, 2023-03-09:
Apologies if I'm missing something obvious.
I'm trying to apply the DSDT patch https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching#dsdt-broken-4g-decoding-fix-haswell-and-up to my Asrock Z97 Extreme 3 on BIOS 2.10: https://www.asrock.com/mb/intel/z97%20extreme3/#BIOS
However, even with all of the _OSI methods removed, my DSDT mod is significantly larger than the original, almost 700 bytes larger using iasl from ACPICA 20221020.  Any ideas if there's any other parts that are safe to remove, or any other way to fix this?

[dsdt.zip](https://github.com/xCuri0/ReBarUEFI/files/10936647/dsdt.zip)

xCuri0, 2023-03-10:
@264-bit You can try

* Remove all ```ADBG``` lines
* Remove device ```BAT0``` / ```BAT1`` / ```BAT2``
* Remove device ```DOCK```
* Remove device ```LID0```

All of these are unneeded on desktop.

264-bit, 2023-03-11:
Thanks, that helped a lot.
I was still a few bytes short, so I also shortened the SLIC string.  This probably breaks SLI but I'm not using it anyways.
With all of these changes, I am now able to fit my modded DSDT in my amiboardinfo.

closed 2023-03-11 by 264-bit

xCuri0, 2023-03-12:
add a comment to list of working motherboards