xCuri0/ReBarUEFI #38: ASRock z-170 k4/d3 can't open the "above 4g decoding" or "above 4g mmio bios assignment"

jnyk, 2023-03-12:
Motherboard: ASRock z-170 k4/d3
GPU: 6700xt
I can't open the "above 4g decoding" or "above 4g mmio bios assignment"
[image]
when I turn this to enable ,it can't work, can't into windows system .

[image]
[image]

then when i try to use modGRUBShell.efi  
setup_var 0x3 0x0
reboot
setup_var 0x3 0x1
but there is nothing happend

It makes me very depressed. Can  you give me some suggestions

_Originally posted by @jnyk in https://github.com/xCuri0/ReBarUEFI/issues/11#issuecomment-1465085555_

jnyk, 2023-03-12:
[image]

jnyk, 2023-03-12:
[image]

dormancygrace, 2023-03-12:
https://github.com/xCuri0/ReBarUEFI/issues/35

jnyk, 2023-03-13:
> I believe that it's same issue #35
> 
> @jnyk do you have ch341a or another programmer? If it's dsdt issue, I can reinsert dsdt from another ASRock motherboard, that works as intended

yes  I have met  similar situation with him,can not extract DTST from AmiBoardInfo2.I have bought one  ch341a ,meybe it will cost two days from online store to me

Koekieezz, 2023-03-13:
@jnyk here you could try this bios, dsdt patched, uefipatched.
[z17gkd37mod.zip](https://github.com/xCuri0/ReBarUEFI/files/10952783/z17gkd37mod.zip)

xCuri0, 2023-03-13:
DSDT is in ```DsdtAsl```, not ```AmiBoardInfo2```

dormancygrace, 2023-03-13:
> @jnyk here you could try this bios, dsdt patched, uefipatched. Just enable Above 4G like usual, the one that made windows hang/freeze. [z17gkd37mod.zip](https://github.com/xCuri0/ReBarUEFI/files/10952783/z17gkd37mod.zip)

Do you use this mb? Or just made modifications? What you changed in dsdt?

Koekieezz, 2023-03-13:
> > @jnyk here you could try this bios, dsdt patched, uefipatched. Just enable Above 4G like usual, the one that made windows hang/freeze. [z17gkd37mod.zip](https://github.com/xCuri0/ReBarUEFI/files/10952783/z17gkd37mod.zip)
> 
> Do you use this mb? Or just made modifications? What you changed in dsdt?

i dont use this mb but i've done several dsdt fix on asrock boards (yes they're just pAWJPASJd;JDp[ to fix sometimes)
what did i change? well mainly the qwordfield under the first dwordfield, then also correcting the rest of dsdt errors (its a long list lol)

dormancygrace, 2023-03-13:
> @jnyk here you could try this bios, dsdt patched, uefipatched. Just enable Above 4G like usual, the one that made windows hang/freeze. see if it's working with this bios. [z17gkd37mod.zip](https://github.com/xCuri0/ReBarUEFI/files/10952783/z17gkd37mod.zip)

Is there coffee lake mod? Because @jnyk use i7-8700

dormancygrace, 2023-03-13:
> > > @jnyk here you could try this bios, dsdt patched, uefipatched. Just enable Above 4G like usual, the one that made windows hang/freeze. [z17gkd37mod.zip](https://github.com/xCuri0/ReBarUEFI/files/10952783/z17gkd37mod.zip)
> > 
> > 
> > Do you use this mb? Or just made modifications? What you changed in dsdt?
> 
> i dont use this mb but i've done several dsdt fix on asrock boards (yes they're just pAWJPASJd;JDp[ to fix sometimes) what did i change? well mainly the qwordfield under the first dwordfield, then also correcting the rest of dsdt errors (its a long list lol)
> 
> and no, i dont slap other board DSDT on this bios

But there is already qwordfield in dsdt in this board. I'm 90% sure that there is no need to patch uefi not dsdt. I will make bios without patching

dormancygrace, 2023-03-13:
> heres the one with coffee lake mod @jnyk [z17gkd37_2.zip](https://github.com/xCuri0/ReBarUEFI/files/10952835/z17gkd37_2.zip)

Please, make one without any patches.

Koekieezz, 2023-03-13:
> > heres the one with coffee lake mod @jnyk [z17gkd37_2.zip](https://github.com/xCuri0/ReBarUEFI/files/10952835/z17gkd37_2.zip)
> 
> Please, make one without any patches.

He could test it and if it's working it could be the DSDT faulty? he's the one testing it. if anything went good then DSDT was the one causing it, if its not, then it wasnt it, simple. Even if everything seems to be in place, sometimes it's just sketchy XD

dormancygrace, 2023-03-13:
> He could test it and if it's working it could be the DSDT faulty? he's the one testing it. if anything went good then DSDT was the one causing it, if its not, then it wasnt it, simple. Even if [...]

As I mentioned, I've compared dsdt from similar ASRock mb - they are +/- same (identical in qwordfield and m1.. parts)

Koekieezz, 2023-03-13:
> As I mentioned, I've compared dsdt from similar ASRock mb - they are +/- same (identical in qwordfield and m1.. parts)

let him reply first okay? he's the one testing

jnyk, 2023-03-13:
thanks @Koekieezz .I'm working now ， may not be able to reply to a message promptly.
I will try this bios when i get home.

jnyk, 2023-03-13:
By the way ,I try to patched UEFI to my bios.After  I  flashed the ROM whitch I patched, windows can be used as usually, but when i  
insert the usb disk ， it could not start windows  system, utill i remove the  usb disk

dormancygrace, 2023-03-13:
@Koekieezz is it secret, what you changed? i'm asking because stock dsdt and from your file - completely same 
[image]

Koekieezz, 2023-03-13:
> @Koekieezz is it secret, what you changed? i'm asking because stock dsdt and from your file - completely same [image]

no.. both files have 0x7FFFFFFFFF on qword, and for the rbarcfl one have acpi fix on dsdt then apply the dsdt patch

dormancygrace, 2023-03-13:
> > @Koekieezz is it secret, what you changed? i'm asking because stock dsdt and from your file - completely same [image]
> 
> no.. both files have 0x7FFFFFFFFF on 1st qwordfield, and for the rbarcfl one have acpi fix on dsdt then apply the dsdt patch

I`ve attached screenshot of files compare. only difference is disassembling by iasl

jnyk, 2023-03-13:
@Koekieezz  I just instant flash the bios ,
when i open above 4g mmio bios assignment, it can not find any disk drive  as before.
i use asrock's tool to flashed bios,may be use ch431a it will be ok

dormancygrace, 2023-03-13:
> @Koekieezz I just instant flash the bios , when i open above 4g mmio bios assignment, it can not find any disk drive as before. i use asrock's tool to flashed bios,may be use ch431a it will be ok

Ch341a you need to use in case of major failure

dormancygrace, 2023-03-13:
@jnyk  
[z17gkd37.zip](https://github.com/xCuri0/ReBarUEFI/files/10957751/z17gkd37.zip)

try this one

jnyk, 2023-03-13:
i flash this bios with ch341a ,but It's still the same as before

jnyk, 2023-03-13:
@dormancygrace  ok I‘ll try it

Koekieezz, 2023-03-13:
> I`ve attached screenshot of files compare. only difference is disassembling by iasl

i have 2 bioses attached 12h ago, they have different dsdt from normal ones, its this one: [z17gkd37rbarcfl.zip](https://github.com/xCuri0/ReBarUEFI/files/10953131/z17gkd37rbarcfl.zip), check em again

jnyk, 2023-03-13:
@dormancygrace   It's still the same as before . start my conputer can't find any disk drive except  usb disk ,and can not inter into 
neither windows system or winpe system

jnyk, 2023-03-13:
@Koekieezz  yes wo download this [z17gkd37rbarcfl.zip](https://github.com/xCuri0/ReBarUEFI/files/10953131/z17gkd37rbarcfl.zip) one

i can not find disk drive untill setup_var 0x385 0

dormancygrace, 2023-03-13:
@Koekieezz in first bios you've made, is there cfl mod?

jnyk, 2023-03-13:
i was in the same situation as him
[image]

dormancygrace, 2023-03-13:
@jnyk if you used programmer, I think, that you erased your Mac address

jnyk, 2023-03-13:
yes i used ch341a programmer to flash bios

jnyk, 2023-03-13:
> > i can not find disk drive untill setup_var 0x385 0
> 
> oh i think you should use iGPU, try to enable igpu, set display primary to igpu, and enable multi monitor/display on bios, save everything, then try to re do the thing i told for modgrub but this time using iGPU. the person i mentioned from winraid DID have the same problem as you but once he use igpu.. it worked fluently.
> 
> Here is the post: https://winraid.level1techs.com/t/experimental-resizable-bar-dxe-driver/89534/275?u=koekieezz Maybe you could ask that guy since he figure it out. also the bios i gave to you contains the latest orom, efigop and the gop vbt itself for coffeelake all revisions (up till 9900K), igpu should work. the patch i gave to you is modded for coffee lake (acpi table fix, fd unlock, 128GB ram unlock, and patch for 8+ thread for cpu) + uefipatch for 64GB allocation + dsdt fix (this done step by step from coffeemod to dsdt) changing from external reference to 0x7FFFFFFFFF.
> 
> > your mean is i need try to setup_var 0x385 1 gagin?
> 
> Yes but with igpu, to see if above 4g decoding would work.

ok ,i'll try

Koekieezz, 2023-03-13:
> > > i can not find disk drive untill setup_var 0x385 0
> > 
> > 
> > oh i think you should use iGPU, try to enable igpu, set display primary to igpu, and enable multi monitor/display on bios, save everything, then try to re do the thing i told for modgrub but this time using iGPU. the person i mentioned from winraid DID have the same problem as you but once he use igpu.. it worked fluently.
> > Here is the post: https://winraid.level1techs.com/t/experimental-resizable-bar-dxe-driver/89534/275?u=koekieezz Maybe you could ask that guy since he figure it out. also the bios i gave to you contains the latest orom, efigop and the gop vbt itself for coffeelake all revisions (up till 9900K), igpu should work. the patch i gave to you is modded for coffee lake (acpi table fix, fd unlock, 128GB ram unlock, and patch for 8+ thread for cpu) + uefipatch for 64GB allocation + dsdt fix (this done step by step from coffeemod to dsdt) changing from external reference to 0x7FFFFFFFFF.
> > > your mean is i need try to setup_var 0x385 1 gagin?
> > 
> > 
> > Yes but with igpu, to see if above 4g decoding would work.
> 
> ok ,i'll try if i need to enable this option [image]

couldnt you just enable it in bios?

jnyk, 2023-03-13:
@Koekieezz my BIOS can not show hole "System Agent (SA) Configuration" setting,so i can enable igpu us  modGRUBShell.efi or AMIBCP64.exe.
i flashed this bios into mainboard 
[image]

jnyk, 2023-03-13:
just now i try to enable IGPU and do setup_var or setup_var_3 0x385 0x0, reboot (type it and enter), then 0x385 0x1, reboot .
it seems that still not working properly

Koekieezz, 2023-03-13:
> just now i try to enable IGPU and do setup_var or setup_var_3 0x385 0x0, reboot (type it and enter), then 0x385 0x1, reboot . it seems that still not working properly Do I need to connect the monitor to igpu？

yes, that what i meant earlier. enable igpu but also connect display to motherboard.

jnyk, 2023-03-13:
@Koekieezz Thank you for your continued help. It's almost midnight,I had to sleep
I'll try your way tomorrow and then reply to the results.

xCuri0, 2023-03-14:
@jnyk are you able to boot linux with 4g decode ?

jnyk, 2023-03-14:
[image]
[image]
I connected the IGPU and video card at the same time, but I still can't find the boot hard drive

jnyk, 2023-03-14:
@xCuri0 I haven't tried linux, I'll try it now

xCuri0, 2023-03-14:
@jnyk So the issue is 4G decode makes the boot drive disappear ? I thought it was that Windows would freeze during boot or something.

This is identical to #35

jnyk, 2023-03-14:
Yes, the reason it can't enter Windows is that it can't find the boot hard drive

xCuri0, 2023-03-14:
@jnyk Is your Windows installation in GPT UEFI or legacy MBR ?

xCuri0, 2023-03-14:
@jnyk are you still able to boot USB drives ?

jnyk, 2023-03-14:
it able to discover usb drives, but cannot enter winpe.
I found that I only connected igpu it didn't show anything, maybe where else the settings were not correct

xCuri0, 2023-03-14:
@jnyk what about linux usb

is the usb drive formatted mbr or gpt ?

jnyk, 2023-03-14:
I need to make a linux startup disk, which takes a little time.
I found that the boot area of the usb boot disk is in gpt format and the system area is in mbr format

jnyk, 2023-03-14:
@xCuri0  I made a CentOS7 USB disk that looks like it can recognize the boot, but Iit can't get into the installation process.

xCuri0, 2023-03-14:
@jnyk can you show what error or anything your getting ?

jnyk, 2023-03-14:
on It stops in this interface without sending anything.

jnyk, 2023-03-14:
oh ， it works in igpu
[image]
when i try to switch display's input, it show

dormancygrace, 2023-03-14:
You need to use more recent dist, maybe centos 9 or Ubuntu 22

jnyk, 2023-03-14:
is that means I need one nvme driver?

xCuri0, 2023-03-14:
@jnyk Can you provide the output of ```lspci``` while you boot with iGPU ? It also needs to be with your dGPU in your system which I assume you do

I can then check if the BIOS had any issues assigning BARs or something

jnyk, 2023-03-15:
@xCuri0 OK I'll do what you suggest later

Koekieezz, 2023-03-15:
> it able to discover usb drives, but cannot enter winpe. I found that I only connected igpu it didn't show anything, maybe where else the settings were not correct

to use igpu u need to set primary display on bios to igpu. there should be such an option in the bios, do not use amibcp for it.

jnyk, 2023-03-15:
yes you're right ,I have set this two options like this
[image]

> > it able to discover usb drives, but cannot enter winpe. I found that I only connected igpu it didn't show anything, maybe where else the settings were not correct
> 
> to use igpu in bios, u need to set primary display on bios to igpu. there should be such an option in the bios, do not use amibcp for it.

jnyk, 2023-03-15:
When I installed Centos9, I was also unable to complete it  because I could not find the hard drive @xCuri0 
[image]

[image]

dormancygrace, 2023-03-15:
> When I installed Centos9, I was also unable to complete it because I could not find the hard drive @xCuri0 [image]
> 
> [image]

You need just run not install

jnyk, 2023-03-15:
how can i  run without install , Sorry, I don't know much about Linux @dormancygrace

[105 more replies]