xCuri0/ReBarUEFI #43: Verification of rebar support and request to update wiki with last step only

acimanx, 2023-03-24:
Greetings,

My system specifications are Gigabyte Aorus Gaming K3, rx470 8gb, i5 8600k.

A few years ago, the motherboard got update from Gigabyte with 4g decoding rebar support, which was working fine at that time. However, with the release of new AMD drivers, this feature got disabled, and my GPU became unsupported.

I came across your tutorial, but I missed some steps while flashing the BIOS as I wanted to avoid any potential risks of damaging it. Instead, I opted for the .exe tool to increase the capacity to 8192mb (13), which allowed the bar to show up in GPU-Z. Then, I used a .reg file to enable it in the AMD software.

I am now seeking your assistance in verifying whether this solution is working as expected. 

If it is indeed functioning correctly, could you kindly update your wiki to reflect this so that other users can follow only the last step of the tutorial without the fear of damaging their BIOS?

I appreciate the work you have done on this project.

Best regards.

acimanx, 2023-03-24:
I find this quite interesting as my screenshot shows the same result as yours.

It would be greatly appreciated if you could provide any alternative methods to verify the functionality of rebar support, excluding AMD software and GPU-Z as they may potentially provide misleading information. 

I have scoured the internet but have been unsuccessful in finding any reliable methods for verifying rebar support, other than using game benchmarks which, in my opinion, is not a conclusive method.

xCuri0, 2023-03-27:
@acimanx GPU-Z or any other application that can show Vulkan memory information should work.

But the best way is always just benchmarking (DX12 or Vulkan) because your modding firmware to get better fps in games usually anyways

xCuri0, 2023-04-02:
There is now wiki section [Verifying Resizable BAR is enabled](https://github.com/xCuri0/ReBarUEFI/wiki/Flashing-modified-UEFI#verifying-resizable-bar-is-enabled)

closed 2023-04-02 by xCuri0

Donalaren, 2025-11-14:
How to verify if the RebarState is working in Linux? 
After building and running RebarState, set the value to 32. After rebooting and reopening the RebarState, I can see the `Current ReBarState 32/Unlimited`. But `nvidia-smi` still can't see my Tesla V100. `NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.`

Through `dmesg`  I can see:
```
[ 6584.529446] NVRM: This PCI I/O region assigned to your NVIDIA device is invalid:
               NVRM: BAR1 is 0M @ 0x0 (PCI:0000:73:00.0)
[ 6584.529754] nvidia: probe of 0000:73:00.0 failed with error -1
[ 6584.529802] NVRM: The NVIDIA probe routine failed for 1 device(s).
[ 6584.529804] NVRM: None of the NVIDIA devices were initialized.
[ 6584.531040] nvidia-nvlink: Unregistered Nvlink Core, major device number 236
```