xCuri0/ReBarUEFI #48: P9X79 PRO Enabling Above 4G Decoding leads to a black screen (no boot)

Raziel49, 2023-04-29:
Hi there,

I have a ASUS P9X79 PRO motherboard with an i7-4930K and 32GB of ram, gpu is a Sapphire RX 580 Nitro+ 8GB.

The issue is that, after having successfully added your .ffs module to the bios, and also enabled the bios submenu for Above 4G Decoding, and having enabled the 4G Decoding itself, the board POSTs just fine but it doesn't boot anything, it just stucks to a black screen.

If i disable the Above 4G option, i can boot to windows just fine. Also another thing is that adding your module to the bios, breaks the sata contoller somehow (it gives code 10 in device manager, sata controller is a Marvell 91xx, but this is not a problem as i am booting from a NVMe drive).

I didn't do the UEFIPatch part, and the program said that it cannot find anything to patch. 

Also i didn't do the DSDT patching step as you indicated in the guide that HEDT boards do not need it.

(But i still tried to look in the extracted DSDT file from the AmiBoardInfo module if i could find any of the lines you pointed to in the guide, and i couldn't find anything that matched)

I tried also updating the GOP in the vga bios to the last version, but to no avail.

xCuri0, 2023-04-29:
@Raziel49 

You need to disable CSM

>  GPU is a Sapphire RX 580 Nitro+ 8GB

I'm also using the same GPU and GOP doesn't work when using modded VBIOS unless you use "magic GOP" which is incompatible with secure boot. So try with stock VBIOS and if it works then you can try magic GOP if you need modded

Raziel49, 2023-04-29:
Yeah i already reverted to the stock bios. 

Forgot to mention, obviously the CSM is off.

xCuri0, 2023-04-29:
@Raziel49 

Try setting these BIOS settings:

**4G Decode**: On
**Full Screen LOGO Show**: On
**CSM Support**: Never
**Boot Mode Selection**: UEFI
**Storage Boot Option Control**: UEFI Only
**Display Boot Option Control**: UEFI Only
**PCI Device ROM Priority**: UEFI OpRom

Raziel49, 2023-04-29:
I did already try all of them, i even tried changing mmioh size from 64gb to 128gb but with no results.

xCuri0, 2023-04-29:
@Raziel49 are you able to boot a linux live usb with 4g decoding ?

Raziel49, 2023-04-29:
@xCuri0 I will try now, will let you know

Raziel49, 2023-04-29:
Just tried 3 different linux distros, ubuntu, xubuntu and kali linux, but i can't boot in any of them. The bootloader works, but as soon as one selects "install linux" or "try linux without installing" i get the same black screen as if i was trying to boot windows 10.

Raziel49, 2023-04-29:
Of course all of this is with Above 4G option enabled, with the option disabled i can boot no problem.

I think that there is some problem with the motherboard bios and the Above 4G function, maybe it was not well implemented, there has to be a reason if it was hidden by default.

xCuri0, 2023-04-29:
@Raziel49 try to get a log of the kernel output somehow

Raziel49, 2023-04-29:
Ok, do you have any suggestion?

How can i record that output?

xCuri0, 2023-04-30:
@Raziel49 try adding kernel option "earlyprintk=efi".

Also can try disabling acpi to see if it fixes with "acpi=off" kernel option.

worst cause u will need to debug over serial port if no other way gives you early kernel output

Squall-Leonhart, 2023-05-03:
These asus x79 motherboards have a hidden MMIOH setting that defaults to 64GB, it might be a factor.

You need to enable the second pcie settings panel in a bios editor to access it.

[image]

xCuri0, 2023-05-03:
@Squall-Leonhart is that the MMIOH size or base ? Either way it should be fine unless you have 64GB+ RAM or a GPU with 32GB+ BAR

Raziel49, 2023-05-03:
I remember trying all of these different MMIOH values but with no result. 

Sadly i can't continue trying things as i don't own the board anymore.

franckythebest, 2023-05-03:
> 

hello i am on rampage iv extreme and the mmioh parameter is set to 64G by default. when I lower it to 32G, the PC reboots permanently.

franckythebest, 2023-05-06:
> Ces cartes mères asus x79 ont un paramètre MMIOH caché qui est par défaut de 64 Go, cela peut être un facteur et doit être réduit pour prendre en charge ReBar sur cette plate-forme.
> 
> Vous devez activer le panneau des paramètres de l'agent système caché (configuration IOH) dans un éditeur de bios pour y accéder.
> 
> [image]

Hello squall-leonhart my shore card is configured on mmioh 64g by default. when I decrease the value at 32G, there is a black screen while when I put it on 128G, my pc starts correctly. do you have any idea of ​​the cause?

xCuri0, 2023-05-06:
@franckythebest so 4g decode works with 128g ?

franckythebest, 2023-05-08:
> @franckythebestdonc le décodage 4g fonctionne avec 128g ?

I haven't tried to activate "ABOVE 4G". the pc starts correctly with a "mmioh" value of 64G and 128G, but a value less than 64G the pc does not start. you think that the "mmioh" value can influence the "ABOVE 4G" option.

xCuri0, 2023-05-08:
@franckythebest possibly, i don't have x79 to know

franckythebest, 2023-05-15:
I have a question. I read on the forum that it is necessary to modify the DSDT of the bios which could be responsible for the black screen. But is it for windows or MAC?.

xCuri0, 2023-05-15:
@franckythebest if it happens on linux even with `acpi=off` then it's not an ACPI or DSDT issue

ThatGuy-Bob, 2023-05-16:
Im getting a similar issue on a Chinese X79 board with a 3930K and 16gb or 32gb of UDIMM memory.

With an RX 5500 XT and Above 4G Decoding enabled the system freezes at kernel init for Windows or Linux.  When windows tries to start, the loading ring is massively corrupted.  The UEFI is text mode on this board, and there is also minor corruption (a few characters 'sticking' on the screen) when flipping through the pages with the option enabled as well.

When using an Arc A770 the screen is frozen and looks like a "magic-eye" poster when Above 4G Decoding is enabled.

Even without a GPU in the system it still freezes at the same part of the boot process.  The debug display steps from b6 to b1 and it freezes.

UEFI has options for serial debug output, but I have not dug into that further to see if it is working or leftover code at this point.

The behavior makes it seem as if the memory mapping is incorrect for the Above 4G option and devices are directed to memory buffer locations that simply do not exist.

I do have a Chinese dual socket 2011-0 board coming in the mail and I have a pair of E5-2697 V2s and 128gb of REG ECC memory for it.  Will have to compare the modules in the firmware to see if they match, as well as testing if 32/64/128gb of RAM changes anything.

xCuri0, 2023-05-16:
@ThatGuy-Bob are you sure csm off and everything legacy off ?

ThatGuy-Bob, 2023-05-16:
CSM was absolutely off, its a two step process on these boards.

Ive been so far down the rabbit hole that Ive even tried patching DSDT info, heh.

xCuri0, 2023-05-16:
@ThatGuy-Bob you need to connect a serial device (such as ch341a) to the motherboard serial port pins and set the linux kernel to use that. don't think it needs any bios settings besides just enabling serial port

xCuri0, 2023-05-16:
@ThatGuy-Bob you don't need debug information about uefi, just from linux kernel

ThatGuy-Bob, 2023-05-16:
OK, need kernel boot parameters if its something other than "console=ttyS0,9600n8".  Have the HW setup, building a Debian live USB.

ThatGuy-Bob, 2023-05-16:
All I get with 4G Decode enabled is this before it dies...

`[    0.091944] x86/cpu: VMX (outside TXT) disabled by BIOS`

xCuri0, 2023-05-16:
@ThatGuy-Bob so i think we've figured out the issue is efifb. you could try `video=efifb:off` and seeing if that gets it booting.

will need to figure out what's wrong with efifb and how it can be fixed

ThatGuy-Bob, 2023-05-16:
Its pretty universal.  Windows uses something similar for its boot screen and it also fails with corruption visible on screen.

For some reason 4G Decode absolutely breaks access to the EFI display.  Like I said earlier, there is corruption in the UEFI as well with it on.

xCuri0, 2023-05-16:
does disabling the efi framebuffer using that command line option get it working ? if it works like that and gpu works in games and stuff also then it's confirmed to just be an efifb issue

ThatGuy-Bob, 2023-05-16:
Yeah, its actually booting.  Gonna be a bit, USB 2.0 and all.

ThatGuy-Bob, 2023-05-16:
Dang, forgot that Debian keeps their basic live images clean, no black box firmware = no drivers loaded.  Grabbing Ubuntu.

ThatGuy-Bob, 2023-05-16:
OK, this is the wildest thing Ive seen in a LONG time, but bear with me as I take a left turn...

HD4350 + CSM Legacy VBIOS + 4G Decoding... works.

HOWEVER, there is no UEFI FBDEV.  Windows boots, but the boot screen looks like XP/Server 2003.  Desktop comes up in 640x480 greyscale only...

xCuri0, 2023-05-16:
Does the AMD graphics driver work?

Also can you send output of `cat /proc/cpuinfo` while your able to ?

ThatGuy-Bob, 2023-05-16:
In Windows 11 the RX 5500 XT is present in device manager.  Code 43, I disable/enable and it comes up, but the driver crashes.

xCuri0, 2023-05-16:
@ThatGuy-Bob I think the BIOS is setting incorrect BAR address or cache settings. Try see what Linux says when running like that since it will tell more than just Code 43

ThatGuy-Bob, 2023-05-16:
OK, over serial console it gets to DRM, then dies.  Going to disable modesetting.

ThatGuy-Bob, 2023-05-16:
cat /proc/cpuinfo

> processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 45
model name      : Intel(R) Core(TM) i7-3930K CPU @ 3.20GHz
stepping        : 7
microcode       : 0x71a
cpu MHz         : 3200.000
cache size      : 12288 KB
physical id     : 0
siblings        : 12
core id         : 0
cpu cores       : 6
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx lahf_lm epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid xsaveopt dtherm ida arat pln pts md_clear flush_l1d
vmx flags       : vnmi preemption_timer invvpid ept_x_only ept_1gb flexpriority tsc_offset vtpr mtf vapic ept vpid unrestricted_guest ple
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit mmio_unknown
bogomips        : 6385.07
clflush size    : 64
cache_alignment : 64
address sizes   : 46 bits physical, 48 bits virtual
power management:

ThatGuy-Bob, 2023-05-16:
lspci -v result for the 5500 XT

> 04:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Upstream Port of PCI Express Switch (rev c5) (prog-if 00 [Normal decode])
        Flags: bus master, fast devsel, latency 0, IRQ 29
        Memory at fbb00000 (32-bit, non-prefetchable) [size=16K]
        Bus: primary=04, secondary=05, subordinate=06, sec-latency=0
        I/O behind bridge: d000-dfff [size=4K] [16-bit]
        Memory behind bridge: fba00000-fbafffff [size=1M] [32-bit]
        Prefetchable memory behind bridge: 3fffc0000000-3fffd01fffff [size=258M] [32-bit]
        Capabilities: [48] Vendor Specific Information: Len=08 <?>
        Capabilities: [50] Power Management version 3
        Capabilities: [58] Express Upstream Port, MSI 00
        Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [150] Advanced Error Reporting
        Capabilities: [270] Secondary PCI Express
        Capabilities: [320] Latency Tolerance Reporting
        Capabilities: [370] L1 PM Substates
        Capabilities: [400] Data Link Feature <?>
        Capabilities: [410] Physical Layer 16.0 GT/s <?>
        Capabilities: [440] Lane Margining at the Receiver <?>
        Kernel driver in use: pcieport

05:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Downstream Port of PCI Express Switch (prog-if 00 [Normal decode])
        Subsystem: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Downstream Port of PCI Express Switch
        Flags: bus master, fast devsel, latency 0, IRQ 31
        Bus: primary=05, secondary=06, subordinate=06, sec-latency=0
        I/O behind bridge: d000-dfff [size=4K] [16-bit]
        Memory behind bridge: fba00000-fbafffff [size=1M] [32-bit]
        Prefetchable memory behind bridge: 3fffc0000000-3fffd01fffff [size=258M] [32-bit]
        Capabilities: [50] Power Management version 3
        Capabilities: [58] Express Downstream Port (Slot-), MSI 00
        Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [c0] Subsystem: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Downstream Port of PCI Express Switch
        Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [150] Advanced Error Reporting
        Capabilities: [270] Secondary PCI Express
        Capabilities: [2a0] Access Control Services
        Capabilities: [400] Data Link Feature <?>
        Capabilities: [410] Physical Layer 16.0 GT/s <?>
        Capabilities: [440] Lane Margining at the Receiver <?>
        Kernel driver in use: pcieport

06:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Navi 14 [Radeon RX 5500/5500M / Pro 5500M] (rev c5) (prog-if 00 [VGA controller])
        Subsystem: Micro-Star International Co., Ltd. [MSI] Navi 14 [Radeon RX 5500/5500M / Pro 5500M]
        Flags: fast devsel, IRQ 11
        Memory at 3fffc0000000 (64-bit, prefetchable) [disabled] [size=256M]
        Memory at 3fffd0000000 (64-bit, prefetchable) [disabled] [size=2M]
        I/O ports at d000 [disabled] [size=256]
        Memory at fba00000 (32-bit, non-prefetchable) [disabled] [size=512K]
        Expansion ROM at fba80000 [disabled] [size=128K]
        Capabilities: [48] Vendor Specific Information: Len=08 <?>
        Capabilities: [50] Power Management version 3
        Capabilities: [64] Express Legacy Endpoint, MSI 00
        Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [150] Advanced Error Reporting
        Capabilities: [200] Physical Resizable BAR
        Capabilities: [240] Power Budgeting <?>
        Capabilities: [270] Secondary PCI Express
        Capabilities: [2a0] Access Control Services
        Capabilities: [2b0] Address Translation Service (ATS)
        Capabilities: [2c0] Page Request Interface (PRI)
        Capabilities: [2d0] Process Address Space ID (PASID)
        Capabilities: [320] Latency Tolerance Reporting
        Capabilities: [400] Data Link Feature <?>
        Capabilities: [410] Physical Layer 16.0 GT/s <?>
        Capabilities: [440] Lane Margining at the Receiver <?>
        Kernel modules: amdgpu

ThatGuy-Bob, 2023-05-16:
Is the memory hole supposed to be there in the MTRR when using Above 4G Decoding in a native 64-bit environment?

cat /proc/mtrr gives me this...
```
root@ubuntu:/proc# cat /proc/mtrr
reg00: base=0x000000000 (    0MB), size= 2048MB, count=1: write-back
reg01: base=0x080000000 ( 2048MB), size= 1024MB, count=1: write-back
reg02: base=0x0be000000 ( 3040MB), size=   32MB, count=1: uncachable
reg03: base=0x100000000 ( 4096MB), size= 4096MB, count=1: write-back
reg04: base=0x200000000 ( 8192MB), size= 8192MB, count=1: write-back
reg05: base=0x400000000 (16384MB), size=16384MB, count=1: write-back
reg06: base=0x800000000 (32768MB), size= 1024MB, count=1: write-back

```

xCuri0, 2023-05-16:
@ThatGuy-Bob can you try the load the AMDGPU driver with serial console connected and see where it crashes ?

That should give most info

ThatGuy-Bob, 2023-05-17:
The 4350 is OLD, it uses Radeon.

5500 XT, as you can see above starts up in the "disabled" state, so no driver.
```
[    8.439348] ATOM BIOS: DDR3
[    8.453283] radeon 0000:03:00.0: VRAM: 1024M 0x0000000000000000 - 0x000000003FFFFFFF (1024M used)
[    8.471003] radeon 0000:03:00.0: GTT: 1024M 0x0000000040000000 - 0x000000007FFFFFFF
[    8.486292] [drm] Detected VRAM RAM=1024M, BAR=256M
[    8.496030] [drm] RAM width 64bits DDR
[    8.503536] [drm] radeon: 1024M of VRAM memory ready
[    8.513448] [drm] radeon: 1024M of GTT memory ready.
[    8.523368] [drm] Loading CEDAR Microcode
[    8.531436] [drm] Internal thermal controller without fan control
[    8.567926] [drm] radeon: dpm initialized
[    8.575992] [drm] GART: num cpu pages 262144, num gpu pages 262144
[    8.588779] [drm] enabling PCIE gen 2 link speeds, disable with radeon.pcie_gen2=0

```

xCuri0, 2023-05-17:
dmesg should tell why amdgpu won't load. what happens when trying to manually load ?

ThatGuy-Bob, 2023-05-17:
Live image does not have kernel modules, they are all compiled in.  Modprobe fails trying to find the module file.

ThatGuy-Bob, 2023-05-17:
Cant manually load the module, getting the system to boot disables some pre-reqs for it.

xCuri0, 2023-05-17:
dmesg should still tell why it's not loading the AMDGPU module

xCuri0, 2023-05-20:
@ThatGuy-Bob update ? just send the whole dmesg of when you can boot with HD 4350 but the 5500 XT isn't working. It should tell why it won't load the module

franckythebest, 2023-05-20:
y graphics card is an rx5700xt thicc iii and I'm having the same issue. the problem could come from there if I understand correctly.

SignalSpectre, 2023-06-24:
Hi there,
I got a similar problem with my system:

MB: Asus ROG Rampage IV Formula
CPU: Xeon E5 1620 v2
GPU: Aorus RX570 Gaming (4GB)

I was able to enable some hidden bios pages and activate 4G decoding from there.
MMIOH is set to 64GB as by default.

The system is booting and UEFI working with no issues, I can hear Windows starting up (some USB plugged sound) and I'm also able to start the shutdown process by pressing the power button (as I set up in the Windows settings).

The screen remains completely black though.

My Mobo has double bios with an hardware switch so if necessary I can experiment without risking it, is there something we can do to troubleshoot the issue?

xCuri0, 2023-06-24:
@SignalSpectre try and get a linux full dmesg log, it will tell why the gpu driver won't load

SignalSpectre, 2023-06-24:
@xCuri0 Got it, I'm creating a persistent live Linux USB so that I can put a startup script on it to get the dmesg automatically (otherwise I wouldn't be able to execute the command).
Hopefully I will soon be able to post the output of the command here.

SignalSpectre, 2023-06-24:
Finally got the dmesg output (took far more time than I thought...).
I looked at it very quickly an didn't see anything particularly weird but probably I don't even know what to look for (and also it's 1 am here so I need to get some sleep).

Here's it:
[2023_06_24_10_53_PM.txt](https://github.com/xCuri0/ReBarUEFI/files/11858363/2023_06_24_10_53_PM.txt)

xCuri0, 2023-06-25:
@SignalSpectre 4G decoding isn't enabled

```
[    0.305327] pci 0000:02:00.0: [1002:67df] type 00 class 0x030000
[    0.305337] pci 0000:02:00.0: reg 0x10: [mem 0xc0000000-0xcfffffff 64bit pref]
[    0.305344] pci 0000:02:00.0: reg 0x18: [mem 0xd0000000-0xd01fffff 64bit pref]
[    0.305348] pci 0000:02:00.0: reg 0x20: [io  0xe000-0xe0ff]
[    0.305352] pci 0000:02:00.0: reg 0x24: [mem 0xfbe00000-0xfbe3ffff]
[    0.305356] pci 0000:02:00.0: reg 0x30: [mem 0xfbe40000-0xfbe5ffff pref]
[    0.305361] pci 0000:02:00.0: enabling Extended Tags
[    0.305372] pci 0000:02:00.0: BAR 0: assigned to efifb
```

What happens when you set rebar to 4GB and boot Linux ?

SignalSpectre, 2023-06-25:
@xCuri0 mmmh, that's weird. It was activated in the bios and I got the usual black screen...
I'll do another test today.

The issue is with that option activated I cannot run the ReBarState exe to set the rebar size because I cannot see anything.

SignalSpectre, 2023-06-25:
It was my bad, I probably shut down the PC too soon and the file didn't get saved by the script I made (and I took one from the previous session which was with 4G decoding off).
Sorry again, it was 1 am and I was extremely tired...

This one is the right one:
[2023_06_25_06_57_AM.txt](https://github.com/xCuri0/ReBarUEFI/files/11859033/2023_06_25_06_57_AM.txt)

Inside that log you can see the amdgpu driver fails with these messages:

```
[    2.796609] amdgpu: 
                failed to send message 254 ret is 0
[    4.476521] amdgpu: SMU load firmware failed
[    4.476525] fbcon: Taking over console
[    4.476527] amdgpu: fw load failed
[    4.476529] amdgpu: smu firmware loading failed
[    4.476533] amdgpu 0000:02:00.0: amdgpu: amdgpu_device_ip_init failed
[    4.476537] amdgpu 0000:02:00.0: amdgpu: Fatal error during GPU init
[    4.476555] usb 2-1.2: new low-speed USB device number 4 using ehci-pci
[    4.476560] amdgpu 0000:02:00.0: amdgpu: amdgpu: finishing device.
[    4.476588] clocksource: Switched to clocksource tsc
[    4.477719] amdgpu: probe of 0000:02:00.0 failed with error -22
```

I post the Call Trace for more context on the crash:
```
[    4.478183] Call Trace:
[    4.478185]  <TASK>
[    4.478188]  amdgpu_uvd_suspend+0x202/0x2d0 [amdgpu]
[    4.478388]  uvd_v6_0_sw_fini+0x16/0xa0 [amdgpu]
[    4.478541]  amdgpu_device_fini_sw+0x116/0x410 [amdgpu]
[    4.478683]  amdgpu_driver_release_kms+0x16/0x30 [amdgpu]
[    4.478826]  drm_dev_release+0x28/0x50 [drm]
[    4.478854]  devm_drm_dev_init_release+0x38/0x60 [drm]
[    4.478874]  devm_action_release+0x15/0x20
[    4.478879]  release_nodes+0x40/0xb0
[    4.478882]  devres_release_all+0x9e/0xe0
[    4.478885]  really_probe.part.0+0x1ad/0x380
[    4.478888]  really_probe+0x40/0x80
[    4.478890]  __driver_probe_device+0x119/0x190
[    4.478893]  driver_probe_device+0x23/0xb0
[    4.478896]  __driver_attach+0xd3/0x1a0
[    4.478898]  ? __device_attach_driver+0x140/0x140
[    4.478901]  bus_for_each_dev+0x7e/0xd0
[    4.478903]  driver_attach+0x1e/0x30
[    4.478906]  bus_add_driver+0x178/0x220
[    4.478908]  driver_register+0x74/0xe0
[    4.478910]  __pci_register_driver+0x68/0x70
[    4.478914]  amdgpu_init+0x7c/0x1000 [amdgpu]
[    4.479055]  ? 0xffffffffc048e000
[    4.479058]  do_one_initcall+0x48/0x1e0
[    4.479062]  ? __cond_resched+0x19/0x40
[    4.479066]  ? kmem_cache_alloc_trace+0x15a/0x420
[    4.479069]  do_init_module+0x52/0x230
[    4.479074]  load_module+0x138d/0x1610
[    4.479078]  __do_sys_finit_module+0xbf/0x120
[    4.479081]  ? __do_sys_finit_module+0xbf/0x120
[    4.479085]  __x64_sys_finit_module+0x1a/0x20
[    4.479088]  do_syscall_64+0x5c/0xc0
[    4.479092]  ? fput+0x13/0x20
[    4.479095]  ? ksys_mmap_pgoff+0x14b/0x2a0
[    4.479098]  ? exit_to_user_mode_prepare+0x3d/0x1c0
[    4.479102]  ? syscall_exit_to_user_mode+0x27/0x50
[    4.479105]  ? __x64_sys_mmap+0x33/0x50
[    4.479107]  ? do_syscall_64+0x69/0xc0
[    4.479110]  ? do_syscall_64+0x69/0xc0
[    4.479113]  entry_SYSCALL_64_after_hwframe+0x61/0xcb
[    4.479116] RIP: 0033:0x7f653875e73d
[    4.479118] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 23 37 0d 00 f7 d8 64 89 01 48
[    4.479123] RSP: 002b:00007ffefb9c3218 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[    4.479126] RAX: ffffffffffffffda RBX: 0000561809c6f060 RCX: 00007f653875e73d
[    4.479128] RDX: 0000000000000000 RSI: 00007f653863eded RDI: 0000000000000018
[    4.479130] RBP: 0000000000020000 R08: 0000000000000000 R09: 0000000000000000
[    4.479132] R10: 0000000000000018 R11: 0000000000000246 R12: 00007f653863eded
[    4.479134] R13: 0000000000000000 R14: 0000561809c7be90 R15: 0000561809c6f060
[    4.479137]  </TASK>
[    4.479139] Modules linked in: pata_acpi amdgpu(+) mfd_aaeon iommu_v2 asus_wmi gpu_sched sparse_keymap i2c_algo_bit video drm_ttm_helper platform_profile mxm_wmi ttm crct10dif_pclmul crc32_pclmul ghash_clmulni_intel drm_kms_helper aesni_intel syscopyarea sysfillrect crypto_simd cryptd sysimgblt fb_sys_fops cec rc_core ahci i2c_i801 xhci_pci i2c_smbus e1000e lpc_ich libahci drm xhci_pci_renesas wmi
[    4.479161] CR2: ffffa1950093d000
[    4.479164] ---[ end trace c3daf93b69ffa6d8 ]---
```

SignalSpectre, 2023-06-25:
By further investigation in the amdgpu code base I fond that the 254 message is the PPSMC_MSG_LoadUcodes message.
Can this kind of operation be influenced by having 4G decoding enable or not? I cannot see the link...

xCuri0, 2023-06-25:
@SignalSpectre BIOS probably has configured the PCIe devices incorrectly which is kinda complex on HEDT systems, I have no idea how this stuff works though.

xCuri0, 2023-06-25:
Hopefully someone who knows more can figure this out

SignalSpectre, 2023-06-25:
Damn, I hoped it could be related to the GPU... That's unlucky.
Thanks anyway, hopefully someone will be able to figure this out in the future.

[264 more replies]