xCuri0/ReBarUEFI #50: The computer cannot enter bios and system after running ReBarState.exe

HowieHz, 2023-04-30:
**System**
 - Motherboard: [x99z 102v](https://xeon-e5450.ru/socket-2011-3/x99z-v102/)
 - BIOS Version: 
 - GPU: [蓝宝石 RADEON RX 5600 XT 6G D6 白金版PRO OC](https://www.sapphiretech.com.cn/zh-cn/consumer/pulse-pro-rx-5600-xt-6g-d6-oc_c)
 - CPU: E5-2667v4
 - SYSTEM: Windows10 22H2
 - [ ] CSM is turned off
 - [ ] 4G decoding is enabled
 - [x] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [x] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

(My English is not very good, so I use a translator)

What I have done?
1. Adding FFS module and Writing bios(I used freedos to write bios, but the screen did not display, I used the keyboard to type directly, then restarted the computer, the bios options were reset, I guess the write is successful)
2. I have restarted the computer several times, but I still can't find the option in the bios (I try to find Advanced-PCI Subsystem Settings-Above 4G Decoding) (Repeated restarts are also to verify that the bios is not damaged).
3. Because I saw in [this video](https://www.bilibili.com/video/BV1vd4y1s7KR/) that csm is turned on, as long as they all select uefi, I did not close csm as in the video.My system and graphics card are all uefi booted, and uefi is also checked in gpuz.
4. I ran [ReBarState.exe](https://github.com/xCuri0/ReBarUEFI/releases/download/0.3/ReBarState.exe) on the system, just like in the video link above, type 32 and press Enter. (Running the program failed before writing to the bios. This time the program is displayed as successful) 
5. I reboot the computer

**Current situation**
When I press the power button, all the fans are spinning, but the screen doesn't respond

1. There is no problem with the power supply, memory, cpu
2. I found my spare graphics card, gt610 (This card is not damaged, because I checked it the day before yesterday), and plugged it in. But it still doesn't show.
3. I unplugged all the hard drives and still couldn't display the bios.
4. I made a pe system with uefi boot and plugged it in, but it still didn't boot
5. I pulled out the battery from the motherboard and tried to reset the bios, but it still didn't work
6. The remote desktop is enabled on my system, but I cannot connect to it, and I cannot see the device in the router system. I think I have not entered the windows10 system

**My question** : What I should do next to save my computer?

Thank you for reading the issue
(I can upload the original bios file if needed)

xCuri0, 2023-04-30:
@HowieHz hi

GPU-Z reporting of CSM is often incorrect so the user in the Bilibili video probably actually had it disabled in the BIOS. Without disabling CSM you will get a black screen with 4G decoding on. Some motherboards still will eventually boot Windows and show video but most don't.

You also need to make sure 4G decoding is on before Resizable BAR like the ReBarState program warns you. If you can't find the option then 90% of the time [Enabling hidden 4G decoding](https://github.com/xCuri0/ReBarUEFI/wiki/Enabling-hidden-4G-decoding) will work. CSM **must** be disabled if you enable 4G Decoding

Your GT610 should work though since it's a non Resizable BAR GPU. Maybe it's possible that you actually disabled CSM (maybe I'm wrong because translation) and the GT610 doesn't support UEFI.

I've had similar situation happen with 4G decoding off and ReBarState set 13 but my PC still booted once I removed the GPU (I have iGPU).

Were you on the latest version of ReBarDxe downloaded from GitHub ? Version 0.3 and above contain CMOS reset detection but I don't think that's the issue since it would work with GT610 anyways.

If you were on latest 0.3 version I suggest trying your 5600 XT again after leaving CMOS battery out for a few minutes.

Last option you can try is using an external BIOS flasher (CH341A) to flash your BIOS to stock. Miyconst did a video explaining how to use it on these Chinese X99 boards [https://www.youtube.com/watch?v=4qX2zihB6UE](https://www.youtube.com/watch?v=4qX2zihB6UE).

HowieHz, 2023-04-30:
@xCuri0 Thank you very much for your reply

Unfortunately, I probably wasn't using the latest 0.3 version (I uploaded the files I was using to the [site](https://wwzt.lanzoul.com/iMlIR0ucagxc)), but a file from one of Bilibili's tutorial videos.

Thanks for your advice. I have placed an order for CH341A.After I receive the goods, I will try again to solve the problem.

closed 2023-04-30 by HowieHz

HowieHz, 2023-05-07:
> @HowieHz hi
>
> GPU-Z reporting of CSM is often incorrect so the user in the Bilibili video probably actually had it disabled in the BIOS. Without disabling CSM you will get a black screen with 4G [...]

@xCuri0 I tried three bios files, among which the original one and [x99zv102_iEngineer](https://xeon-e5450.ru/wp-content/plugins/download-attachments/includes/download.php?id=O9yyYfrqRMWQnxdqajakwNTJCx4kGkmG4CtT7D-8rT4,) still couldn't find 4G decoding options after adding ffs, and 4G Decod couldn't be searched using UEFITool.Finally, I used [x99zV102amodbio_VK239&kostiaprofrom](https://xeon-e5450.ru/wp-content/plugins/download-attachments/includes/download.php?id=XZhnKr5M2173ab05R9IHKdnxHEz_2phxgYRDrS3gNeo,), added ffs, and found 4G decoding options.I enabled it, and it worked!!!

You can find these bios files from the bottom of this [site](https://xeon-e5450.ru/socket-2011-3/x99z-v102/)

Thanks for what you did. :)