xCuri0/ReBarUEFI #51: [LINUX] ReBarState doesn't build with GCC v13.1.1

Sid127, 2023-05-14:
```
[sidpr@strogg build]$ make
[ 50%] Building CXX object CMakeFiles/ReBarState.dir/ReBarState.cpp.o
/home/sidpr/ReBarUEFI/ReBarState/ReBarState.cpp:89:9: error: ‘uint32_t’ does not name a type
   89 |         uint32_t attr;
      |         ^~~~~~~~
/home/sidpr/ReBarUEFI/ReBarState/ReBarState.cpp:15:1: note: ‘uint32_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
   14 | #include <sys/ioctl.h>
  +++ |+#include <cstdint>
   15 | #endif
/home/sidpr/ReBarUEFI/ReBarState/ReBarState.cpp:90:9: error: ‘uint8_t’ does not name a type
   90 |         uint8_t value;
      |         ^~~~~~~
/home/sidpr/ReBarUEFI/ReBarState/ReBarState.cpp:90:9: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
/home/sidpr/ReBarUEFI/ReBarState/ReBarState.cpp:97:1: error: ‘uint8_t’ does not name a type
   97 | uint8_t GetState() {
      | ^~~~~~~
/home/sidpr/ReBarUEFI/ReBarState/ReBarState.cpp:97:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
/home/sidpr/ReBarUEFI/ReBarState/ReBarState.cpp:111:17: error: ‘uint8_t’ was not declared in this scope
  111 | bool WriteState(uint8_t rBarState) {
      |                 ^~~~~~~
/home/sidpr/ReBarUEFI/ReBarState/ReBarState.cpp:111:17: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
/home/sidpr/ReBarUEFI/ReBarState/ReBarState.cpp: In function ‘int main()’:
/home/sidpr/ReBarUEFI/ReBarState/ReBarState.cpp:142:9: error: ‘uint8_t’ was not declared in this scope
  142 |         uint8_t reBarState;
      |         ^~~~~~~
/home/sidpr/ReBarUEFI/ReBarState/ReBarState.cpp:142:9: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
/home/sidpr/ReBarUEFI/ReBarState/ReBarState.cpp:151:9: error: ‘reBarState’ was not declared in this scope
  151 |         reBarState = GetState();
      |         ^~~~~~~~~~
/home/sidpr/ReBarUEFI/ReBarState/ReBarState.cpp:151:22: error: ‘GetState’ was not declared in this scope; did you mean ‘setstate’?
  151 |         reBarState = GetState();
      |                      ^~~~~~~~
      |                      setstate
/home/sidpr/ReBarUEFI/ReBarState/ReBarState.cpp:186:23: error: ‘WriteState’ cannot be used as a function
  186 |         if (WriteState(reBarState)) {
      |             ~~~~~~~~~~^~~~~~~~~~~~
make[2]: *** [CMakeFiles/ReBarState.dir/build.make:76: CMakeFiles/ReBarState.dir/ReBarState.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/ReBarState.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```
using Gnu Make 4.4.1, GCC 13.1.1, on Arch Linux

closed 2023-05-14 by Sid127

xCuri0, 2023-05-14:
@Sid127 how did u fix it ?

Sid127, 2023-05-14:
Ah, sorry, forgot to put this here ^^'
```
diff --git a/ReBarState/ReBarState.cpp b/ReBarState/ReBarState.cpp
index 26d26de..40efed0 100644
--- a/ReBarState/ReBarState.cpp
+++ b/ReBarState/ReBarState.cpp
@@ -5,6 +5,7 @@ SPDX-License-Identifier: MIT
 #include <iostream>
 #include <string>
 #include <cmath>
+#include <cstdint>
 
 #ifdef _MSC_VER
 #include <Windows.h>
@@ -203,4 +204,4 @@ exit:
 exit:
        #endif
        return ret;
-}
\ No newline at end of file
+}
```

xCuri0, 2023-05-14:
@Sid127 you could submit a PR for it, or I could just do the changes

Sid127, 2023-05-14:
Feel free to go ahead, just including a single header anyway