xCuri0/ReBarUEFI #56: The guide is very confusing.

Ckosmick, 2023-05-24:
**System**
 - Motherboard:
 - BIOS Version:
 - GPU:
 - [ ] CSM is turned off
 - [ ] 4G decoding is enabled
 - [ ] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [ ] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

I don't know what I'm looking at in any of the screenshots. You're told to search for a GUID but UEFITool doesn't jump you there so you need to dig around all willy nilly.

There's screenshots that supposedly show the difference between what a good bios patch and a bad bios patch looks like but it does not clearly indicate where to look, only what to look for, a pad file, which there are many in the bios.

I grabbed NE UefiTool to try and see if it was verbose enough to make it less confusing and what I find is that I have a message in the UefiTool window of my patched rom that says 

FfsParser::checkProtectedRanges: AMI v2 protected range [290000h:5C6A00h] hash mismatch, opened image may refuse to boot

Is this borked or not? One can *assume* it is but the guide does not helpfully explain any of this in any way. It's frustrating

Ckosmick, 2023-05-25:
Update: I ignored the ami v2 protected range warning and tried flashing patched f22f anyway. Q-flash didn't work even after downgrading to f22c, so I used rufus to make a FreeDOS image on my usb, copied over my patched f22f BIOS and a modded version of efiflash. https://winraid.level1techs.com/t/flashing-gigabyte-while-avoiding-invalid-bios-image/31185

So far so good with 1GB BAR.

xCuri0, 2023-05-25:
The reason the guide is so confusing and many workarounds is due to UEFITool NE not supporting modification of UEFI images. So you have to use UEFITool 0.28 which has problems handling newer (~2015+) UEFI firmware and non-aligned pad files.

Once UEFITool NE supports modification the guide will become alot easier and no workarounds like MMTool will be needed.

If u have any suggestions for the guide you can mention them

closed 2023-05-25 by xCuri0

Ckosmick, 2023-05-25:
That sounds good, appreciate all the work done on this. 
[image]

Seems to work set to unlimited with the only issue being the inability to see bios before boot. That said, I have not modified this RX 6700 XT vBIOS in any way. Would connecting to iGPU work in this case? Or would I be looking to try and update GOP too?

xCuri0, 2023-05-25:
@Ckosmick see https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes)#no-video-output-during-boot-before-os-loads

Ckosmick, 2023-05-25:
Ahh, will try that, thanks. I currently have above 4G decoding on and csm off.

Ckosmick, 2023-05-25:
Good news and bad news, the bad news is the fixes in common issues don't work. turning above 4gb decoding off doesn't work.

However, good news, setting "initial display" to iGFX means that when I need to change bios I still can, just with a display connected to iGPU. And I still get above 4G decoding and unlimited resizable bar this way.

xCuri0, 2023-05-25:
@Ckosmick you supposed turn off the **Above 4G Decoding Patch** option, not **Above 4G Decoding**. it's possible that your bios doesn't have the option

Ckosmick, 2023-05-25:
I toggled off the "Above 4G decoding" setting in my bios, my system supports that, just not rebar (until now haha)

Enterprise12nx01, 2023-09-02:
I am also very confused..have Asus Z170I Pro Gaming, i see its suppose to work.
i did all the steps, it named my file (name).CAP.patched
i removed the patched extension and just renamed it, but it wont let me install it as its not a recognized bios or whatever.

Opening the modified bios it says "parseImageFile: Aptio capsule signature may become invalid after image modifications"
i tried to understand the "pad" file thing but i have no idea what im looking or comparing to, if its needed and how i can flash it

xCuri0, 2023-09-02:
@Enterprise12nx01 

> parseImageFile: Aptio capsule signature may become invalid after image modifications

this is safe to ignore.

ASUS almost always has pad file issue though, you can try mentioning someone in the working motherboards list who has something similar for help

Enterprise12nx01, 2023-09-02:
> @Enterprise12nx01
> 
> > parseImageFile: Aptio capsule signature may become invalid after image modifications
> 
> this is safe to ignore.
> 
> ASUS almost always has pad file issue though, you can try mentioning someone in the working motherboards list who has something similar for help.
> 
> Once UEFITool NE gets released with editing support I can simplify the guide a lot since we won't need workarounds for issues like this.

@X3Shadowknight would you be willing to help? I have the same motherboard you marked as working! I would really appreciate it

Enterprise12nx01, 2023-09-05:
@68k-dude @dsanke @necr0n @Nimayer @Aushyro @Phil-Libra
sorry for the tags, i saw some of you have done Z170 boards, i was hoping somebody could help, pretty sure im good up until the pad part of the guide, and also flashing it as it seems ezflash3 will deny it

dsanke, 2023-09-13:
@Enterprise12nx01 you need a spi programmer(hardware) to flash modified/unofficial bios, this is the most convenient way.
i can provide you modified bios if you are able to flash unofficial bios.

Enterprise12nx01, 2023-09-14:
> @Enterprise12nx01 you need a spi programmer(hardware) to flash modified/unofficial bios, this is the most convenient way. i can provide you modified bios if you are able to flash unofficial bios.

Ooof no other way? i will just settle without it ig, im not even sure how to do that