xCuri0/ReBarUEFI #57: May I use it as a hot patch in Linux?

FNsi, 2023-05-25:
If I can, may pls tell me how?

Laptop problem
😅

xCuri0, 2023-05-25:
Do the DSDT Patching "Linux no 4G decoding fix" but use the method described on Arch Wiki to extract and apply modified DSDT so you don't have to flash BIOS.

And also boot with `pci=realloc`

FNsi, 2023-05-25:
> Do the DSDT Patching on wiki "Linux no 4G decoding fix" but use the method described on Arch Wiki to extract and apply modified DSDT so you don't have to flash BIOS.
> 
> 
> 
> And also boot with `pci=realloc`

Thank u, I'd like to try soon.

FNsi, 2023-05-25:
> Do the DSDT Patching on wiki "Linux no 4G decoding fix" but use the method described on Arch Wiki to extract and apply modified DSDT so you don't have to flash BIOS.
> 
> 
> 
> And also boot with `pci=realloc`

What if... my bios is insyde...

closed 2023-05-25 by xCuri0

FNsi, 2023-05-25:
> @FNsi it doesn't matter 

Then how can I know which module should I extract....

And should rename the extracted module name to AmiBoardINfo.efi?

xCuri0, 2023-05-25:
@FNsi don't extract from bios, extracting from running system following arch wiki guide

FNsi, 2023-05-25:
> @FNsi don't extract from bios, extracting from running system following arch wiki guide 

Thank u, I gonna do it now.

FNsi, 2023-05-25:
> @FNsi don't extract from bios, extracting from running system following arch wiki guide 

I am sorry to brother again....

The code I see is 
```

If ((TOPM != Zero))
                {
                    BTX1 = TOPM /* \TOPM */
                }

                BTMX = (PCBA - One)
                BTLN = (PCBA - BTMN) /* \_SB_.PCI0._CRS.BTMN */
                BTN1 = (PCBL + One)
                BTL1 = (BTX1 - BTN1) /* \_SB_.PCI0._CRS.BTN1 */
                BTL1 += One
                If ((MM64 == One))
                {
                    CreateQWordField (CRES, \_SB.PCI0._Y02._MIN, M1MN)  // _MIN: Minimum Base Address
                    CreateQWordField (CRES, \_SB.PCI0._Y02._MAX, M1MX)  // _MAX: Maximum Base Address
                    CreateQWordField (CRES, \_SB.PCI0._Y02._LEN, M1LN)  // _LEN: Length
                    M1MN = HMB1 /* \HMB1 */
                    M1MX = HMM1 /* \HMM1 */
                    M1LN = HML1 /* \HML1 */
                    CreateQWordField (CRES, \_SB.PCI0._Y03._MIN, M2MN)  // _MIN: Minimum Base Address
                    CreateQWordField (CRES, \_SB.PCI0._Y03._MAX, M2MX)  // _MAX: Maximum Base Address
                    CreateQWordField (CRES, \_SB.PCI0._Y03._LEN, M2LN)  // _LEN: Length
                    M2MN = HMB2 /* \HMB2 */
                    M2MX = HMM2 /* \HMM2 */
                    M2LN = HML2 /* \HML2 */
                }

                M460 ("PLA-ASL-\\_SB.PCI0._CRS End\n", Zero, Zero, Zero, Zero, Zero, Zero)
                Return (CRES) /* \_SB_.PCI0.CRES */
            }

```

That really stuck me in...
Any suggestions 🙏?

FNsi, 2023-05-25:
That wired code let me assume the resize bar could be working direct in my system, 

Then I try to use Rebarstate, set value to 32 and reboot, now it looks like working, at least show in the ReBarstate current 32, but I see cpu-x still show resize bar disabled...

xCuri0, 2023-05-25:
@FNsi rebarstate doesn't matter you are not using rebardxe

the DSDT code shows your system probably already has a 4g decode option probably hidden (see wiki method) you just have to enable it

FNsi, 2023-05-25:
> @FNsi rebarstate doesn't matter you are not using rebardxe
> 
> 
> 
> the DSDT code shows your system probably already has a 4g decode option probably hidden (see wiki method) you just have to enable it

I know, in my system it's mimo and I check it in txt file show it can read 1t... but what should I do now?

xCuri0, 2023-05-25:
Just follow the enabling hidden 4g instructions on wiki

FNsi, 2023-05-25:
> Just follow the enabling hidden 4g instructions on wiki
> 
> 
> > @FNsi <https://github.com/FNsi> rebarstate doesn't matter you are not
> > using rebardxe
> >
> > the DSDT code shows your system probably already has a 4g decode option
> > probably hidden (see wiki method) you just have to enable it
> >
> > I know, in my system it's mimo and I check it in txt file show it can read
> > 1t... but what should I do now?

I mean, if the above 4g is the same as mimo, then I am already in it... and I cannot find any set name in above 4g decoding in my bios 😅

xCuri0, 2023-05-25:
@FNsi 4gb mmio is the same thing. 

It should give you rebar in Linux with `pci=realloc` kernel option.

If it doesn't then send your full dmesg

FNsi, 2023-05-25:
> @FNsi 4gb mmio is the same thing. 
> 
> 
> 
> It should give you rebar in Linux with `pci=realloc` kernel option.
> 
> 
> 
> If it doesn't then send your full dmesg

I suppose I should rebuild the kernel?

xCuri0, 2023-05-25:
@FNsi why rebuild, that's just a kernel command line option

xCuri0, 2023-05-25:
> I don't know it😂 sorry... and .. yes... how to do it?

It's different across distros just search Google or something

FNsi, 2023-05-25:
> It's different across distros just search Google or something
> 
> 
> > @FNsi <https://github.com/FNsi> why rebuild, that's just a kernel command
> > line option
> >
> > I don't know it😂 sorry... and .. yes... how to do it?

I assume it's like config***-generic in /boot? All I need to do is add a line called pci=realloc in that file?

histfuture, 2024-10-16:
> Do the DSDT Patching on wiki "Linux no 4G decoding fix" but use the method described on Arch Wiki to extract and apply modified DSDT so you don't have to flash BIOS.
> 
> And also boot with `pci=realloc`

Does this mean I do not flash my bios if I use linux (i.e. skip UEFITool and UEFIPatch part) . Just do dsdt patching is  enough to open my resizable bar?