xCuri0/ReBarUEFI #59: Enabled CSM with the modded BIOS and I lost all access to the UEFI setup

gmiranda, 2023-05-27:
**System**
 - Motherboard: Asus X99-S
 - BIOS Version: 4101
 - GPU: AMD Radeon 5700 XT
 - [] CSM is turned off
 - [X] 4G decoding is enabled (had to mod)
 - [X] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [X] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

So I flashed the modded BIOS according to your instructions and it works all well under Windows. However, I wanted to boot a non-uefi formatted USB drive and had to enable CSM. Now it boots directly into Windows, showing a Windows 7-esque loading screen and completely skipping GRUB, which is my preferred boot option. It's just a blank screen, then a text message saying something like "Booting the operating system" and then the Windows loading screen.

I can't get into the setup either via DEL key or from Windows.

Is there anything I can do besides clearing CMOS (which may not work) or reflashing?

Thanks!

xCuri0, 2023-05-27:
You can use setup_var tool to edit bios settings, you can boot USB devices from windows 10 advanced startup

Or just reset cmos

closed 2023-05-27 by xCuri0

gmiranda, 2023-05-27:
Was this the expected behavior for enabling CSM?

Seems like it didn't like the reset cmos... Now I get a D4 boot code. I guess I'll have to reflash again

xCuri0, 2023-05-27:
@gmiranda yes it is even on stock bios, csm is incompatible with 4g decoding

gmiranda, 2023-05-27:
@xCuri0 I meant this weird boot behavior where I can't enter the UEFI setup?

I almost pooed. I could no longer boot with my 5700XT plugged in ever after flashing back the non-patched BIOS. I had to plug in a GT 610 (with no uefi support) and then I was able to go back to the 5700XT.

There's one thing I didn't mention: I enabled CSM in the first place to boot with the GT610 as a 2nd GPU. It must have been related to this.

Thanks!

xCuri0, 2023-05-28:
Yes 4g decoding is incompatible with CSM