xCuri0/ReBarUEFI #60: Weird CPU support

GiulianoB-1, 2023-06-14:
not a bug but a question. i have a motherboard ROG strix x299-E, a i7-7740x cpu and rtx 3070 FE. My gpu and motherboard support resizable bar tho according to intel, it seems that my cpu isn't supported. However, when I enable resizable bar in my bios, nvidia control panel and GPU-z reports resizable bar as enabled. So is cpu support really is needed or not? and if cpu supported is needed, do i need to use the ReBarUEFI too to fix it?

xCuri0, 2023-06-14:
Any CPU from ~2010 should be able to support rebar even if it's PCIe 2.0. I'm unsure about earlier stuff because no one has tried it.

closed 2023-06-14 by xCuri0

GiulianoB-1, 2023-06-15:
So I dont need to use ReBarUEFI?

xCuri0, 2023-06-15:
@GiulianoB-1 if you're motherboard has rebar option in BIOS no, because you can use that instead