xCuri0/ReBarUEFI #62: rebar turn off after reset bios and i cant reactive

yaggoto, 2023-06-22:
**System**
 - Motherboard: machinist x99 rs9
 - BIOS Version: original bios modded turbo unlock and rebar
 - GPU: rx 6600
 - [y ] CSM is turned off
 - [y ] 4G decoding is enabled
 - [ ] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [ ] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

I was using rebar "on" during a week, my pc was freezing so i reset bios. now rebar is off and ReBarState.exe isnt working proprely. show Successful message but after reboot, system no boot!

xCuri0, 2023-06-22:
Enable 4G decoding and disable CSM

yaggoto, 2023-06-22:
Solved! thx!!

closed 2023-06-22 by xCuri0