xCuri0/ReBarUEFI #63: Enable 4G Decoding for Tesla P40 none of the described methods worked until now

SirNoName2705, 2023-06-24:
**System**
 - Motherboard: [GA-Z97P-D3 rev. 1.1](https://www.gigabyte.com/Motherboard/GA-Z97P-D3-rev-11/sp#sp) from Gigabyte
 - BIOS Version: [F9b](https://download.gigabyte.com/FileList/BIOS/mb_bios_ga-z97p-d3_f9b.zip?v=c982f3f352bd854fd100f81f693cd2e0)
 - GPU: [Nvidia Tesla P40](https://www.nvidia.com/content/dam/en-zz/Solutions/design-visualization/documents/nvidia-p40-datasheet.pdf)
 - CPU: [Intel I7 4770](https://www.intel.de/content/www/de/de/products/sku/75122/intel-core-i74770-processor-8m-cache-up-to-3-90-ghz/specifications.htmls) (Haswell)
 - OS: Ubuntu 22.04 (flavor is Kubuntu)
 - [X] CSM is turned off, but sometimes it does turn it on again, automatically
 - [ ] 4G decoding is enabled
 - [X] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [X] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
Background:
We want to use an old PC as a server for large language models; hence, I established an SSH connection from my working PC to the server PC. But I do have physical access to the llm-server-pc. 
Problem Drivers for Tesla P40 were not working, and I found out, that Above 4G decoding is needed. I have tried everything on the driver side and after 2 weeks of trying everything lead me to see that the graphics card cannot be addressed properly. I saw a lot of errors that said something like: 

> [ 2805.584742] NVRM: This PCI I/O region assigned to your NVIDIA device is invalid:
               NVRM: BAR1 is 0M @ 0x0 (PCI:0000:01:00.0)
[ 2805.585711] nvidia: probe of 0000:01:00.0 failed with error -1
[ 2805.585725] NVRM: The NVIDIA probe routine failed for 1 device(s). 

Finally, it has led me to this repo. Now I have been trying to get this working with UEFI modification using tutorials from this repo. I started with that about one week ago.

Goal: We really do not need ReBar, but it would be nice. What we really need is Above 4G decoding enabled so that we can use the Graphics card for ML.

Issues:
PC won't boot after I applied the UEFI Patch with DSDT Patch. If I start it, there is no video output at all. Hence, I cleared the CMOS and will try other solutions.
I am pretty convinced that I messed up something, as I find the tutorial perplexing. But I think it is more on my side than on your side, as I sincerely appreciate your help and all the efforts you all have put into the wiki.

I decided to document very thoroughly all the things I have tried. I will go through all the tutorials a second time and make retroactively screenshots of everything.

# Here is what I have done, in chronological order:

## 1. [linux-no-4g-decoding-fix](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching#dsdt-broken-4g-decoding-fix-haswell-and-up--linux-no-4g-decoding-fix) (Wrong!)
The first thing I tried was the DSDT Only fix. But I think I messed up this part quite good. Therefore, I do not recommend following these steps. I saw a link on the [DSDT-Patching site](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching#dsdt-broken-4g-decoding-fix-haswell-and-up--linux-no-4g-decoding-fix) there was a Link leading to an Arch wiki page.
**[... override DSDT without any BIOS modification on Linux ...](https://wiki.archlinux.org/title/DSDT#Using_modified_code)**
I started by reading the arch wiki page from the beginning, after I have read the part [2. Recompiling it yourself](https://wiki.archlinux.org/title/DSDT#Recompiling_it_yourself), I started to implement it as well.
In hindsight, I now see that I should not have done that because there was a reason that the link on the [DSDT-Patching site](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching#dsdt-broken-4g-decoding-fix-haswell-and-up--linux-no-4g-decoding-fix) was leading me to [Part 3. of the tutorial](https://wiki.archlinux.org/title/DSDT#Using_modified_code) and not to the [second part of the arch wiki tutorial](https://wiki.archlinux.org/title/DSDT#Recompiling_it_yourself).
<details><summary>Details</summary>
<p>

I decompiled the DSDT-file, changed the version number from 0x0...088 to 0x0... and compiled it. Of course, I had some errors.
<details><summary>How I have changed the Version number</summary>
<p>

I changed this line:
[image]
to this:
[image]

</p>
</details> 

```shell
sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.dat
iasl -d dsdt.dat
iasl -tc dsdt.dsl
iasl -ve -tc dsdt.dsl (This one is only for documentation)
```
> Intel ACPI Component Architecture
> ASL+ Optimizing Compiler/Disassembler version 20200925
> Copyright (c) 2000 - 2020 Intel Corporation
> 
> Compiler aborting due to parser-detected syntax error(s)
> dsdt.dsl   6751:             Zero
> Error    6126 -   syntax error ^ 
> 
> Input file:    dsdt.dsl - Compilation aborted due to parser-detected syntax error(s)
> Hex Dump:      dsdt.hex -  641565 bytes
> 
[... 856 lines cut ...]
I then cleared the CMOS by removing the battery, and hence the bios is now on the latest working version, which is the 9b as before.

I have appended a zip file containing images of my BIOS config. I cannot embed them and do not know why.
[BiosConfig.zip](https://github.com/xCuri0/ReBarUEFI/files/11858097/BiosConfig.zip)
</p>
</details> 

## Miscellaneous
I know it is a long description, but I hope that it helps someone.
I also hope that it is not too bad to read. I tried my absolute best to organize it and use the supported functions to bring structure into my description. (mostly details)
If anyone has any plan how to fix that, I would much appreciate your help.
Thanks to all of you who helped to build this repo and its wiki.

# Here is what I plan to do:
If anyone of you has an educated guess as to what to try next, please let me know.
1. [X] write an Issue in case I miss something obvious to other people
2. [ ] See if any of the previous versions of the bios has hidden 4G decoding
3. [ ] Brutforce the placement of the code of the DSDT patch (I mean trying all sensible placement of the Tutorial code)
4. [ ] Use the Linux only version of the created DSDT using half of your Tutorial and half of the Arch Linux wiki tutorial
5. [ ] Using the uefi flash mod from [win raid](https://winraid.level1techs.com/t/flashing-gigabyte-while-avoiding-invalid-bios-image/31185#msg24127) even I thought I did not get the error described there.
6. [ ] Use the older mb Z87-G41 PC Mate from msi, which I have lying around.
7. [ ] Trying to change this TOLUD thing.
8. [ ] Using the older version of UEFIExtract and Retry 3.
9. [ ] Try to flash the Patched UEFI without the DSDT Haswell Fix
10. [ ] Try to compile the DSDT again to see if I accidentally compiled it on the client PC

xCuri0, 2023-06-25:
@SirNoName2705 DSDT Modification will fix it but only for Linux, you don't need to flash BIOS just follow the Arch wiki guide for overriding it.

Your BIOS probably getting corrupt from pad file problem which needs MMTool to fix

The Z87 likely won't work

SirNoName2705, 2023-06-25:
So, then it was right that I extracted the DSDT like it was described on the Arch wiki page?
Then I would redo that again, apply the fix from your Tutorial and modify grub again.
I write back when I have done that.

SirNoName2705, 2023-06-25:
Okay I did the following. I changed the dsdt file like it was described in my last post.
Then I checked the kernel Config
<details><summary>Kernel Config</summary>
<p>

[image]

</p>
</details> 
which looked fine.
So I Followed this part:

[image]

Except for not having a `ssdt1.aml`, as this was not described by the tutorial. I could get this if needed via `apcidump`.

When I have made the archive and loaded it in my grub config.
I named the archive acpi_override and placed the following line into the grub menu entry:
`initrd	/boot/acpi_override2`
<details><summary>Details</summary>
<p>

[image]

</p>
</details> 

this would boot into my OS. Sadly, with `dmesg | grep ACPI` revealed not the same output as described in the arch wiki. And also under DSDT it showed the old version of the dsdt, as it was 0x...880 and my dsdt file had the version 0x.....890. So the arch wiki way does not work to load my dsdt file. But maybe I am missing something here.

Then I googled and found out there is [another way](https://askubuntu.com/questions/1027552/including-a-custom-acpi-dsdt-with-kubuntu-18-04-rc1) of loading a custom dsdt. So I removed the previous line.
 Then I copied my compiled dsdt.aml file into the /boot folder and renamed it to `modded_dsdt.aml`.
Then I added the following to my grub menu entry:
`acpi /boot/modded_dsdt.aml`
<details><summary>The Whole Grub menu entry</summary>
<p>

[image]

</p>
</details> 

But this won't work. Every time I start it, it will reboot to the Grub menu.

Do you have any Idea of what I could have done wrong?
Maybe I did wrong by replacing the existing code of my dsdt with the code from your tutorial, but then again, I do not know where to put it then.

xCuri0, 2023-06-25:
@SirNoName2705 I don't have Linux installed anymore but I used the initrd method successfully with it

SirNoName2705, 2023-06-25:
Okay, which OS did you used?

Could you maybe tell me If my replacement of the original code in the dsdt was right?

xCuri0, 2023-06-25:
Ubuntu

SirNoName2705, 2023-06-25:
I think I got it now. But to my dismay, the card died.

closed 2023-06-26 by xCuri0

Nhayes05, 2024-02-16:
I reckon I have this same problem. hp z230 sff, I7 4790, lga 1150, win 10, c226, newest BIOS version. I did check the oldest BIOS I could find as well, no dice. This seems to be my only real option, but I don't know if a ton of time for only 1gb decoding would be worth it in the slightest. My card is an asrock a380lp with 6gb.