xCuri0/ReBarUEFI #64: Not working on Memphis2-S

kaitlyndotmoe, 2023-06-30:
**System**
 - Motherboard: Memphis2-S (HP Pavilion 550-153w)
 - BIOS Version: A0.17
 - GPU: NVIDIA GTX 1050 Ti 4GB (yes, I'm well aware my GPU is not officially ReBar supported.)
 - [X] CSM is turned off
 - [X] 4G decoding is enabled
 - [X] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [X] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

I did everything correctly, enabled 4G decoding, etc. But attempting to turn on the ReBar DXE via ReBarStatus does not work (returns GetLastError: 5). Checking the wiki says I need the NVRAM whitelist patch, but the problem is that none of those whitelist patches work on my system, so I attempted to make my own NVRAM whitelist by patching VarNotInEfiGlobalVarList to return false. Unfortunately, this *still* did not work. Any ideas?

I have attached the patch I created (may or may not work), but I doubt it's correct.
```
# NvramSmi NVRAM whitelist unlock for Memphis2-S and potentially other motherboards (patches VarNotInEfiGlobalVarList, by @kaitlyndotmoe)
842A454A-75E5-408B-8B1C-36420E4E3F21 10 P:48895C240848896C24104889742418574155:48895C240848896C2410488974241830C0C3
```

xCuri0, 2023-06-30:
@kaitlyndotmoe The NVRAM patches are the only ones I didn't develop or assist with so I'm not sure how they work. But you can try replacing one of the entries in the whitelist that are non-critical with name `ReBarDxe` GUID `a3c5b77a-c88f-4a93-bf1c-4a92a32c65ce`

kaitlyndotmoe, 2023-06-30:
> @kaitlyndotmoe The NVRAM patches are the only ones I didn't develop or assist with so I'm not sure how they work. But you can try replacing one of the entries in the whitelist that are non-critical with name `ReBarState` GUID `a3c5b77a-c88f-4a93-bf1c-4a92a32c65ce`. To find the GUID you could look in the NVRAM variables list from dump opened in UEFITool NE or Linux efivars

So, I actually took a Z97-PRO ASUS BIOS and patched that and took it's NvramSmi and replaced my NvramSmi with that one, and sure enough, it worked!

[image]

But now there is another problem, ReBar isn't detected by GPU-Z. I'm starting to think this BIOS lacks proper 4G decoding despite the option being enabled?

[image]

xCuri0, 2023-06-30:
@kaitlyndotmoe it's not going to be detected without a rebar capable gpu

kaitlyndotmoe, 2023-06-30:
> @kaitlyndotmoe it's not going to be detected without a rebar capable gpu

Understood, thanks then!

closed 2023-06-30 by kaitlyndotmoe

xCuri0, 2023-06-30:
Have you confirmed that nvram write works and they stay across power cycles ?

kaitlyndotmoe, 2023-06-30:
> Have you confirmed that nvram write works and they stay across power cycles ?

As far as I'm aware, it works perfectly fine. But I'll check that and report back if it doesn't.

kaitlyndotmoe, 2023-06-30:
> Have you confirmed that nvram write works and they stay across power cycles ?

Update: it seems to persist and stay across power cycles.

MaxAbarcaC, 2023-10-26:
> > @kaitlyndotmoe The NVRAM patches are the only ones I didn't develop or assist with so I'm not sure how they work. But you can try replacing one of the entries in the whitelist that are non-critical with name `ReBarState` GUID `a3c5b77a-c88f-4a93-bf1c-4a92a32c65ce`. To find the GUID you could look in the NVRAM variables list from dump opened in UEFITool NE or Linux efivars
> 
> So, I actually took a Z97-PRO ASUS BIOS and patched that (as it's NVRAM whitelist can be patched) and took it's NvramSmi and replaced my NvramSmi with that one, and sure enough, it worked!
> 
> [image]
> 
> But now there is another problem, ReBar isn't detected by GPU-Z. I'm starting to think this BIOS lacks proper 4G decoding despite the option being enabled?
> 
> [image]

Hi, @kaitlyndotmoe, can you tell me what was the BIOS version you took? I'm afraid I'm facing the same issue than you with an ASUS Z97-AR 3503

Icedrow, 2024-02-08:
Can you please add more details? I have the same shitty hp motherboard