xCuri0/ReBarUEFI #67: Supermicro X10DRG-OT+-CPU Broadwell Re-Bar on linux working and not.

Ph0rk0z, 2023-07-06:
I have the latest bios on this board. It has 3 GPU. 2x3090 and a P40.

Above 4g is enabled and supported. No bios setting for re-bar.

The P40 is using a 32gb bar. One 3090 supports resizing, the second I need to flash a vrom update on.
Unfortunately both 3090s only use 256MB bar. I have fairly recent nvidia 5.xx driver. Is there some reason one 3090 doesn't use 32g bar right now and how can I force it?

xCuri0, 2023-07-06:
@Ph0rk0z You need to use option in NVIDIA driver `NVreg_EnableResizableBar=1` to do resizing by Linux kernel

or use ReBarUEFI to do it by UEFI.

closed 2023-07-06 by xCuri0

Ph0rk0z, 2023-07-06:
Thank you, I will try to enable it. Am trying to avoid patching this bios and fiddling with backing up with HW programmer.

Ph0rk0z, 2023-07-06:
Unfortunately enabling it did not help. I set nvidia.NVreg_EnableREsizeableBar=1 in kernel and it shows enabled but my 3090 still says 256 selected but 32gb supported. I am trying flashed bios now. None of the patches applied except for the broadwell USB3 fix. I will build the linux executable and see if it enables after I run that.

Ph0rk0z, 2023-07-06:
Ok.. I have ran the PE and set size of 32. 2 of my cards show correct big BAR right now.. I have to flash the 3rd to support it. You can call this mobo working.

Ph0rk0z, 2024-12-29:
It's likely 3.3v chip, its a server. No need to unlock FD or do any crazy stuff here. The BMC can recover bad flashes if you got one. On mine, bios is signed but no bootguard was enabled. Try to read the whole chip with intel FPT.

Ph0rk0z, 2024-12-29:
Serials and macs should be in the nvram part of the bios. I guess for this version, you're right, there's no linux FPT, only for newer gens.

BMC seemed to read the whole thing when I used it. All update files are the full image but I didn't edit the FD.

Didn't try to turn off ME because of the same worry. 

[image]

Ph0rk0z, 2024-12-31:
On mine it did the USB patch only I think.