xCuri0/ReBarUEFI #68: ASUS Prime Q270M-C ReBAR remains disabled after module + patches

LatiosAeon, 2023-07-06:
Hi, I cannot seem to get ReBAR to actually turn on no matter what I try with this motherboard.

So far I've tried:
-disabling CSM, enabled 4G decoding, even setting it by default in AMIBCP
-following the guide exactly, rebar driver placed at end of table containing pci, patches applied too
-also tried following the steps exactly using older files/extracted ffs and older patches from the time I successfully did it with two ASUS Z170 boards.

The ReBarState exe does successfully write the value and remember it on reboot but still remains disabled.

No matter what, the result is always the same, ReBAR stays disabled in both hwinfo and gpu-z.
the RTX 3080 being used does support it as tested in my W790 system.

xCuri0, 2023-07-07:
Have you tried using MMTool for inserting the module ?

What method was used to flash it ?

LatiosAeon, 2023-07-07:
Tried MMTool to insert the module and secure boot disabled, still the same result, it simply will not enable.

Been flashing the bios chip with a CH341A

LatiosAeon, 2023-07-07:
[Q270.zip](https://github.com/xCuri0/ReBarUEFI/files/11977319/Q270.zip)

This BIOS is the latest one I tried, has the ReBar module inserted with MMTool and the patches applied

xCuri0, 2023-07-07:
@LatiosAeon Maybe try inserting it in a different place ? Such as directly after PciBus or before IntelGigabitLan

LatiosAeon, 2023-07-07:
Just tried inserting it directly below pcibus, same result, this board is flat out refusing to just work.

xCuri0, 2023-07-07:
@LatiosAeon I guess it's because ReBarDxe ffs isn't compressed, it seems like every other module in this BIOS is. Are the other Z170 boards you have similar (which it works on)

LatiosAeon, 2023-07-07:
I'm not too sure how similar this board is to the Z170's, this one has the classic grey/blue UI but still works the same.
Last time it was quite simple to get it working, add module, extract and replace then run patches and it just worked, I did the same  on this board after the first failed attempt, everything I have tried so far, it acts as if the module doesn't exist.

xCuri0, 2023-07-07:
It should work if you can build the module compressed with LZMA or EFI 1.1

It's likely that ASUS removed support for uncompressed modules.

LatiosAeon, 2023-07-07:
I'll have to call it quits here as I've no clue how to do that just yet..

xCuri0, 2023-07-08:
@LatiosAeon You can try this build, it's compressed EFI 1.1

[ReBarDxeEFI11.zip](https://github.com/xCuri0/ReBarUEFI/files/11991411/ReBarDxeEFI11.zip)

I built it with the following commands

```
GenSec -o pe32.sec ReBarDxe.efi -S EFI_SECTION_PE32
GenSec -o name.sec -S EFI_SECTION_USER_INTERFACE -n ReBarDxe
GenSec name.sec pe32.sec -o comp.sct -S EFI_SECTION_COMPRESSION
GenFfs -g a8ee1777-a4f5-4345-9da4-13742084d31e -o ./ReBarDxeEFI11.ffs -i comp.sct -t EFI_FV_FILETYPE_DRIVER --checksum
```

Can also try LZMA compression if this doesn't work

[image]

LatiosAeon, 2023-07-08:
Tried again with the new build using a fresh BIOS + patches, still the same result sadly.

xCuri0, 2023-07-09:
Maybe you could try replacing RaidDriver PE32 image with that from ReBarDxe (you need to extracted it from a BIOS with ReBarDxe)

[image]

And make sure secure boot for "Internal FV" is off

LatiosAeon, 2023-07-09:
I can't find such option in BIOS or AMIBCP, this board is not very good also, i had attempted to disable every single secure boot option i could see by default, every single thing that has been tried is always the same result, it acts as if the module isn't present.

LatiosAeon, 2023-07-09:
Ok, some good news, I had started again from scratch inserted new .ffs module both from the main page and the compressed one and applied patches, for no reason I can find its now working perfectly with both modules, I don't know why it kept failing before even though I started fresh almost every time but somehow its enabled now. tested with both a 3080 and a 3090
Briefly tested with a game and it does show some improvement so it does appear to be properly working on the unlimited size.

ASUS Prime Q270M-C, RTX 3090 FE
[image]

xCuri0, 2023-07-09:
Does inserting 2 uncompressed modules also work ? If so then I won't have to add a compressed build.

Was this using MMTool or uefitool ?

LatiosAeon, 2023-07-09:
I managed to get another Q270M-C board and used one BIOS with the normal module and one with the compressed, they are both working, module was inserted with MMTool only.

I'll attach the latest bios that is working as I really don't know why it wouldn't enable before.. This one is using just the normal module
[Q270.zip](https://github.com/xCuri0/ReBarUEFI/files/11995593/Q270.zip)

xCuri0, 2023-07-09:
Can I close this as it's considered working ? But probably need to figure out why it didn't work initially

LatiosAeon, 2023-07-09:
Yes, it can be closed, I'll try experimenting with the spare board to see what caused it to fail before

closed 2023-07-09 by xCuri0

jonronnquist, 2024-06-14:
Hopefully someone is still following this discussion or I'm going to be writing to myself. Oh well, here it goes. 

I just purchased a brand new ASUS Prime Q270M-C board which I am using in a system with an Intel Arc A380 GPU. I only bought the A380 as I needed a card with 3 x DP and I got a good deal on it. It was only after being reminded by ARC Control utility that the card would be a lot happier with Resizable Bar enabled that I even considered looking into it. And that's when I found this thread. 

Cut to the present and I've bought a CH431 programmer and a new blank BIOS chip to try out the working file posted here on 9 July 2023. The problem I have run into is that the BIOS in the zip file above is a *.fd file and the software I am using with the CH431 seems to want either a *.rom, *.hex or *bin file. I did attempt to download the latest bios straight from ASUS and program the new chip with this just to see if it was working, but that was a *.cap file and the software complains that it's too long. 

Anyway, I'm guessing there is a relatively easy way to get this *.fd file onto the chip and working on the motherboard, but that my inexperience with this field has placed it out of my reach. I did consider doing a deep dive into the subject to see if I could figure it out, but I'm hampered by the fact that it's all a bit too complicated for me and I don't really have the time it would take. The fact that I don't really need to do this as I have no intention of using the A380 to do anything demanding such as play games (not that it really can) also makes it hard to muster the needed resolve. 

That said, if the solution is relatively simple, it would be kind of cool to get ReBar up and running on this system all the same. So if anyone who took part in this discission could possibly point me in the right direction, I would be truly grateful for the help. 

Fingers crossed!

jonronnquist, 2024-06-14:
Ps. LatiosAeon, I had hoped to contact you directly but sadly it appears there is no way to DM on this platform. I can only hope you are still subscribed to this thread and receive notifications.