xCuri0/ReBarUEFI #69: hidden above 4g decoding and rebar in bios not showing after doing the steps

zigiiiii, 2023-07-07:
So i tried the to do the tutorial and idk why but in my gpu z above 4g decoding is turned off and also the resizable Bar enabled in the bios
[image]

if someone is willing to teach me please reply and if you can i can give my discord and i can share screen it there

my specs
board: fm2a68m-dg3+
16 gb ram
rx 460 Sapphire 2gb single fan
amd a8 7600

xCuri0, 2023-07-07:
follow the guide for enabling hidden 4g decoding ?

zigiiiii, 2023-07-07:
i did it and still says its turned off

zigiiiii, 2023-07-07:
does above 4g decoding affect on 2gb gpus?

zigiiiii, 2023-07-07:
since i my 4g decoding is showing its turned off in gpu z, I cant do DSDT patching

Electron-PSO, 2023-07-08:
Hello @zigiiiii,

Your BIOS file doesn't have hidden Above 4G Desode.

I would add the ReBarDxe.ffs to the latest bios file. Then set ReBAR to 256MB, 512MB, 1GB and 2GB and testing it for performance improvements. You might find that the improvements drop off as you get to 2GB. In which case, you can save yourself the effort of trying to get the full BAR size working.

If at any point it fails to boot, clear the CMOS and set the last value that worked.

Best regards,

-68k

zigiiiii, 2023-07-08:
Hi, im sorry this is a dumb question but how do i set it to that exact rebar value?

Electron-PSO, 2023-07-09:
It is powers of 2

2^8=256
2^
2^
2^

xCuri0, 2023-07-16:
closing due to inactivity

closed 2023-07-16 by xCuri0