xCuri0/ReBarUEFI #75: Does ReBarUEFI sacrific multi-GPU support after patching BIOS?

terryjx, 2023-08-13:
**System**
 - Motherboard:  ASUS X99m WS, ASUS X99-E WS,
 - BIOS Version:  Any of the newest version
 - GPU:               RTX A6000 (Ampere),  NVIDIA A100, NVIDIA Titan V

**Description**

After ReBar UEFI is successfully enabled, and RTX A6000 is installed on the 1st slot, if the 2nd NVIDIA high end GPU (for example 2nd GPU is Titan V) is installed on the other PCI-E 16x slot, the entire system will either not boot or will not recognize the 2nd GPU at the BIOS level. 

Tried different slots when the 1st slot (nearest to CPU socket is installed with RTX A6000), and it's the same as above.

One setup that I find working is having one AMD graphic card (very low end, 2013 model, single slot, low profile one), and one high end NVIDIA GPU like A6000 or A100 on the other slot. In this case, it works without any problem. 

Not entirely sure how ReBarUEFI managed to get Resizable Bar work on X99 mobo, I really like the idea, but just curious know whether multi-GPU is meant to be sacrificed, and whether it could be fixed.  Just purely trying to have some brain storming from technology perspective to understand the possibility.

xCuri0, 2023-08-14:
@terryjx it's probably that your motherboard doesn't have enough BAR space for multiple GPUs with resizable BAR activated. this can be fixed most of the time though as the cpu often supports more than what the bios uses

send a screenshot of this device manager -> view by resources expand large memory with one card
[image]

look in your BIOS for settings related to MMIOH, make sure its size is set to the largest

xCuri0, 2023-08-27:
@terryjx update ?

terryjx, 2023-08-29:
I'm not with my computer this month, will try and update once I'm back.

Will keep you posted

beyawnko, 2024-09-09:
@terryjx do you have a copy of a good working bios for the x99m WS? The one posted somewhere in here seems to get corrupted over time, and all my attempts do the same. Sends it into a no post situation until i spam cmos clear forever

zhangyu1216-SC, 2026-04-02:
@terryjx do you have a copy of a good working bios for the x99m WS?