xCuri0/ReBarUEFI #78: P8Z77-V unable to enable Above 4G

DunoCZ, 2023-08-30:
**System**
 - Motherboard: Asus P8Z77-V
 - BIOS Version: 2104
 - GPU: RX 6700XT
 - [ yes ] CSM is turned off
 - [ yes - but ] 4G decoding is enabled
 - [ yes - but ] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [ yes ] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

Greetings everyone,
I have an problem, I'm trying to patch bios for P8Z77-V, if I only include the ReBarUEFI module it works but I can't somehow enable Above 4G Decoding even with i7 3770K. (btw yes I tried the method https://github.com/xCuri0/ReBarUEFI/wiki/Enabling-hidden-4G-decoding and it returns values I entered)
So I tried to patch my bios (https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) this results in no post and debug LED blinking between CPU and RAM, the PADs are correctly placed.

Can anyone help me?

DunoCZ, 2023-09-01:
I checked the PAD files where I put the rebar module, they seem to be the same, I will try the AMIBCP method but have to delid my i7 3770k, when I can use it I will update.

xCuri0, 2023-09-02:
ASUS almost always has pad file issue are you sure

Baks456, 2023-09-03:
Same issue with asus p8z77-v. 
Cannot unavle to turn on Above 4g.
original 2104 bios https://drive.google.com/file/d/17BOy0uhYxdhqKIqxHWP8-6_Yw_uIstA_/view?usp=sharing
2104 bios + NVME mode https://drive.google.com/file/d/1ApQzEJT-XIAIzMFrEvKksMbryViXNBeE/view?usp=sharing

"Enabling hidden 4G decoding" method do not change anything. CPU-Z says "disable"
amibcp_v4-55-0070 method also do not show anything in MB bios
[image]
[image]

Baks456, 2023-09-03:
Some part fixed. You need change to user in only ONE place.
And you will get new menu in the bios
[image]

xCuri0, 2023-09-04:
@romulus2k4 able to help with this ?

DunoCZ, 2023-09-04:
> Some part fixed. You need change to user in only ONE place. And you will get new menu in the bios where you can activate 4g [image]
> 
> my bios nvme +4g + rebar without patches. https://drive.google.com/file/d/1jDvRtMwrMmT-B4CaMsFkQ-b9r7eb9tOA/view?usp=sharing

Thank you, I was able to patch it, and now it's working perfectly!

@xCuri0, you were right, even though there was no obvious difference, something happened to those PAD files so I had to use the MMTool method.
Without patches, in case there was a POST interruption, like the CPU fan spinning way too slowly, the system would be soft-bricked.
Now it works flawlessly. I used the patch.txt and IvyUSB3.txt files.

[image]

If allowed, I can share the resulting BIOS.

Baks456, 2023-09-04:
> If allowed, I can share the resulting BIOS.

Do it. 
Have you made the following changes yet? [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) ? 
Or it is fine without?

DunoCZ, 2023-09-04:
> > If allowed, I can share the resulting BIOS.
> 
> Do it. Have you made the following changes yet? [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) ? Or it is fine without?

I looked into it and it seems to me that it doesn't need to be patched, I played Forza 5 without a problem so far.
Here is how it looks like:
```
                Else
                {
                    CreateQWordField (BUF0, \_SB.PCI0._Y0F._LEN, M2LN)  // _LEN: Length
                    CreateQWordField (BUF0, \_SB.PCI0._Y0F._MIN, M2MN)  // _MIN: Minimum Base Address
                    CreateQWordField (BUF0, \_SB.PCI0._Y0F._MAX, M2MX)  // _MAX: Maximum Base Address
                    M2MX = 0x0000000FFFFFFFFF
                    If ((TUUD >= 0x1000))
                    {
                        M2MN = (TUUD << 0x14)
                    }
                    Else
                    {
                        M2MN = 0x0000000100000000
                    }

                    M2LN = ((M2MX - M2MN) + One)
                }
```

Baks456, 2023-09-04:
my mod bios (seems all ok) Asus p8z77-v
based on 2104  + NVME + above 4g + rebar + patch rebar
https://drive.google.com/file/d/1iZCs8gfIdqxcELeb-RKBOa-EzGWiN4W6/view?usp=sharing

DunoCZ, 2023-09-04:
> my mod bios (seems all ok) Asus p8z77-v based on 2104 + NVME + above 4g + rebar + patch rebar https://drive.google.com/file/d/1iZCs8gfIdqxcELeb-RKBOa-EzGWiN4W6/view?usp=sharing

Sadly for me yours when rebar is set to 16GB turns on red light for PCI-E.
So here is mine for anyone else in case they face same issue.

Closing issue since for me everything is solved (except missing cursor when I enable rebar, but that isn't big deal for me.)
[Z77VB.zip](https://github.com/xCuri0/ReBarUEFI/files/12516874/Z77VB.zip)

closed 2023-09-04 by DunoCZ

xCuri0, 2023-09-05:
ASUS Z77 often don't need DSDT patching yeah

Kur0ex, 2024-03-07:
> > my mod bios (seems all ok) Asus p8z77-v based on 2104 + NVME + above 4g + rebar + patch rebar https://drive.google.com/file/d/1iZCs8gfIdqxcELeb-RKBOa-EzGWiN4W6/view?usp=sharing
> 
> Sadly for me yours when rebar is set to 16GB turns on red light for PCI-E.
> 
> Closing issue since for me everything is solved (except missing cursor when I enable rebar, but that isn't big deal for me.)

hello, I flash it, but i dont have any above 4g options. Could you tell me how to open it?

I flash your mod via usb flashback

dsanke, 2024-03-08:
[Z77VB.CAP.zip](https://github.com/xCuri0/ReBarUEFI/files/14533479/Z77VB.CAP.zip)
based on P8Z77-V-ASUS-2104, modifications :
1. uefipatched (bar size limit removal, usb3 whitelist)
2. add nvme support
3. above 4g decoding = on by default and can not set off ( i didnt unhide the option)
4. video oprom policy = uefi only by default
5. add rebardxe with unlimited BAR size, regardless of value set by rebarstate
feedback from @Kur0ex , works without issue.

Kur0ex, 2024-03-08:
> [Z77VB.CAP.zip](https://github.com/xCuri0/ReBarUEFI/files/14533479/Z77VB.CAP.zip) based on P8Z77-V-ASUS-2104, modifications :
>
> 1. uefipatched (bar size limit removal, usb3 whitelist)
> 2. add nvme [...]

Thanks a lot. Very perfect!

Nisetie, 2025-03-30:
> Thanks a lot. Very perfect!

How did you enable Rebar? I installed dsanke's BIOS using USB Flashback. Everything seems unchanged. GPU-Z says OK in Requirements but Rebar disabled in BIOS.