xCuri0/ReBarUEFI #79: ASRock Z77 Extreme4 Will Sometimes Fail to Boot With Error 62

aolszowka, 2023-08-31:
**System**
- Motherboard: [ASRock Z77 Extreme4](https://www.asrock.com/mb/intel/z77%20extreme4/) - Modifications Start from Latest 3.00 Beta BIOS Available From ASRock
- GPU: [ASRock Intel Arc A380 Challenger ITX 6GB OC](https://www.asrock.com/Graphics-Card/Intel/Intel%20Arc%20A380%20Challenger%20ITX%206GB%20OC/)
- OS: Windows 10 22H2
 - [x] CSM is turned off
 - [x] 4G decoding is enabled
 - [x] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [x] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
I was able to successfully modify the BIOS to allow ReBAR to work as per my post here: https://github.com/xCuri0/ReBarUEFI/issues/11#issuecomment-1605868162

However, every so often, most noticeably during a Windows Update "Install Updates and Restart" Cycle my board will fail to boot and display an Error 62 on the ASRock Motherboard Dr. Debug Readout. Looking online this error code relates to "Chipset initialization error." Clearing the CMOS and then restarting always has resolved the issue, but it just means I need to reapply my settings.

This doesn't appear to occur when a "normal" reboot happens or I shutdown and restart the machine. Again I have only noticed this on a Windows Update "Install Updates and Restart" Cycle or when automatic updates are executed. Any ideas on what might be different here?

I have the modified BIOS if that would help.

xCuri0, 2023-08-31:
@aolszowka have you checked your CMOS battery voltage ? hwinfo should say it

aolszowka, 2023-08-31:
@xCuri0 This is VBAT right? Seems to read at 3.248? I honestly don't know what expected range should be:

[image]

I'd be happy to swap out the battery its just a CR2032; I don't think I've ever replaced this (build occurred in ~2011).

Is there something that leads you to think this might be the issue?

xCuri0, 2023-09-01:
@aolszowka voltage looks ok

Maybe you could redo the BIOS modification but use MMTool to replace modules (pad file workaround) because I've heard UEFITool sometimes causes corruption like this.

aolszowka, 2023-09-01:
@xCuri0 Thank you for the reply, honestly its not that big a deal and really only happens on Windows Updates. It seems so bizarre that an `Update and Restart` is somehow different than a normal restart. I'll keep this open until Patch Tuesday (2023/09/12) and see if it reproduces again if that is alright?

aolszowka, 2023-09-12:
@xCuri0 Closing the loop on this, Patch Tuesday (2023/09/12) after installing Windows Updates and selecting `Update and Restart` I experienced the same issue. This machine rebooted (`Restart`) and Shutdown just fine. Absolutely bizarre. I am going to close out the issue as I have not encountered any other issues with this mod, but perhaps at some point in the future I'll be able to figure out just what is different between an `Update and Restart` and `Restart`. Next Patch Tuesday I am going to attempt an `Update and Shutdown` to see if that yields any different results.

closed 2023-09-12 by aolszowka

Naddly, 2024-03-12:
@aolszowka I hope I'm not bothering you, I have this same motherboard and the desire to have ReBar working but I have no experience with BIOS modification and I hope you can provide me with the already modified BIOS file if you still have it, accompanied by any tips that might be useful in the process. Thank you in advance. 😃

aolszowka, 2024-03-13:
@Naddly I've got more details in my post here: https://github.com/xCuri0/ReBarUEFI/issues/11#issuecomment-1605868162

This part was critical I feel:

"During my entire process I actually had a CH341 Programmer available to me along with a bunch of spare W25Q64BVAIG's. The chip is socketed on this motherboard which makes swaps super simple and I always had a way to fall back when something went wrong. I had to use AsProgrammer_2.1.0.13_fix that I found off some YouTube guide, I picked up the programmer from Amazon and the spares from eBay from China (took 2 weeks to get here, but was worth it)."

Feel free to contact me @ my username from a mail service from Google.

Naddly, 2024-03-14:
@aolszowka I just sent you an email, though I'm not quite sure if I got it right. Is it aolszowka @ (google mail) com?