xCuri0/ReBarUEFI #83: Performance problem on Linux but not on Windows with ReBAR v0.3 on Jingsha X99 D8-DDR3

ms178, 2023-09-11:
**System**
 - Motherboard: Jingsha X99 D8-DDR3
 - BIOS Version: [CX99DE26](https://github.com/BIOS-iEngineer/HUANANZHI-X99-TF/releases/tag/CX99DE26)
 - GPU: AMD 6950 XT - XFX Speedster Merc
 - [x] CSM is turned off
 - [x] 4G decoding is enabled
 - [ ] UEFIPatch is applied
 - [x] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

I use Windows 11 on a SSD and CachyOS (Arch-derivative) on a NVMe drive on the same X99 machine. With the 0.3 version of the ReBAR ffs driver, I see problematic performance in the first benchmark scene of Total War: Troy ("battle benchmark"). I used to get 148 fps avg on Linux and 140 fps avg on Windows. With ReBAR 0.3, performance tanks to around 118 fps avg for both operating systems in that scene. For Windows, I am able to recover the performance by running the ReBARState script and set the recommended value of 32 which then completes the ReBAR enabling process. GPU-Z also reports that ReBAR is working fine then. The strange thing is that the two other benchmark scenes of that game show their expected results on Linux. It is only the first benchmark that still misbehaves. I also haven't noticed performance problems with other Linux games yet.

Linux also shows the complete ReBAR being used. With an earlier version of the ReBAR ffs driver (I think it was 0.1), I used to get great performance on either Windows and Linux and without/with completing the ReBARState script. I can reproduce this behavior with using both the Windows and Linux version of the script.

I appreciate any help, I could try to set different values in the script but would need guidance on the optimal value for my case.

```
❯ sudo dmesg | grep "root bus resource"
[    1.532852] pci_bus 0000:ff: root bus resource [bus ff]
[    2.557224] pci_bus 0000:00: root bus resource [io  0x0000-0x0cf7 window]
[    2.557227] pci_bus 0000:00: root bus resource [io  0x1000-0xffff window]
[    2.557230] pci_bus 0000:00: root bus resource [mem 0x000a0000-0x000bffff window]
[    2.558711] pci_bus 0000:00: root bus resource [mem 0x90000000-0xfbffbfff window]
[    2.558712] pci_bus 0000:00: root bus resource [mem 0x380000000000-0x38ffffffffff window]
[    2.558714] pci_bus 0000:00: root bus resource [bus 00-fe]
```

```
[  +0,000006] [drm] vm size is 262144 GB, 4 levels, block size is 9-bit, fragment size is 9-bit
[  +0,000006] amdgpu 0000:05:00.0: amdgpu: VRAM: 16368M 0x0000008000000000 - 0x00000083FEFFFFFF (16368M used)
[  +0,000003] amdgpu 0000:05:00.0: amdgpu: GART: 512M 0x0000000000000000 - 0x000000001FFFFFFF
[  +0,000002] amdgpu 0000:05:00.0: amdgpu: AGP: 267894784M 0x0000008400000000 - 0x0000FFFFFFFFFFFF
[  +0,000008] [drm] Detected VRAM RAM=16368M, BAR=16384M
[  +0,000001] [drm] RAM width 256bits GDDR6
[  +0,000137] [drm] amdgpu: 16368M of VRAM memory ready
[  +0,000001] [drm] amdgpu: 15970M of GTT memory ready.
[  +0,000009] [drm] GART: num cpu pages 131072, num gpu pages 131072
[  +0,000097] [drm] PCIE GART of 512M enabled (table at 0x0000008000300000).
```

xCuri0, 2023-09-11:
So 4G decoding is causing fps drop for you ?

0.1 and 0.3 function identically besides the CMOS reset detection

ms178, 2023-09-11:
Just to be clear, 4G decoding was enabled all the time. As to the performance, I've seen some fluctuation depending on the used BIOS settings (I could get it to 131 fps avg on Linux with some tweaks, but it is still well below 148 fps I got before in that particular benchmark scene).

I don't know if vkd3d-proton or Mesa or other parts of the Linux software stack are contributing to that problem, I saw that vkd3d-proton disables reBAR for some games specifically due to performance problems. I've experimented a lot with different BIOS settings, forcing SAM with RADV_PERFTEST=sam - without success yet. I also already re-installed CachyOS and used the default packages, that also had no effect. I first saw the regression when using the newly-modded BIOS however.

xCuri0, 2023-09-11:
Linux can use resizeable bar without the BIOS supporting it btw, it only needs 4g decoding

ms178, 2023-09-11:
I've also seen a lot of PCI-related problems in the past with X99 and Linux, e.g. PCI AER errors when using my i350 network card. I need to use `pcie_aspm=off` to get rid of that, so there is potential that some obscure PCI problem hits me here, too.

ms178, 2023-09-11:
I've now gone back to my old BIOS that used to work fine (I cannot bench the game right now due to game copy protection limits, but will edit this post later if the downgrade got rid of the issue). I found a few differences while comparing the Kernel logs: https://www.diffchecker.com/TL1qvBHp/ (the old BIOS is on the right, the newer BIOS on the left).

1. ACPI tables
- the old BIOS does load a couple more ACPI tables: MPST, MSCT, SLIT, SRAT

2. The values for the PCI bridge window are different (values below are for the GPU)

- for the old BIOS: bridge window [mem 0x383800000000-0x383c0fffffff 64bit pref]
- for the new BIOS: bridge window [mem 0x38f800000000-0x38fc0fffffff 64bit pref]

ms178, 2023-09-11:
Closing, as I can still reproduce the issue with the older BIOS.

closed 2023-09-11 by ms178