xCuri0/ReBarUEFI #86: [Asking for help] P40 with ASUS P8H67-M PRO

crashr, 2023-09-20:
**System**
 - Motherboard: ASUS P8H67-M PRO
 - BIOS Version: 2.10.1208
 - GPU: NVIDIA Tesla P40
 - [✓] CSM is turned off
 - [✓] 4G decoding is enabled
 - [✓] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [✓] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
Some more info:
i5-3570K 
32GB RAM 
Ubuntu 22.04.3 LTS

I think I face the last issue from here https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes) (Using 32GB or larger BAR on LGA1155/1150) but don't know how to proceed. As far as I know I don't need Resizable BAR enabled to use a single P40 but it doesn't harm either. If I undertsand right I need to enable Above 4G so I (think) I did that. I assumed I need to do the DSDT Patching. But when I flashed the ROM, the PC shutdown suddenly and didn't do anything after that. No POST, now fans starting to spin, no screen wakeup anymore. Luckily I was able to flash the UEFIPatched ROM back via RPi SPI header.  Now I am unsure how to proceed. Is it even possible to use this GPU with this PC? I saw this issue https://github.com/xCuri0/ReBarUEFI/issues/77 . Is that related to my problem? Do I need to reduce to 8GB RAM to make the GPU work?

xCuri0, 2023-09-21:
Does your PC POST with the P40 in if so you can just follow what's written

crashr, 2023-10-01:
You mean before making changes to the BIOS? Yes, the PC POSTs and ich can see the GPU with lspci. After patching the BIOS PC POSTs as well but I had no luck getitng the P40 to run. Then when trying to flash the DSDT patched ROM something went wrong. After that the PC didn't POST anymore with or without GPU. I hat to flash the patched (not DSDT patched) ROM vi SPI. My problem is that I don't know if this was just a random fail or if it was caused by the ROM. The sizes of all ROMs look good to me:

$ stat -c "%s %n" ROMs/* 
4194304 ROMs/P8H67-M-PRO-ASUS-3904.ROM 
4194304 ROMs/P8H67-M-PRO-ASUS-3904.ROM.dsdt.patched 
4194304 ROMs/P8H67-M-PRO-ASUS-3904.ROM.patched 

I am not sure what you mean with what's written. As fas as I understand I need to modify the BIOS to get above 4G decoding ans also I need DSDT patching which requires to flash the now twice patched ROM. Do I missunderstand thomething? Sorry for asking dump questions. And thanks for providing this.

xCuri0, 2023-10-01:
You do not need to flash BIOS

**Disable** 4G decoding and do the DSDT modification on #77 using grub or initrd to do the override

And try using 8GB RAM at first

xCuri0, 2023-10-01:
See https://github.com/xCuri0/ReBarUEFI/wiki/Sandy-and-Ivy-Bridge-32GB-BAR-DSDT-modification

crashr, 2023-10-01:
Thank you so much! The GPU works now with 8GB RAM. I don't understand in detail, what other restrictions a have to deal with but I am so glad that the GPU now does at least something. I will still have a look at your linked wiki article.

xCuri0, 2023-10-01:
Tried using more than 8GB RAM ?

The wiki guide is identical to what you did btw

closed 2023-10-01 by xCuri0

crashr, 2023-10-01:
Yes, I tried 16GB but then again got these "BAR ... invalid .." messages.