xCuri0/ReBarUEFI #88: Asus Z170i pro gaming ReBar NotWorking

ShockTime, 2023-10-04:
**System**
 - Motherboard: Asus Z170i pro gaming ( CPU Xeon E-2186m)
 - BIOS Version: 3805
 - GPU: sapphire 6700XT
 - [X] CSM is turned off
 - [X] 4G decoding is enabled
 - [X] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [X] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

Describe the issue here. 
I used UEFIPatch. Mod bios with coffemod 0.99 and .ffs file and flash afudos. This hack not working for me. Please help!
[https://disk.yandex.ru/i/w7zcu3hwBwPRMg](url)

xCuri0, 2023-10-04:
Did you run ReBarState ?

ShockTime, 2023-10-04:
Yes of course
[https://disk.yandex.ru/i/_7lECdmcSRPmWA](url)

xCuri0, 2023-10-04:
@ShockTime your rebarstate is set to 256mb. that is why it won't work.

you're supposed to set it to 32 (unlimited) or the size you want (16gb in this case)

ShockTime, 2023-10-04:
@xCuri0 
Anything more than 256Mb does not allow you to load the system in a normal state.
Error with loading W11. "b1initialize library faild 0xc00000BB" and stop system.

xCuri0, 2023-10-04:
@ShockTime what size did you try ?

try doing 32, some of these z170 boards don't like 2/4gb bar size.

it should work fine, i've seen some people have POST issues but windows always worked.

ShockTime, 2023-10-04:
@xCuri0 
8,9,10
All but 8 caused problems and I stopped the experiments. 8 it half-work default.

xCuri0, 2023-10-04:
@ShockTime use 32

ShockTime, 2023-10-04:
Goodjob. 
'32-unlimited" worked. But pc speaker do it 1 long 3 short (2 cycle) and load windows normaly. 
Start menu bios "black screen".
[image]

xCuri0, 2023-10-04:
@ShockTime that is normal others report the same

closed 2023-10-04 by xCuri0

ShockTime, 2023-10-04:
@xCuri0 
It's worked for me but realtekHD error ^_^

xCuri0, 2023-10-04:
See #4 same issue

Enterprise12nx01, 2025-03-30:
I want to do this modification for the same board, what should i do?
first step when searching "4G Decod" i get 2 options from search 

[image]