xCuri0/ReBarUEFI #90: Linux Rebar N

Halteproblem, 2023-10-14:
**System**
 - Motherboard: MSI B150M MORTAR
 - BIOS Version: 7972vCD
 - GPU: Intel ARC A770
 - [I have switched from Legacy+UEFI -> UEFI] CSM is turned off
 - [ x] 4G decoding is enabled
 - [ ] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [ x] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
 I am using POP OS 22.04 and the "Mesa Intel® Arc(tm) A770 Graphics (DG2)" drivers.

After updating my bios with the newest version from MSI, I activated the 4G Decoding in the BIOS.
I have updated the kernel parameters using 'sudo kernelstub -a "pci=realloc" '
And I have set the rebar value to different values ranging from 1GB to 16GB using 'sudo ./ReBarState'.
After reboot I can see the correct value being set after runing 'sudo ./ReBarState' again.

However dmesg returns:

'
[    0.338209] pci 0000:03:00.0: can't claim BAR 2 [mem 0x2f00000000-0x2f0fffffff 64bit pref]: no compatible bridge window
[    0.363873] pci 0000:00:1d.0: BAR 14: assigned [mem 0x90000000-0x901fffff]
[    0.363877] pci 0000:00:1d.0: BAR 15: assigned [mem 0x2000000000-0x20001fffff 64bit pref]
[    0.363881] pci 0000:00:1d.0: BAR 13: assigned [io  0x2000-0x2fff]
[    0.363884] pci 0000:03:00.0: BAR 2: assigned [mem 0x2fe0000000-0x2fefffffff 64bit pref]
[    0.426959] caller get_primary_reg_base+0x4f/0xb0 mapping multiple BARs
[    1.781157] i915 0000:03:00.0: BAR 0: releasing [mem 0xde000000-0xdeffffff 64bit]
[    1.781159] i915 0000:03:00.0: BAR 2: releasing [mem 0x2fe0000000-0x2fefffffff 64bit pref]
[    1.781175] pcieport 0000:02:01.0: BAR 15: releasing [mem 0x2fe0000000-0x2fefffffff 64bit pref]
[    1.781177] pcieport 0000:01:00.0: BAR 15: releasing [mem 0x2fe0000000-0x2fefffffff 64bit pref]
[    1.781183] pcieport 0000:01:00.0: BAR 15: no space for [mem size 0x400000000 64bit pref]
[    1.781185] pcieport 0000:01:00.0: BAR 15: failed to assign [mem size 0x400000000 64bit pref]
[    1.781187] pcieport 0000:02:01.0: BAR 15: no space for [mem size 0x400000000 64bit pref]
[    1.781189] pcieport 0000:02:01.0: BAR 15: failed to assign [mem size 0x400000000 64bit pref]
[    1.781191] i915 0000:03:00.0: BAR 2: no space for [mem size 0x400000000 64bit pref]
[    1.781193] i915 0000:03:00.0: BAR 2: failed to assign [mem size 0x400000000 64bit pref]
[    1.781195] i915 0000:03:00.0: BAR 0: assigned [mem 0xde000000-0xdeffffff 64bit]
[    1.781330] i915 0000:03:00.0: [drm] Failed to resize BAR2 to 16384M (-ENOSPC)
[    1.781333] i915 0000:03:00.0: BAR 2: assigned [mem 0x2fe0000000-0x2fefffffff 64bit pref]
[    1.781362] i915 0000:03:00.0: [drm] Using a reduced BAR size of 256MiB. Consider enabling 'Resizable BAR' or similar, if available in the BIOS.
'

Furthermore I can not access the bios anymore.
I get a black screen. According to the msi forum i can reset the bios to default by entering
F6 + F10 after getting the black screen.

xCuri0, 2023-10-14:
@Halteproblem Are you sure you need to use `pci=realloc`.

Atleast on my system using ReBarUEFI + Linux I don't see any messages like this of kernel trying to resize BAR. Only with 4G decoding on and **Resizable Bar off** I see these.

Also you need to use UEFIPatch.

closed 2023-10-14 by xCuri0

Halteproblem, 2023-10-14:
Yes now it works for me on Windows with 16 GB Bar Size.
The problem was just setting UEFI+LEGACY to UEFI did not deactivate CMS.
I had to enable 10WHQL Support in the MSI Mainboard.
See: https://forum-en.msi.com/index.php?threads/above-4g-decoding-kills-the-bios.301338/

Now I have to check my Pop Os Distribution. Currently it is not shown in the boot list after deactivating CMS.