xCuri0/ReBarUEFI #91: HP Z440 Above 4G Decoding

wilsoniain, 2023-10-31:
**System**
 - Motherboard: HP Z440
 - BIOS Version: 0.257
 - GPU: ARC A750
 - [N/A ] CSM is turned off
 - [? ] 4G decoding is enabled
 - [N/A] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [N/A ] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

I've looked through the BIOS for any descriptions relating to 4G Decoding / Above 4G with no success. 

I wondered: 

1. Is there any other terminology relating to this feature which could be used in the hidden menu item?
2. I assume this is the end of the road regarding this system for enabling 4G decoding if no option is found within the BIOS?

I thought this information would be useful for other to see for future reference.

Also, I've added a few BIOS files here available for searching: https://github.com/wilsoniain/Z440-BIOS

xCuri0, 2023-11-01:
I think it can also be called MMIO, 64-bit, etc 

You need to edit DSDT and in some cases on HEDT systems modify the IOH registers before OS loads using an EFI program. **This method only works on Linux unless you can resize and assign the GPU before Windows loads using an EFI program**

wilsoniain, 2023-11-01:
Any thoughts on this parameter for enabling above 4G?

`0xF1C 	Form: Miscellaneous Configuration, FormId: 0x9 {01 86 09 00 1A 16}
0xF22 		Subtitle: Statement.Prompt: Miscellaneous Configuration, Flags: 0x0 {02 87 1A 16 00 00 00}
0xF29 		End {29 02}
0xF2B 		Subtitle: Statement.Prompt: --------------------------------------------------, Flags: 0x0 {02 87 4D 16 00 00 00}
0xF32 		End {29 02}
0xF34 		Subtitle: Statement.Prompt: , Flags: 0x0 {02 87 25 0D 00 00 00}
0xF3B 		End {29 02}
0xF3D 		One Of: PCI 64-Bit Resource Allocation, VarStoreInfo (VarOffset/VarName): 0x54, VarStore: 0x1, QuestionId: 0x10, Size: 1, Min: 0x0, Max 0x1, Step: 0x0 {05 91 1C 16 1D 16 10 00 01 00 54 00 10 10 00 01 00}
0xF4E 			One Of Option: Disable, Value (8 bit): 0x0 (default) {09 07 70 08 30 00 00}
0xF55 			One Of Option: Enable, Value (8 bit): 0x1 {09 07 6F 08 00 00 01}
0xF5C 		End One Of {29 02}`

xCuri0, 2023-11-01:
@wilsoniain yes

wilsoniain, 2023-11-01:
Issue with setting variable value.

```
grub>setup_var 0x54
Looking for Setup variable.....
var name: Setup, var size: 12, var guid: ec87d643-eba4-4bb5 - a1-e5....etc

--> GUID does not match expected GUID, taking it nevertheless...
expected a different size of the Setup variable (got 5063 (0x13c7) bytes). Continue with care...
successfully obtained "Setup" variable from VSS (got 5063 (0x13c7) bytes).
offset 0x54 is: 0x00

var name: Setup, var size: 12, var guid: 492d16ab-4969-49ac - a1-f7....etc
expected a different size of the Setup variable (got 5063 (0x14ca) bytes). Continue with care...
successfully obtained "Setup" variable from VSS (got 5063 (0x14ca) bytes).
offset 0x54 is: 0x00
```
```

grub>setup_var 0x54 0x1
setting offset 0x54 to 0x01
error: can't set variable using efi (error: 0x80000000000000f)

```

wilsoniain, 2023-11-01:
Is this an issue with locked EFI? 

Can the default value be updated in the BIOS to 0x1 file along with adding FFS module? i.e. no need for setup_var method?

xCuri0, 2023-11-01:
@wilsoniain Is there no option like that in the regular BIOS menu ?

https://github.com/datasone/grub-mod-setup_var/issues/4 seems like it's some kind of BIOS lock.

> Can the default value be updated in the BIOS to 0x1 along with adding FFS module? i.e. no need for setup_var method?

I think some people know how to do this like the user Sweet_Kitten on win-raid forum, try asking there.

wilsoniain, 2023-11-01:
@xCuri0 Cant see it in the BIOS. Guess this would be the header in the BIOS? 

`Form: Miscellaneous Configuration, FormId: 0x9 {01 86 09 00 1A 16} `

But can't see any 'Suppress' text around that form. Would that be expected to hide the menu option?

Ok, thanks for the pointer. Will try and post in win-raid about the next step. 

Attached the IFRextract in case anyone wanted to look. 

[4G_Z440.txt](https://github.com/xCuri0/ReBarUEFI/files/13228678/4G_Z440.txt)

xCuri0, 2023-11-01:
@wilsoniain what about "Security Configuration" under Security do you have that in the menu ?

wilsoniain, 2023-11-01:
[image]
[image]
[image]

wilsoniain, 2023-11-01:
@xCuri0 Can't see BIOS lock.

wilsoniain, 2023-11-01:
@xCuri0 .....really sorry. Just found it under performance. Not sure how I missed before.

[image]

closed 2023-11-01 by wilsoniain

rippiedoos, 2023-11-13:
@wilsoniain did you manage to flash a modified BIOS to the Z440 motherboard? I also want to use ReBAR but I'm not sure about all the other changes in the HP BIOS.

Joel-De, 2023-11-29:
@wilsoniain Also interested if you were able to get this working, I have the same system and was hoping to get an A770 at some point.

phoang0693, 2023-12-16:
Can you help me, I got this 
[image]

EdrumZ, 2024-01-26:
> @wilsoniain Also interested if you were able to get this working, I have the same system and was hoping to get an A770 at some point.

Same here does anybody have any info out there?.

phoang0693-a, 2024-01-26:
Sadly, no one able to turn rez bar or overclock hp z440.

lquinones89, 2024-02-06:
To add more info. Using xCuri0 method to modify the bios will not work. When you use the UEFItool to open the Bios some parts will confuse the app and not read properly. When you save those parts will not save either (seen in screenshot.) Regardless I have tried and HPZ will reject the bios and not install. I believe the contents have something to do with the bios Digital signature. the only way I think will work is if you use a different Bios mod tool specifically for the HPZ series, I just don't know where to find something like that.

[image]

xCuri0, 2024-02-06:
@lquinones89 what about MMTool ?

And there are several others too you can try. Go to UEFITool GitHub page it lists alternatives at the bottom

lquinones89, 2024-02-06:
Unfortunately not. I tried MMtool then went to the super user forums to see what could be done and the end result is you would need a custom bios modder to edit it.

[image]

EdrumZ, 2024-02-09:
Thanks Xcuri0

Sent from Tim's phone.

nkshschdv, 2024-08-02:
If you are looking into HP Z420 BIOS Version 2.15.1229 then option is under ESC (BIOS SETUP)  > Advanced > Bus options > MIMO Assignment Mode > 65 Bit

bibikalka1, 2024-08-14:
@xCuri0 

The modded Z440 bios will not flash. Checksum verification will fail! Need to find a way to bypass that.

The modded bios could be flashed on ZX20 series via a kludgy workaround:
https://github.com/bibikalka1/HP_Z420_Z620_Z820_BOOTBLOCK2013_BIOS_mod/tree/main

bibikalka1, 2024-08-15:
@xCuri0 @lquinones89 @wilsoniain @rippiedoos @Joel-De

The Z440 / Z640 / Z840 modding up to date page is here:
https://github.com/bibikalka1/HP_Z420_Z620_Z820_BOOTBLOCK2013_BIOS_mod/issues/3

DarkJoney, 2025-12-12:
Guys! Any success on this?